SUSTAINA-OP/SUSTAINA-OP #269: サーボモータ関連のタスク確認

MasatoKubotera, 2022-02-18:
## 管理システム

現状の管理システムは下記の２つ．（閲覧操作にはチームGoogleドライブ権限が必要）
- [B3M管理システム](https://docs.google.com/spreadsheets/d/1bP7htNyyIz2KI1jB_ZaqXPhJWibj-PJ-OF8uCFB9jfw/edit#gid=824152206)
- [B3M管理システム回答フォーム](https://docs.google.com/forms/d/e/1FAIpQLSdb1IHi3cyBIFQgdWhbcbmZKQIVtRktk1ASsYKrLtSqyldkWw/viewform)

## 現状

- 管理のために新規パーツケース [PS-360-CL](パーツストッカー（ブルー・クリア）)を４個導入．

- 修理用の新品 [出力軸ホーン](https://kondo-robot.com/product/02031)/[フリーホーン](https://kondo-robot.com/product/02137)はそれぞれClampHorn/FreeHornとInventorモデル・管理システムで命名し，在庫がClampHorn：6個，FreeHorn：48個確保している．ClampHornが足りない恐れがあり．

- ClampHorn軸の締結ネジを純正プラスねじから六角穴付フランジボルトの[620103008](https://jp.misumi-ec.com/vona2/detail/221000862966/?PNSearch=620103008&HissuCode=620103008)に変更している．170機についてはすでにヨー軸の3つのサーボを除き実装済み．現状の在庫数は203個．参照：#241

- 世界大会中の故障対応ように，[B3M-SC-1170-A](https://kondo-robot.com/product/03092)・[B3M-SC-1040-A](https://kondo-robot.com/product/03090)を新品状態で10個確保・隔離している．

## タスク

※優先順位で記載しているわけではありません．

作業単体でissueを必要とする場合は，このページを参照したうえで作成してください．

- [ ] #228
モータ自体の性能評価を行う
→ 評価方法を確立し，マニュアルを整備する必要がある．また，評価する際には記録するためのシステム/場所を作成する必要がある．

対象はすべての在庫．

参照：https://github.com/citbrains/citbrains_humanoid/issues/1613

- [ ] #248
バックラッシュ等によるガタの大小区分を行う
→ 調査方法を確立し，記録するためのシステム/場所を作成する必要がある．

- [ ] マーキング
マーキングの方法が確立されているため，B3M本体にマーキングを行う．
→現状のマーキング方法がissueにしか存在していなく扱いにくいため，新規パーツケースに概要を印刷し貼付してください．

- [ ] 管理システムの改良
現状，機体への使用登録が重複して登録が可能となっており，これにより一つの箇所に二つのサーボモータが使用登録されていた事例が発生した．
→ @hiroki-0001 が新規システムを設計している．

- [ ] 新規パーツケースへの移行
> 管理のために新規パーツケース [PS-360-CL](パーツストッカー（ブルー・クリア）)を４個導入．

>現状のマーキング方法がissueにしか存在していなく扱いにくいため，新規パーツケースに概要を印刷し貼付してください．

パーツケースに管理番号，管理システム・回答フォームへのQRコードの貼付も出来ればお願いします．

- [ ] 出力軸ホーンの状態確認
雌ネジがつぶれていて挿入できない箇所がある．Ver3.0までは90度間隔でネジを挿入していたが，Ver4.0では腕などでネジの挿入間隔が異なるため，すべての雌ネジが通常にネジが挿入できる必要がある．

また，出力ホーンのサイドから挿入されているクランプ用のネジが純正と異なるものが使用されていることがあるため，状態を把握する必要がある．

- [ ] 洗浄
中古サーボでは，ロックタイトを使用して締結しているものがある．
以前にロックタイトを使用した箇所で再度ネジを挿入すると，残りカスなどで適切に締結できない．
対象は，ClampHorn/FreeHornのすべての雌ネジ，サーボ本体の出力軸/フリー軸．
参照：https://github.com/citbrains/OpenPlatform_ver3_0/issues/199#issuecomment-997322243

- [ ] 出力ホーンネジの挿入
> ClampHorn軸の締結ネジを純正プラスねじから六角穴付フランジボルトの[620103008](https://jp.misumi-ec.com/vona2/detail/221000862966/?PNSearch=620103008&HissuCode=620103008)に変更している．170機についてはすでにヨー軸の3つのサーボを除き実装済み．現状の在庫数は203個．参照：#241

挿入するときには，ロックタイトと塗布する必要があります．参照：#199 ，#227 
量については，2022/2/18時点で指標が決まっていません．作業するときなどはあらかじめ@MasatoKubotera に確認してください．

また，クランプのネジについてもロックタイトを挿入したいと考えている．

- [ ] B3Mの初期設定
量産の組み立て時にBaudrate＆ID，ゲインやリーバースモードなどの各種パラメータを設定している余裕がない．そのため，あらかじめ設定しておきたい．
Baudrate＆IDについてはB3Mmanagerを使って手動設定した後に，そのほかんもパラメータついては，https://github.com/citbrains/citbrains_humanoid/issues/1573 で扱っているプログラムを使用し書き込む．

プログラム開発担当をしている @hiroki-0001 には，期限を3月8日（火）までとしている．

準備が終わった量産機体で使用するサーボモータについては，管理システムに登録し機体ごとに区分・組み立て時に管理ページを参照しなくてもいいように本体にIDを貼付してほしい．

Baudrateとパラメータの書き込みについては，量産時に使用しない在庫サーボふくめすべてが対象である．（実装中の機体にも書き込みが必要な場合がある）
>- 世界大会中の故障対応ように，[B3M-SC-1170-A](https://kondo-robot.com/product/03092)・[B3M-SC-1040-A](https://kondo-robot.com/product/03090)を新品状態で10個確保・隔離している．

→　どこかにパラメータ書き込みしたサーボを区別できるページ・リストを作成する必要がある．

参照：
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/174#issuecomment-961669421
- https://github.com/citbrains/citbrains_humanoid/issues/1573
- https://github.com/citbrains/citbrains_humanoid/issues/1569

- [ ] 固定ネジ抜き
フレームなどと締結するための固定穴表裏4か所ずつがあるが，表裏ともに4か所のうち1か所にアッパーケースとボトムケースを固定するためにネジが締結されている．
組み立て時に，ネジを抜き忘れ組み立ててしまい分解，再度組み立てする事案が発生している．そのためにあらかじめ抜いておく．抜いた時にケース自体が分裂することは無いためネジを抜くことを問題視していない．
参照：https://github.com/citbrains/OpenPlatform_ver3_0/issues/174#issuecomment-960564058

- [ ] 例外サーボの扱い
下記の3つのサーボモータの扱いを決めて対処する．このモータは現状 @MibuchiYuta が管理している．
  - *40-1918 ボトムケースが3Dプリンター製
  - 70-1982　171 | 5 (右足付け根ロール)に重複して登録されていた．RCAP前から状態不明．
  - 70-2005 ボトムケース紛失
 
[image]

- [ ] 修理申請の用意
B3M-SC-1040-Aを優先的に修理に出す．修理日は，4月1日以降である必要がある．この日程については修理日が近づいたら要確認．
基本的には，管理システムの「[［B3M］故障リスト](https://docs.google.com/spreadsheets/d/1bP7htNyyIz2KI1jB_ZaqXPhJWibj-PJ-OF8uCFB9jfw/edit#gid=372255127)」を参照することで故障ログは取得できる．
以前の修理に関して：#99

aiba0921, 2022-02-21:
@hiroki-0001 
163機体からサーボモータを取り出したときに40に「＊」を付け忘れて在庫登録していたものがありました。

hiroki-0001, 2022-02-21:
@aiba0921 @

aiba0921, 2022-02-21:
@hiroki-0001 
＊40-1908：右肩ピッチ
＊40-1909：左肘ピッチ
＊40-1910：左肩ピッチ
＊40-1911：首ヨー
＊40-1914：右肘ピッチ
以上のサーボが「＊」無しで在庫登録されています。

Keita1Mori, 2022-02-21:
# 報告
出力軸ホーンの状態確認
## 担当者
@Keita1Mori @KentaTsukino @aiba0921 
## 現状終わっている個数
月野　20個　
相場　18個
森　15個　1時間半
## 洗浄が必要なサーボ
70-1916
70-1990
70-1991
70-1993
70-2006
70-2008
## 側面に傷あり
70-1973
70-1974

aiba0921, 2022-02-21:
## 追加
10個　1時間
## 洗浄が必要
＊40-1909
## 側面に傷あり
＊40-1907
＊40-1917

MasatoKubotera, 2022-02-21:
>側面に傷あり

出力ホーン（ClampHorn）に傷ありということでしょうか．
SBシリーズなので今回は割とどうでもよいですが，ホーンやサーボ自体に傷（凹凸）があると嵌め合いを行う場合があり影響がある可能性があります．
そのため，できれば画像の貼付をお願いしたいです．
※最終的な報告に盛り込んでもらえれば大丈夫です．

Keita1Mori, 2022-02-21:
# 02/21進捗報告
## 担当者
@Keita1Mori @KentaTsukino @aiba0921 
## 作業内容
出力ホーンの状態確認
確認内容
- ネジ剤が使用されていたか
- 出力ホーンの傷の有無
- サーボ自体の傷の有無
- サーボ出力ホーンクランプ固定用ネジの確認
## 作業工程
画像のように穴にネジを挿入して確認してください。
[image]

サーボ出力ホーンクランプ固定用ネジの確認は以下の画像で左のサーボモータが純正品を使っています。
[image]

## 報告場所
[**こちらのスプレッドシート**](https://docs.google.com/spreadsheets/d/1VIxgXyjoNEmh7lO8G6sGWJy9Pbur3THwH7KlOdJ8XuE/edit?usp=sharing)からお願いします。

MibuchiYuta, 2022-02-22:
## 出力ホーンのクランプネジについて

純正品以外のネジが見つかっています。現在は70-2001系統（高須先輩の機体に使われていた）サーボモータで多く確認されています。以下に計測したデータを載せます。

長さ10mm
頭径3.65mm
頭高2mm
[image]<width=320>

Keita1Mori, 2022-02-22:
# サーボモータ出力ホーン穴の洗浄
## 必要な物
- +2番ドライバー　　　　　　　　×１
- 薄い－ドライバー　　　　　　　×２
- 面幅2.6の六角ドライバー　　　　×１
- 無水エタノール　　　　　　　　×１
- パーツクリーナ　　　　　　　　×１
- 歯間ブラシ(Seria)　　　　　　　×１
- 金属ブラシ(Viva home)　　　　　×１
- 管理していない旧機体のパーツ　×１
- 面幅2.6のネジ　　　　　　　　　×２

[image]

## 作業工程
[**動画**](https://drive.google.com/file/d/14LNF0FGPiad_Yj1uVNhhLI_o401wKmOf/view?usp=sharing)にしました。

Keita1Mori, 2022-02-22:
# サーボモータ出力軸報告
## 作業日
2022/02/21, 22
## 担当者
@Keita1Mori @KentaTsukino @aiba0921 
## 作業内容

B3Mの在庫登録されている以下の物の出力ホーンのネジ穴の状態確認と洗浄を行いました。
*40-　8個
40-　6個
70-　28個
計　52個

---
### 状態報告
状態確認に関しては[**こちら**](https://github.com/citbrains/OpenPlatform_ver3_0/issues/269#issuecomment-1047081563)に記載した通りの方法でやりました。
状態報告に関しては[**こちらのスプレッドシート**](https://docs.google.com/spreadsheets/d/1VIxgXyjoNEmh7lO8G6sGWJy9Pbur3THwH7KlOdJ8XuE/edit#gid=1643826419)にまとめました。

---

### 洗浄報告
洗浄方法は[**こちら**](https://github.com/citbrains/OpenPlatform_ver3_0/issues/269#issuecomment-1048057655)洗浄の有無も記載されています。
洗浄報告の有無は状態報告内のスプレッドシートに記載されています。

### まとめ報告
計52個のうち6個ネジ山死んでいました。型番はスプレッドシートで列が赤くなっている項目です。

## 所要時間
合計52個　5時間
状態確認　2時間
洗浄　3時間

## 次回の作業内容と担当者
ここのissueが三渕先輩が仕事を振るという認識なので三渕先輩の指示待ちになるかと思います。担当者もタスクが決まり次第一緒に報告をします。

MasatoKubotera, 2022-02-23:
>計52個のうち6個ネジ山死んでいました。型番はスプレッドシートで列が赤くなっている項目です。

Githubと実物報告の差分がありました．
スプレッドシートで関数を使うことを覚えた方が良いと思います．
[image]

hiroki-0001, 2022-02-23:
[220222_サーボ状態の確認](https://docs.google.com/spreadsheets/d/1VIxgXyjoNEmh7lO8G6sGWJy9Pbur3THwH7KlOdJ8XuE/edit#gid=133131651)
こちらを確認し、上の実物報告とGithubの差分。また出力ホーンが現状足りるかどうかを調査しました。

調査した結果、使用しているSC1040・SC1170の出力ホーンは10個不足している状況です。そこでSBのサーボモータから正常な出力ホーンを10個入れ替えることで対応します。

死んでる出力ホーンがついているサーボモータと出力ホーンを入れ替える用のSBのサーボモータは以下のような箱に入れておきます。箱自体は久保寺くんの机の上に置いておきます。

[image]
[image]

MiyawakiKaito, 2022-02-23:
# サーボモータの出力軸・フリーホーン軸洗浄

## 担当者
@MiyawakiKaito 

---
## 作業内容
前回洗浄したB3Mの洗浄で出力軸とフリーホーン軸の洗浄が抜けていたので追加で洗浄を行った

---
## 洗浄報告
下図の箱の中にあるサーボモータ以外はすべて洗浄を終えました。
[image]

hiroki-0001, 2022-02-24:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/269#issuecomment-1048663397の出力ホーンの入れ替えに関しては自分が行います。

hiroki-0001, 2022-02-24:
# 例外サーボの処理に関して

- 70-2006
- 70-1982

の2つに関しては、他の2つ例外サーボ(70-2005, *40-1918)からボトムケースを移植しました。
一応簡易的に動作確認を行いましたが特に問題はなさそうなので在庫登録しておきます。しかし、出力トルクと回転速度に関しては調査していないので、それに関する調査はhttps://github.com/citbrains/OpenPlatform_ver3_0/issues/228 で行う方針でお願いします。

- 70-2005
- *40-1918

に2つに関してはボトムケースや出力ホーンを止めるネジの２種類が不足しているため、修理によってボトムケースを補填してくれるかどうか確認が済むまで例外登録しておきます。

Keita1Mori, 2022-02-25:
# 再洗浄
## 担当者
@Keita1Mori 
## 作業内容
宮脇君が行った軸の洗浄を野口先輩と久保寺先輩がチェックした。しかし8個完了していないものが発覚したので再洗浄をしている。
- 70-1913
- 70-1915
- 70-1919
- 70-1922
- 70-1923
- 70-1924
- 70-1929
- 70-1930

hiroki-0001, 2022-02-25:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/269#issuecomment-1048663397 にて記述されている。出力ホーンの入れ替えを行いました。以下の通りに出力ホーンとクランプネジを入れ替えました。

- *40-1910 → 40-1949
- *40-1901 → 40-1922
- *40-1908 → 70-2001
- *40-1917 → 70-1993
- *40-1914 → 70-2002
- *40-1907 → 70-1991
- *40-1929 → 70-1916
- *40-1911 → 70-2110
- *40-1903 → 70-1939
- *40-1918 → 70-2006

入れ替えが終了したものは現時点では他の在庫サーボと分けてありますが、特に問題がなければ在庫サーボと同じケースに入れて管理します。しかしSBのモータに関しては今後使用しない方式をとっているため、別の箱に入れて管理していこうと考えています。

MibuchiYuta, 2022-02-28:
## サーボケースに関して

ケースに貼付するB3Mについてまとめたものは[このように](https://docs.google.com/presentation/d/1dNYn63iJMcLRtrZLj5xuEzmqulErvnwHLOnz9bEbQVM/edit?usp=sharing)作成しました。
実際に貼付した様子は以下のようになっています。
[image][image]

Keita1Mori, 2022-03-04:
# 報告(3/4)
サーボモータの出力ホーンの洗浄が完了しました。(3/4)#277 
## 次回の作業内容
出力ホーンの洗浄が終わったので出力軸の洗浄および出力軸の固定ネジの変更を行います。
## 質問
@MasatoKubotera 
出力軸の洗浄および固定ネジの変更の期限はいつまでですか？

hiroki-0001, 2022-03-09:
> 状態の良いサーボ19個(1セット分が最低)にboundrateとIDの書き込みをする(3/7〆切)

こちら170の脚部のサーボモータのみ状態のいいものを選定しました。以下の通りに変更しました。

- 1 (右足首ロール)
    -   40-1933 → 40-2204
- 2 (右足首ピッチ)
    -   70-1977 → 70-2105
- 4 (右足付け根ピッチ)
    -   70-1987 → 70-2010
- 5 (右足付け根ロール)
    -   70-1996 → 70-1991
- 6 (右足付け根ヨー)
    -   70-1979 → 70-1993
- 10 (左足首ロール)
    -   40-1956 → 40-2203
- 11 (左足首ピッチ)
    -   70-1981 → 70-2115
- 12 (左足膝ピッチ)
    -   70-1976 → 70-2102
- 13 (左足付け根ピッチ)
    -   70-1972 → 70-1916
- 14 (左足付け根ロール)
    -   70-1978 → 70-2006
- 15 (左足付け根ヨー)
    -   70-1980 → 70-2103

11個の状態のいいB3Mサーボモータを交換に使用しました

hiroki-0001, 2022-03-25:
https://github.com/citbrains/SUSTAINA-OP/issues/274 で選定された新たなクランプネジを現在の非純正・状態悪のクランプネジと交換します。

hiroki-0001, 2022-04-07:
>出力ホーンネジの挿入
ClampHorn軸の締結ネジを純正プラスねじから六角穴付フランジボルトの[620103008](https://jp.misumi-ec.com/vona2/detail/221000862966/?PNSearch=620103008&HissuCode=620103008)に変更している．170機についてはすでにヨー軸の3つのサーボを除き実装済み．現状の在庫数は203個．参照：https://github.com/citbrains/SUSTAINA-OP/issues/241

こちら、現在の在庫対象のB3Mは交換完了しました。
ネジ剤に関しては指標が出ていないので塗布していません。

> 非純正・なめった純正ClampHornのクランプねじを交換・赤ネジを塗布する．https://github.com/citbrains/SUSTAINA-OP/issues/274

こちらは在庫・故障対象のB3Mの交換完了しました。
こちらに関してもネジ剤は塗布していません。

hiroki-0001, 2022-04-11:
> 固定ネジ抜き
フレームなどと締結するための固定穴表裏4か所ずつがあるが，表裏ともに4か所のうち1か所にアッパーケースとボトムケースを固定するためにネジが締結されている．
組み立て時に，ネジを抜き忘れ組み立ててしまい分解，再度組み立てする事案が発生している．そのためにあらかじめ抜いておく．抜いた時にケース自体が分裂することは無いためネジを抜くことを問題視していない．
参照：https://github.com/citbrains/SUSTAINA-OP/issues/174#issuecomment-960564058

こちら在庫・故障のB3Mのサーボモータすべてに行いました。