SUSTAINA-OP/SUSTAINA-OP #271: 制御スイッチの再設計

MasatoKubotera, 2022-02-20:
オープンプラットフォーム仕様の制御スイッチの設計を行う．

ハードウェア・ソフトウェア両方の部分において実装が容易である持続可能な制御スイッチである必要がある．

---
# 現状
## バージョン
GankenKun Ver3.0から現在において制御スイッチは，「StartStopSwitch」を使用してきた．

タクトが2個搭載されたStartStopSwitchのバージョンは，3種類．
- [初代StartStopSwitch](https://github.com/citbrains/electronics/tree/master/StartStopSwtich/schematics_and_board
)
- StartStopSwitch ver2.0：2022/8/6作成　既製品USBケーブルが使用できるモデル
- [StartStopSwitch ver2.1](https://github.com/citbrains/OpenPlatform_ver3_0/tree/ver4_0/electronics/StartStopSwitch_ver2_1)：2022/10/16 作成　ver2.0のサイズ変更版

現状の機体では，RCAP前からすべてver2.0またはVer2.1に切り替わっている．

これらは，すべて Auvidea製 Carrier Board J120 のUSB OTG端子に対応したモデルである．

## 回路の仕様
スタートスイッチ・ストップスイッチそれぞれのピン状態を[ＰＩＣマイコン　ＰＩＣ１８Ｆ２５５０－Ｉ／ＳＯ（ＵＳＢ内蔵）](https://akizukidenshi.com/catalog/g/gI-02140/)で取得し送信している．

Jetson側では，ジョイスティックとして認識され，スタートスイッチは〇ボタン，ストップスイッチは×ボタン扱いになっている．

https://github.com/citbrains/citbrains_humanoid/blob/6c604a50fd18563a36f05ddf768773101b5010c9/for2050/src/HPL/startstopswitch.h#L17-L20

マイコンに書き込むプログラムはライセンス的問題で非公開(private repository)となっており，複雑で読みにくい．
https://github.com/citbrains/electronics/tree/master/StartStopSwtich

また，マイコンへの書き込みにはPICkit2が使われ専用ソフトウェアMPLAB X IDE v3.30も必要となり難易度が高い．

---
# 再設計の方針

1. 入手性が高い部品を使用すること
※半導体不足の近頃でも安定して供給されている部品などが好ましい．
2. 導入の難易度が低く，初心者でも扱えること．拡張性があること
※送信プログラム・書き込みの複雑さを低減する．
3. 安価であること

MasatoKubotera, 2022-02-20:
Arduino Pro Micro を使った制御スイッチ

PICマイコンは専用の書き込み機が別途必要である点やマイコンに書き込むプログラムの作成難易度が高いため，初心者でも扱いやすいArduinoを使った制御スイッチを作成したい．

ボードの候補は入手性が高く，互換品がAmazonでも大量に出回っているArduino Pro Microである．

キャリアボードとの接続は，USB接続である．
キャリアボード自体のGPIOを使用しても良いが，EN715のGPIOの汎用性が悪く，割り込み処理が実装可能であるかなどは確認できていないためである．
USBデバイスであれば，キャリアボードが変わったとしても汎用性が高いと考えている．

制御スイッチを現状と同様なUSBデバイスとしたとき，ジョイスティックとして認識させる必要は特になく，キーボード入力でも可能であるようだ．

そのため，Arudino公式が出しているkeyboardライブラリが使用できる． https://www.arduino.cc/reference/en/language/functions/usb/keyboard/

下記の記事が製作の参考になりそう．
[第63回 USBデバイスが作れるArduino？Arduino Pro Micro(Leonardo)でデバイス作成に挑戦！](https://deviceplus.jp/hobby/entry_063/)

また，ArudinoをHID(Human Interface Device)として扱えるライブラリも存在する．
https://github.com/NicoHood/HID

回路の基本構成としては，Arduinoとタクトスイッチを接続する容易なもの．
Aruidnoはプルアップがソフトウェア側から行えるため，プルアップ用抵抗を実装する必要が無い．
チャタリング防止のためにCR回路は実装しても良いかもしれない．
https://www.marutsu.co.jp/pc/static/large_order/1405_311_ph

MasatoKubotera, 2022-03-20:
Aruduino Pro Microとタクトスイッチ2個を使って容易な制御スイッチを作成した．

実験回路は下図に示す．

[image]

配線は，緑色タクトスイッチ（Start） -> D9，赤色タクトスイッチ（Stop） -> D10に接続し押下するとデジタルピンとGNDが短絡する簡易的なものでプルアップ抵抗などはArduinoプログラム/マイコン側で行えるため必要ない．

プログラムは下記のhttps://github.com/MHeironimus/ArduinoJoystickLibrary の[JoystickButton.ino](https://github.com/MHeironimus/ArduinoJoystickLibrary/blob/master/examples/JoystickButton/JoystickButton.ino)を用いた．

サンプルプログラムを特に変更せずに，Arduino Pro Microに書き込むことでJoystick（ゲームコントローラー）として認識される．

[image]

---

## windowsで動作確認

windowsで動作確認するためには，[Arduino Leonardo/Micro As Game Controller/Joystick](https://www.instructables.com/Arduino-LeonardoMicro-as-Game-ControllerJoystick/)を参考にした手順で確認できる．

1. <code>コントロール パネル\ハードウェアとサウンド\デバイスとプリンター</code>に「Arduino Leonardo」と表示・認識される．

[image]

2. 「Arduino Leonardo」を右クリックし，ゲームコントローラの設定を開く．
[image]

3. 「ゲームコントローラー」ウィンドウ中のArduino Leonardoをクリックし，「Arduino Leonardoのプロパティ」を表示させる．
[image]

4. タクトスイッチを押下してプロパティ上の「ボタン」項目が下記の画像のように変化することを確認する．
[image]
[image]

---

## Jetson TX2(Ubuntu 16.04)での動作確認

USB-Aポート下段に接続し<code>lsusb</code>すると下図のように<code>Bus 001 Device 004: ID 2341:8036 Arduino SA Leonardo (CDC ACM, HID)</code>と認識される．
[image]

[【Ubuntu】ゲームパットを使う](https://blog2.k05.biz/2021/05/ubuntu-gamepad.html)を参考にした手順で確認できる．

1. joystickの導入 下記の2つのパッケージをダウンロード
```
sudo apt-get install joystick
sudo apt-get install jstest-gtk
```
sudo apt-get install joystick　：ドライバー
sudo apt-get install jstest-gtk　：動作確認用のアプリjstest-gtk

2. アプリ 「jstest-gtk」を実行すると接続されたゲームパットが表示される．

[image]

3. 「Arduino LLC Aruduino Leonardo」をクリックする．

4. タクトスイッチを押下して，Bottomsが下記の画像のように変化することを確認する．

[image]「StopSwitch押下」
[image]「StartSwitch押下」

167機に搭載されたStartStopSwitchは下記のように認識された．
[image]「StopSwitch押下」
[image]「StartSwitch押下」

StartStopSwitchからの変更点
StopSwitch押下がbutton1->button1 で変更なし
StartSwitch押下がbutton2->button0 に変化

MasatoKubotera, 2022-03-20:
サンプルプログラムでは，制御スイッチとして使用できそうではありますが連打やそのほかの追加機能を使うには限界があるため，[ライセンス](https://github.com/MHeironimus/ArduinoJoystickLibrary/blob/master/LICENSE)に従って改変・再頒布を行いたいと思います．

AD58-3104, 2022-03-22:
作って頂いたsssをテストする簡単なプログラムを書きました.  
結果から言うと既存のコードそのままで動いています.完璧です.  
https://github.com/citbrains/citbrains_humanoid/tree/master/for2050/src/tools/joystick_test

MasatoKubotera, 2022-03-31:
上記案ではコックピットの横幅に合わないことが判明しましたので，Arduino Pro Microを表面実装コネクタを用いて実装するバージョンを作成しました．

表面実装コネクタは，[Preci-Dip DILソケット 2.54mm ピッチ 24 ピン](https://jp.rs-online.com/web/p/dil-sockets/7020597)を想定しています．

[image]
[image]
[image]

データについては下記の階層にあります．
https://github.com/citbrains/SUSTAINA-OP/electronics/StartStopSwitch_ver3_0

MasatoKubotera, 2022-04-03:
上記基板で使用するファームウェア（Arduino IDE コード）を作成しました．

外部割込みに対応したコードになっており，連打にも対応しています．（実検証はまだ行っていない）

ファームウェアについては，下記の階層にあります．
https://github.com/citbrains/SUSTAINA-OP/tree/sustaina_ver1_0_171/electronics/StartStopSwitch_ver3_0/firmware/StartStopSwitch

ファームウェア内で使用している，ライブラリについては https://github.com/citbrains/SUSTAINA-OP/issues/271#issuecomment-1073223967 で使用したものであり，ライセンスに従い著作権表示を保ち，同ライセンスで再頒布，ひとまず改変せずに再頒布しています．

ライブラリについて使用していない部分が多いですが，創作性を残すためにあえて無改変にしています．

```.cpp
#include "src/Joystick.h"

Joystick_ Joystick;

void setup() {
  // Initialize Button Pins
  pinMode(9, INPUT_PULLUP);
  pinMode(10, INPUT_PULLUP);

  // Initialize Joystick Library
  Joystick.begin();

  // Initialize external interrupts
  attachInterrupt(2, button0_press, FALLING);
  attachInterrupt(3, button1_press, FALLING);
  attachInterrupt(2, button0_release, RISING);
  attachInterrupt(3, button1_release, RISING);
}

void loop() {
}

void button0_press(){
  Joystick.setButton(0, true);
}

void button1_press(){
  Joystick.setButton(1, true);
}

void button0_release(){
  Joystick.setButton(0, false);
}

void button1_release(){
  Joystick.setButton(1, false);
}
```

MasatoKubotera, 2022-04-03:
キャリアボードEN715との接続は，[USBMCH-LLL20【変換名人製】](https://www.amazon.co.jp/gp/product/B01G1ECBLO/ref=ppx_yo_dt_b_asin_title_o09_s00?ie=UTF8&amp&psc=1)を使って接続することを想定していますが，キャリアボード側のUSB Micro B端子が使えない場合には，USB A - USB Micro Bケーブルで接続します．

MasatoKubotera, 2022-04-12:
2022/4/10 Elecrowに基板を発注しました．
発注データについては，https://github.com/citbrains/SUSTAINA-OP/issues/271#issuecomment-1084724563 で示しているデータ保管の階層にあります．

StartStopSwitch Ver3.0 での使用部品については下記の通りです．

- [DEV-12640](https://www.mouser.jp/ProductDetail/SparkFun/DEV-12640?qs=WyAARYrbSnb2ipr1B6qalg%3D%3D) x1個
- [細ピンヘッダ　１×４０　（黒）](https://akizukidenshi.com/catalog/g/gC-06631/) x1個　※40Pの内24P使用
- [JF15SP1C](https://www.mouser.jp/ProductDetail/NKK-Switches/JF15SP1C?qs=/ha2pyFadug6XMB03BkuswUPYFG07/KtB9uj2vYZzN2ozyWFBgrvgQ==) x1
- [JF15SP1F](https://www.mouser.jp/ProductDetail/NKK-Switches/JF15SP1F?qs=/ha2pyFadug6XMB03Bkus0Mu9c37rOrIdqRyNSNGAfUzUG04tLkntA==) x1

MasatoKubotera, 2022-04-22:
https://github.com/citbrains/SUSTAINA-OP/issues/271#issuecomment-1086904908 に記載したプログラムについて，Arduinoを用いて検証したが，動作しなかった．

原因としては，外部割込みを同一ピンで複数使えないことがあった．
http://www.musashinodenpa.com/arduino/ref/index.php?f=0&pos=3069v

そのため，連打に対応するために押した状態を外部割込みで感知し，離した状態は通常のdigitalReadで感知する方式に変更した．

変更後のプログラムは下記に記す．
```.cpp
#include "src/Joystick.h"

Joystick_ Joystick;

void setup() {
  // Initialize Button Pins
  pinMode(0, INPUT_PULLUP);
  pinMode(1, INPUT_PULLUP);

  // Initialize Joystick Library
  Joystick.begin();

  // Initialize external interrupts
  attachInterrupt(2, button0_press, FALLING);
  attachInterrupt(3, button1_press, FALLING);
}

void loop() {
}

void button0_press(){
  Joystick.setButton(0, true);
  while (digitalRead(0)==LOW);
  Joystick.setButton(0, false);
}

void button1_press(){
  Joystick.setButton(1, true);
  while (digitalRead(1)==LOW);
  Joystick.setButton(1, false);
}

```

MasatoKubotera, 2022-04-22:
elecrowに発注したときの詳細を下記に記します．

Layer : 2 layers
Dimensions : 58 x 22.5
PCB Qty : 10
Different PCB Design : 1eg
PCB Thickness : 1.6
PCB Color : Black
Surface Finish : OSP
Castellated Hole : No
Copper Weight : 1oz
PCB Stencil : No

[image]

Surface Finish（基板の表面処理）をOSPにしているのは発注ミスです．
本来は，HASL Lead Freeの予定でした．

MasatoKubotera, 2022-04-22:
下図は届いた基板になります．

表面処理がOSPであるため，フットプリントが銅色にになっています．

[image]

MasatoKubotera, 2022-04-22:
https://github.com/citbrains/SUSTAINA-OP/issues/271#issuecomment-1096867625 で示した部品を実装しました．

表面実装部品が含まれますが，裏面にスイッチがあり行えためすべて手はんだで行います．

[image]
[image]

MasatoKubotera, 2022-04-22:
同時押しにも対応するファームウェアを作成しました．

作成したファームウェアは下記の階層にあります．
https://github.com/citbrains/SUSTAINA-OP/blob/master/electronics/StartStopSwitch_ver3_0/firmware/StartStopSwitch/StartStopSwitch.ino

```.cpp
#include "src/Joystick.h"

Joystick_ Joystick;

/*
 * The correspondence between physical buttons and output buttons is as follows.
 * button0 : Start Button
 * button1 : Stop Button
 */

// button state 0: release ,1:press
volatile int button0_state = 0;
volatile int button1_state = 0;

void setup() {
  // Initialize Button Pins
  pinMode(0, INPUT_PULLUP); // Start Button
  pinMode(1, INPUT_PULLUP); // Stop Button

  // Initialize Joystick Library
  Joystick.begin();

  // Initialize external interrupts
  attachInterrupt(2, button0_press, FALLING); // pin num 0 -> int.2
  attachInterrupt(3, button1_press, FALLING); // pin num 1 -> int.3
  /*
   * Refer to the following URL for attachInterrupt.
   * https://www.arduino.cc/en/Reference/AttachInterrupt
   */
}

void loop() {
  if(button0_state == 1 && digitalRead(0) == HIGH){
    Joystick.setButton(0, false);
    button0_state = 0;
  }
      
      
  if(button1_state == 1 && digitalRead(1) == HIGH){
    Joystick.setButton(1, false);
    button1_state = 0;
  }     
}

void button0_press(){
  Joystick.setButton(0, true);
  button0_state = 1;
}

void button1_press(){
  Joystick.setButton(1, true);
  button1_state = 1;
}

```

MasatoKubotera, 2022-04-22:
Jetsonを使わない簡易的な動作確認はhttps://github.com/citbrains/SUSTAINA-OP/issues/271#issuecomment-1073223967 を参照してください．

また，ジョイスティックライブラリの仕様上，コマンドが送信されたとき（ボタンを押した瞬間・離した瞬間）Arudino pro micro に実装されたD13ピンのLEDが点灯するようです．

MasatoKubotera, 2022-04-23:
上記で提示していたファームウェアですが，出力がStart・Stopで逆でした．

Startを押したときにbutton0で出力，Stopを押したときにbutton1で出力するように変更しました．

このようにしているのは，BrainsではこのSwitchがStartStopSwitchと呼ばれているため，Start：button0，Stop：button1にすると覚えやすいというのがあります．

訂正したプログラムは下記の通りです．GitHubにも反映しています．

```.cpp
#include "src/Joystick.h"

Joystick_ Joystick;

/*
 * The correspondence between physical buttons and output buttons is as follows.
 * button0 : Start Button / pin num : 1 / int.3
 * button1 : Stop Button  / pin num : 0 / int.2
 */

// button state 0: release ,1:press
volatile int button0_state = 0;
volatile int button1_state = 0;

void setup() {
  // Initialize Button Pins
  pinMode(0, INPUT_PULLUP); // Start Button
  pinMode(1, INPUT_PULLUP); // Stop Button

  // Initialize Joystick Library
  Joystick.begin();

  // Initialize external interrupts
  attachInterrupt(3, button0_press, FALLING); // pin num 1 -> int.3
  attachInterrupt(2, button1_press, FALLING); // pin num 0 -> int.2
  /*
   * Refer to the following URL for attachInterrupt.
   * https://www.arduino.cc/en/Reference/AttachInterrupt
   */
}

void loop() {
  if(button0_state == 1 && digitalRead(1) == HIGH){
    Joystick.setButton(0, false);
    button0_state = 0;
  }
      
      
  if(button1_state == 1 && digitalRead(0) == HIGH){
    Joystick.setButton(1, false);
    button1_state = 0;
  }     
}

void button0_press(){
  Joystick.setButton(0, true);
  button0_state = 1;
}

void button1_press(){
  Joystick.setButton(1, true);
  button1_state = 1;
}
```

出力を間違えていた要因として，Arduino pro micro のピン配置が１，０，２という順番で紛らわしいことがありました．

[image]

MasatoKubotera, 2022-04-23:
Jetson Xavier NX + CarrierBoard EN715の開発において，CarrierBoardのmicro usb type b とArduino pro microを接続することを想定していましたが認識しなかったようです．

usb type aでは動作確認ができているようですので，こちらの方法で接続していくこと前提に進めたいと思います．

https://github.com/citbrains/citbrains_humanoid/issues/1735#issuecomment-1107420145
<blockquote>
startstopswitchについても動作確認をしました．
ただし本来であればswitch側(arduino)micro-b，en715側もmicro-bで接続を行うはずが，en715側のmicro-bが使えず，switch側(arduino)がmicro-bでen715側はusb-Aでの接続となっています．
</blockquote>

MasatoKubotera, 2022-04-23:
下記のマニュアルにOTG/USB Micro Type Connectorはrecovery専用のようでした．

https://storage.avermedia.com/web_release_www/EN715/AVerAI_UM_EN715_NO111B_NX211B_TN111B.pdf

MasatoKubotera, 2022-05-06:
176機分までを量産しました．
製作担当は下記の通りです．
- 171：@MasatoKubotera
- 172：@MasatoKubotera
- 173：@TakuyaNagamine
- 174：@sekitorimisaki
- 175：@Keita1Mori 
- 176：@KentaTsukino

参考程度ですが，実装時の様子を動画に残しました．
回路の簡単な解説を含めて1時間程度かかりました．

慣れている人であれば製作に5分もかからないかと思います．

@TakuyaNagamine が引継ぎ用の[ドキュメント](https://docs.google.com/document/d/1LLo_39rDERS-zUw8RerwoWCL0OPBO0tpu2KG_FWQNdQ/edit?usp=sharing)を作成してくれています．

こちらの詳細は，[wiki](https://github.com/citbrains/SUSTAINA-OP/wiki/StartStopSwitch)にも記載されています．

MasatoKubotera, 2022-05-06:
上記の量産分は，https://github.com/citbrains/SUSTAINA-OP/issues/271#issuecomment-1073223967 の方法に従って動作確認を行いました．
Jetson 本体での動作確認はしていません．

MasatoKubotera, 2022-05-27:
https://github.com/citbrains/SUSTAINA-OP/issues/271#issuecomment-1107431595 上記で提示したファームウェアだが，GankenKun用のStartStopSwitch Ver2.1以下のバージョンとの互換性がない．

これは，ボタンを押したときにVer3.0（Arduino仕様）ではStart→button0，Stop→button1として信号を送っていたが，Ver2.1以下ではStart→button1，Stop→button2の信号であるためである．

この別々な仕様としたときに，コンピュータボード（Jetson）側のソフトウェアのでSUSTAINAとGankenKunで分岐させる必要があるため，一旦GankenKunと互換性がある信号を送るようにする．

そのためにファームウェアを下記のように変更する．
```.cpp
#include "src/Joystick.h"

Joystick_ Joystick;

/*
 * The correspondence between physical buttons and output buttons is as follows.
 * button1 : Stop Button  / pin num : 0 / int.2
 * button2 : Start Button / pin num : 1 / int.3
 */

// button state 0: release ,1:press
volatile int button1_state = 0;
volatile int button2_state = 0;

void setup() {
  // Initialize Button Pins
  pinMode(0, INPUT_PULLUP); // Stop Button
  pinMode(1, INPUT_PULLUP); // Start Button

  // Initialize Joystick Library
  Joystick.begin();

  // Initialize external interrupts
  attachInterrupt(2, button1_press, FALLING); // pin num 0 -> int.2 // Stop
  attachInterrupt(3, button2_press, FALLING); // pin num 1 -> int.3 // Start
  /*
   * Refer to the following URL for attachInterrupt.
   * https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/
   */
}

void loop() {
  if(button1_state == 1 && digitalRead(0) == HIGH){
    Joystick.setButton(1, false);
    button1_state = 0;
  }     
  
  if(button2_state == 1 && digitalRead(1) == HIGH){
    Joystick.setButton(2, false);
    button2_state = 0;
  }
}

void button1_press(){
  Joystick.setButton(1, true);
  button1_state = 1;
}

void button2_press(){
  Joystick.setButton(2, true);
  button2_state = 1;
}
```

MasatoKubotera, 2022-06-06:
177機用，予備用の計2つを製作し，windowsでの動作確認まで行いました．

[image]

また，171~174機用のStartStopSwitchのファームフェアをGankenKun互換に変更し実機に搭載しています．
現状問題なく動作しているようです．

今後行うタスクリストを下記に記載します．

- 175機用
  - [ ] ファームウェア書き換え
  - [ ] 実ロボットを使った動作確認

- 176機用
  - [ ] ファームウェア書き換え
  - [ ] 実ロボットを使った動作確認

- 177機用
  - [ ] 番号テプラ貼付 **No.7**
  - [ ] 実ロボットを使った動作確認

- 予備用(No.8)
  - [ ] 番号テプラ貼付 **No.8**
  - [ ] 実ロボットを使った動作確認

メモ：テプラ貼付後の写真を報告していないのでコメントする．

MasatoKubotera, 2022-06-06:
GankenKun互換バージョンに対応した，動作確認方法など最新情報は下記に記載しました．

https://github.com/citbrains/SUSTAINA-OP/wiki/StartStopSwitch

MasatoKubotera, 2022-06-06:
実装の時の動画をYouTubeにアップしました．

https://youtu.be/ntGQOd9pe1A

MasatoKubotera, 2022-06-08:
番号テプラについては下図のように貼付しています．
[image]

MasatoKubotera, 2022-06-08:
製作者・製作日をまとめておきます．

- No.1 2022/4/26 @MasatoKubotera   -> 171機体用
- No.2 2022/5/5 @MasatoKubotera  -> 172機体用
- No.3 2022/5/5 @TakuyaNagamine  -> 173機体用
- No.4 2022/5/5 @sekitorimisaki  -> 174機体用
- No.5 2022/5/5 @Keita1Mori  -> 175機体用
- No.6 2022/5/5 @KentaTsukino   -> 176機体用
- No.7 2022/6/6 @MasatoKubotera  -> 177機体用
- No.8 2022/6/6 @MasatoKubotera  -> RoboCup2022 Thailand予備

closed 2022-06-27 by MasatoKubotera