SUSTAINA-OP/SUSTAINA-OP #273: CAMテンプレートの修正と追加

MasatoKubotera, 2022-02-24:
- [ ] φ4エンドミルのパラメータ・設定の追加

- [ ] [XALシリーズ超硬スクエアエンドミル 4枚刃/刃長4D（ロング）タイプ (XAL-PEM4L2)](https://jp.misumi-ec.com/vona2/detail/110600204820/?HissuCode=XAL-PEM4L2)用のパラメータの追加
詳細：https://github.com/citbrains/OpenPlatform_ver3_0/issues/101#issuecomment-1024753849

- [ ] PPクラフトシート用のパラメータ，テンプレートアセンブリの作成
詳細：https://github.com/citbrains/OpenPlatform_ver3_0/issues/265

反映はブランチ ver4_1_prototype にお願いします．

MasatoKubotera, 2022-02-27:
@NaokiIwasawa 
ここに各種パラメータもしくは検証などで使用したCAMファイル類をGoogleドライブにアップしたリンクを貼ってください．
無い場合には，変えた個所の記載でも構いません．

NaokiIwasawa, 2022-02-27:
# 変更点まとめ　@MasatoKubotera 
@NaokiIwasawa が検証などを踏まえて次回のテンプレートに反映して欲しいことは以下である

1. アルミのみ4枚刃の追加
変更するのはアルミのテンプレートの **刃数**， **切削送り速度**で
2→4枚刃
300→600mm/min

2. **切削送り速度**を2種類にする
切削送り速度が600mm/minだと穴精度が出ないため，輪郭以外は300mm/minのままとする

### アルミ(4枚刃)
- ポケット　600mm/min
- ポケット切り抜き　300mm/min
- 輪郭　600mm/min
- 輪郭・穴用　300mm/min

### POM(2枚刃)
- ポケット　600mm/min
- ポケット切り抜き・穴用　300mm/min
- 輪郭　600mm/min
- 輪郭・穴用　300mm/min

### GFRP(2枚刃)
- ポケット　600mm/min
- ポケット切り抜き・穴用　300mm/min
- 輪郭　600mm/min
- 輪郭・穴用　300mm/min

### PPクラフトシート(2枚刃)
- ポケット切り抜き・穴用　250mm/min
- 輪郭　250mm/min
主軸回転速度　8000 rpm

NaokiIwasawa, 2022-03-31:
# エンドミル径
現在主にφ2を使っているがそれ以外の径について

## φ4
以前コックピットの切削時に時間短縮の目的でポケットのみφ4を用いようとしていたが，結果設定でミスを犯し失敗してしまった．そのため扱いに慣れていない・ミスが許容できない環境では使用を推奨できず，テンプレートには追加しない．

## φ1.5
コックピットの一部(電圧計の取り付け部など)でM2用の下穴φ1.6が必要なためテンプレートにも導入が必要だと考える．

yasuohayashibara, 2022-03-31:
> 設定でミスを犯し失敗してしまった．

ご存知のように，マシニングセンターは，エンドミルを自動的に変更できます．
利用できるようになるとスピードがぐっと上がりますが，利用方法にまず慣れることが必要になりますので，SUSTAINAの切削が一通り終わったら挑戦したいところですね．

MasatoKubotera, 2022-08-23:
テンプレートについて実際使用している方が少ないため，整備する必要性が薄いためCloseします．
テンプレートよりも，実際に使われたCAMを再加工することの方がテンプレートを使用するよりも早いようです．