SUSTAINA-OP/SUSTAINA-OP #278: サーボモータのマーキング整備

NaokiIwasawa, 2022-02-26:
マーキングの方法が確立されているため，B3M本体にマーキングを行う．
→現状のマーキング方法がissueにしか存在していなく扱いにくいため，新規パーツケースに概要を印刷し貼付してください．
参照：https://github.com/citbrains/OpenPlatform_ver3_0/issues/174#issuecomment-940862575

終わったら以下のissueに反映すること
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/269#issue-1143097732

hiroki-0001, 2022-04-30:
B3Mに添付されているラベルの各項目ごとのマーキングする条件を定めました。
以下の通りです。

● ... サーボの初期位置が変更されている  
　→ B3Mのパラメータ 『Center Position』(初期位置)の値が0以外であればマーキングする。

★ ... 出力軸のギアが欠けている  
　→ B3M管理シートの『型番による抽出』の「ギア欠けの状態」が**限定使用可** or **ガタ異常**になっている場合はマーキングする。

▲ ... 故障している  
　→ 故障判定を受けたもの。また「ギア欠けの状態」が**ガタ異常**になって場合マーキングする

■ ... その他  
　→ 特筆すべき注意点がある場合マーキングする。(詳細はB3M管理シートの備考に記載)

hiroki-0001, 2022-04-30:
このマーキング処理は現在在庫管理されているB3Mで行われていないものも多数あるため、上の条件にしたがってマーキングする必要があります。

hiroki-0001, 2022-05-24:
マーキングの情報に関して今までissueにしかなかったのでwikiに記載しておきました。

https://github.com/citbrains/citbrains_humanoid/wiki/B3M%E3%80%80%E7%AE%A1%E7%90%86%E3%82%B7%E3%82%B9%E3%83%86%E3%83%A0

closed 2022-06-13 by MasatoKubotera