SUSTAINA-OP/SUSTAINA-OP #279: サーボモータを新規パーツケースへ移行する

NaokiIwasawa, 2022-02-26:
新規パーツケースには管理番号，管理システム・回答フォームへのQRコードの貼付をお願いします．

関連issue
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/278

MibuchiYuta, 2022-03-01:
## サーボケースに関して

ケースに貼付するB3Mについてまとめたものは[このように](https://docs.google.com/presentation/d/1dNYn63iJMcLRtrZLj5xuEzmqulErvnwHLOnz9bEbQVM/edit?usp=sharing)作成しました。
実際に貼付した様子は以下のようになっています。
[image][image]

MibuchiYuta, 2022-03-01:
緩衝材は以下のサイズ(mm)に切り取り、貼り付けるようにします。
- A：45×110
- B：50×160
両面テープで貼り付けると画像のようになりました
[image][image]

MibuchiYuta, 2022-04-02:
> 緩衝材は以下のサイズ(mm)に切り取り、貼り付けるようにします。
こちらの緩衝材は @matsuyamayusaku @ShunHoshina が４箱全てに両面テープで貼り付け、その後 @MibuchiYuta が確認し問題ありませんでした。

MibuchiYuta, 2022-04-02:
@NaokiIwasawa @MasatoKubotera 箱に管理番号を貼付することは可能ですが行いますか?

MasatoKubotera, 2022-04-02:
@MibuchiYuta 
同じ箱が４箱あり，ケース上面も説明紙でふさがれており内部の様子が見えないため，側面などに（ケースの）管理番号と簡単に内容物（B3M-SC-1040-A，B3M-SC-1170-A，変換基板など）を書いておいていただけると嬉しいです．

変換基板やUSBケーブル，検証・執権用サーボケーブルなどをこの箱に入れて保管するときは，内容物をA4紙に記載しなくならないように管理できればと思います．

MibuchiYuta, 2022-04-04:
@MasatoKubotera 
@hiroki-0001 に相談した結果、大会時などはわかりませんが通常は別の箱で変換基板やケーブルを保管することにしました。 現在は仮の箱に入れてあります。　専用の箱を購入し次第、保管用の紙を作成し箱に貼付します!<imp src ="https://user-images.githubusercontent.com/53966307/161463873-e351264a-91a6-4e84-b1ef-9a2ac031723c.jpeg" width="320">[image]

MasatoKubotera, 2022-04-05:
@MibuchiYuta 

>専用の箱を購入し次第

予算での購入予定はありません．
量産が始まるとサーボの在庫個数が少なくなっていき，現状の箱が余ると思うのでそちらを利用することも視野に入れてください．

MibuchiYuta, 2022-04-06:
こちら私物の箱で対応しました。
内容物を示す紙はまだ用意していません。（というかそもそも内容物が確定していない気もします）

MibuchiYuta, 2022-04-14:
[貼付する紙](https://docs.google.com/presentation/d/1dNYn63iJMcLRtrZLj5xuEzmqulErvnwHLOnz9bEbQVM/edit?usp=sharing)を作成しました。個数を記入しておきたいのですが、不明です。 @hiroki-0001 教えていただけますか?
[image]

MasatoKubotera, 2022-04-14:
@MibuchiYuta 口頭でお伝えしたとおり、サーボケーブルについて2種類ありますのでご注意ください。

4pサーボケーブル
4pの通信のみサーボケーブル

hiroki-0001, 2022-04-14:
@MibuchiYuta 

紙に明記する各備品の個数です

- RS485USBシリアル変換アダプター
    - 黒(自作) × 2
    - 緑(純正) × 3

- USB A - Mini Bケーブル
    - 2個(内1つは4股の検証に使用中)

- B3Mサーボケーブル
    - 新品ケーブル × 2
    - 信号線のみ × 2

- バナナプラグ-XT60 変換ケーブル
    - 1個

- バナナプラグ-VH 変換ケーブル
    - 1個

- XT60-VH 変換ケーブル
    - 2個

MibuchiYuta, 2022-04-15:
[image]
optionパーツのボックスが完成しました。
特に問題がなければcloseします

closed 2022-04-26 by MasatoKubotera

hiroki-0001, 2022-04-30:
https://github.com/citbrains/SUSTAINA-OP/issues/279#issuecomment-1054982133
これの表記をよく見ると、★と■の内容が被っていたので修正を加えました。

[image] 

[image] 

これを新たに箱に貼り直しておきます