SUSTAINA-OP/SUSTAINA-OP #280: 歩行速度の比較検証

MasatoKubotera, 2022-02-27:
Ver3.0 と prototypeの歩行速度 比較検証結果をまとめる

MasatoKubotera, 2022-02-27:
歩行検証用データフォルダ
https://drive.google.com/drive/folders/1-J2dtfhUEl7F-RSIsEC7oBeD6TFKgcd7?usp=sharing

hiroki-0001, 2022-02-28:
## <検証方法について>

新機体(Ver4.1)prototypeとVer3.0でほぼ同一の歩行パラメータで3m歩行させて、歩行速度や歩行の状態などで、どの程度変化があるかを調査する。また検証には新機体(Ver4.1)prototypeには170機体。Ver3.0には161機体を使用した。

### 検証時の歩行のパラメータに関して
歩行に関するパラメータ(eeprom.txt)を以下のパラメータ以外を統一した。

- 重心高さ(xp_mv_walk.h_cog(*))
  - 161 = 300.0
  - 170 = 320.0

また歩行時のstrideやperiodに関しては以下のように設定した。

1回目の計測  
stride X = 19  
period = 10

2回目の計測  
stride X = 22  
period = 10

3回目の計測  
stride X = 24  
period = 10

4回目の計測  
stride X = 25  
period = 10

5回目の計測  
stride X = 26  
period = 10

### 結果

[歩行時の様子の動画集](https://drive.google.com/drive/folders/1-J2dtfhUEl7F-RSIsEC7oBeD6TFKgcd7?usp=sharing)

[各strideごとの歩行速度の比較](https://docs.google.com/spreadsheets/d/1CLpALmZDkvbexsiF2rR6F_bkFhRo09jVxWtreC6JLCs/edit?usp=sharing)

新機体(Ver4.1)prototypeとVer3.0の歩行を比較すると新機体(Ver4.1)prototypeの歩行速度は同等またはそれ以上の歩行速度を発揮する事がわかった。

しかし、新機体(Ver4.1)prototypeはstride X が25を超えたあたりから、歩行終了時に過電流エラーと思われるエラーが発生してしまう。これは推測だが歩行の停止時にかかるブレーキによる過電流が原因だと思われる。

closed 2022-03-13 by MasatoKubotera