SUSTAINA-OP/SUSTAINA-OP #281: サーボモータ出力軸ホーンの状態確認

NaokiIwasawa, 2022-02-27:
雌ネジがつぶれていて挿入できない箇所がある．Ver3.0までは90度間隔でネジを挿入していたが，Ver4.0では腕などでネジの挿入間隔が異なるため，すべての雌ネジが通常にネジが挿入できる必要がある．

終わったら以下のissueに反映すること
https://github.com/citbrains/OpenPlatform_ver3_0/issues/269#issue-1143097732

aiba0921, 2022-02-28:
70-2009にクランプホーンが挿入されていなかったため*40-1911から純正のクランプホーンを移植しました。
↓移植前
[image]

aiba0921, 2022-02-28:
70-2009にクランプホーンが挿入されていなかったため*40-1910から移植しました。
[image]

KentaTsukino, 2022-03-05:
> 歪みというものがどいうものかどうか詳細に教えていただけると嬉しいです．

この70-1922は特に歪みが酷かったものです
[image][image]
[image][image]
[image][image]

ここから下は上の写真から出力ホーンを180°回転させたものです
<img src = https://user-images.githubusercontent.com/83112596/156693414-5cf45c72-083e-46d4-bf4f-8061067706a7.jpg
width = 150 >[image]

[image][image]

[image][image]

yasuohayashibara, 2022-03-05:
ちょっと話題がずれるかもしれませんが，どこの関節で使用されていたかのログがあると良いですね．
特定の関節のサーボモータに不具合が発生する場合は対策したほうが良いかもしれません．

hiroki-0001, 2022-03-05:
一応サーボの管理システムを作成した時期からにはなりますが、各サーボごとに振り分けられている管理番号を元に今までの使用履歴は参照する事ができます。

[image]

70-1922の場合は過去に163の右膝ピッチに使用されていた形跡があります。

MasatoKubotera, 2022-03-05:
<blockquote>
この70-1922は特に歪みが酷かったものです
[image][image]
[image][image]
[image][image]

ここから下は上の写真から出力ホーンを180°回転させたものです
<img src = https://user-images.githubusercontent.com/83112596/156693414-5cf45c72-083e-46d4-bf4f-8061067706a7.jpg
width = 150 >[image]

[image][image]

[image][image]

</blockquote>

@KentaTsukino 
これはサーボ内部や本体のガタ、消耗によるものでしょうか。

この状態を見る限りは、クランプホーンが上手く装着出来てない恐れもあります。
この他のクランプホーンにおいても、適切なトルク・ネジで軸ネジ、クランプネジを締め付けて"歪み"が発生するようであれば修理対象に入れた方が良いと思います。

B3M-SC-1170を使う場所においてこの歪み（70-1922）を許容して使える箇所はありません。

@hiroki-0001 
163機の脚部に使われていたとの事で、当該のサーボモータはRCAPの展示用機体でしたので、試合や調整中に生じた問題と断定出来きないのが辛いですね。

hiroki-0001, 2022-04-01:
https://github.com/citbrains/SUSTAINA-OP/issues/281#issuecomment-1059691199 で議論されていたサーボホーンの歪みですが、出力ホーンを付け直し、適正トルクでネジを締め直したところ、歪みはなくなりました。
現状のB3Mの在庫登録されているものは、すべて確認済みですが現状のところ画像のような歪みは見られていません。

hiroki-0001, 2022-04-01:
またこのissueの目的である。

> 雌ネジがつぶれていて挿入できない箇所がある．Ver3.0までは90度間隔でネジを挿入していたが，Ver4.0では腕などでネジの挿入間隔が異なるため，すべての雌ネジが通常にネジが挿入できる必要がある．

に関しては[新B3M管理シートの状態確認リスト](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit?usp=sharing)を確認したところ、現状出力ホーンを交換しなければ対処できないB3Mは在庫・故障合わせて4つありました。

- 70-1978
- 70-1981
- 70-1988
- 70-1996

上記のB3Mに関しては再度状態を確認し、問題があれば出力ホーンごと交換してしまったほうが良さそうです。

KentaTsukino, 2022-04-05:
4/2
野口先輩が故障しているサーボは調査していないとの事でしたので、調査しました
赤の丸をした所がネジ山がつぶれていた所です
この二つ以外は大丈夫でした
[image][image]
[image][image]

hiroki-0001, 2022-04-05:
> に関しては[新B3M管理シートの状態確認リスト](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit?usp=sharing)を確認したところ、現状出力ホーンを交換しなければ対処できないB3Mは在庫・故障合わせて4つありました。
70-1978
70-1981
70-1988
70-1996
上記のB3Mに関しては再度状態を確認し、問題があれば出力ホーンごと交換してしまったほうが良さそうです。

こちらの状態の確認を行いました。

70-1978

[image] 

赤丸部分のネジ山が死んでる。要交換

70-1981

[image] 

再度洗浄を行なった。対象のネジ穴は挿入可能になったので交換はしない

70-1988

[image] 

内側のギアが摩耗している。要交換

70-1996

[image] 

再度洗浄を行なった。対象のネジ穴は挿入可能になったので交換はしない

---

これらの調査結果から、

- 70-1978
- 70-1988

の2つはclampHornを交換します。

hiroki-0001, 2022-04-05:
上のものに加え月野くんの調査してくれたものも含めて

- 70-1978
- 70-1988
- 40-1945

これらのB3Mの出力ホーンを交換します。

hiroki-0001, 2022-04-05:
40-1945ですが一応clampHornを取り出して、6つのネジ穴が挿入できるかどうか確認したところ、挿入できたので、とりあえず交換しないでおきます。

hiroki-0001, 2022-04-07:
70-1919のclampHornを固定するM2のネジのネジ穴が潰れすぎて、clampHornから取り外せなくなったので、clampHornごと交換しました。

hiroki-0001, 2022-04-07:
在庫・故障対象のB3Mに関しての出力軸ホーン(clamp horn)の状態確認・対応は完了したので、closeします。

closed 2022-04-07 by hiroki-0001