SUSTAINA-OP/SUSTAINA-OP #282: 模擬試合にサーボモータが壊れる原因の調査

RikuYokoo, 2022-03-03:
161と170で模擬試合終了後にサーボモータの故障が原因でhr時に末端ではないサーボに力が入らなくなるという現象が起きた(力が入らないサーボの先のサーボには力は入っていた)．
これは以前にも起きたらしくその時はサーボ内部の基板が壊れていたことが原因であった．
模擬試合中に何度か転倒をしていたのでそれが故障の原因の1つとして考えられるがしっかりとした原因が不明であるのでその調査を行う．

RikuYokoo, 2022-03-03:
この現象が起きたのが2022年3月2日である．
その模擬試合の様子はこちら([俯瞰図](https://drive.google.com/drive/u/0/folders/1-oFUIOPvMK-REmEHySt7om78EXoIY7NP), [170の個別図]())

MasatoKubotera, 2022-03-03:
該当のサーボモータ

170機
右膝下ピッチ (ID 3)
管理番号：70-1988

使用履歴を確認しましたが，以前の搭載機は管理システムができる以前のため記録がなく，下記の要項で故障登録が行われていました．

故障日：2020/12/22
故障状況：メンテナンス
故障内容：右回転は新品より軽く，左回転はほぼ回らない，Hブリッチ回路の片方の破損？,通信自体は可能

[メーカ修理記録](https://docs.google.com/document/d/1OC_CzZlHnqysJ16qTEJYPaVCg-WHtrKTncKrh0wkYuQ/edit#heading=h.k5k7mmf47i8u)を参照したところ，FET故障と診断され2021年9月に修理されていました．

MasatoKubotera, 2022-03-03:
ハードウェア側に問題があるとして考察します．

該当のサーボモータは，ケーブルの配線的に末端に位置しないためケーブルの断線とは考えにくい状況です．
>[image]

直立状態時に信号線（Ｄ+，Ｄ－）が切断されたとしても，直前の指令で固まるため脱力状態になることはありません．
よって，電源線（ＶＣＣ，ＧＮＤ）が瞬断・切断状態であると考えられます．
電源線が断裂すると，サーボモータ自身がリセットされ，コマンド受け取り，動作するまで脱力状態となります．

この時に，サーボモータ本体に実装されたLEDが点灯・点滅していたとしても断裂箇所によってはモータが動作しないことがあります．また，FETなどモータに電力を加えるあたりやブラシなど内臓モータ本体の故障も考えられます．
※B3M-SCはコアレスモータであり，ブラシレスではないため消耗品になります．
参考：[B3M_SoftwareManual1.2.0.3](https://kondo-robot.com/w/wp-content/uploads/B3M_SoftwareManual1.2.0.3.pdf)

---

対応について

故障内容を考えると，ケーブルを変えて検証するなどは行わず，サーボモータ自体を交換した方が良いかと思います．

故障個所により故障要因が異なりますので，チーム内でサーボモータを分解し調査する余裕・技量が無いため，メーカ側に修理依頼し故障個所を特定，故障個所による故障原因を調査した方が良いかと思います．

現状示せる故障部位による故障要因としては，

ブラシの故障：寿命
FET系統：過剰電流
基板：衝撃負荷，過剰電流

だと思います．

@RikuYokoo @hiroki-0001

closed 2022-04-03 by MasatoKubotera