SUSTAINA-OP/SUSTAINA-OP #283: 164用のStartStopSwitch作成と予備の作成

RikuYokoo, 2022-03-03:
164機体でsssが以下の写真のようMicro-B部分が外れて故障してしまった．
[image]

現状予備もない状態であり、ハード班のタスクオーバ＋回路を触れる人が不足しているため練習として@RikuYoKooが作成を行う

MasatoKubotera, 2022-03-03:
@RikuYokoo 

データ通信に必要な4線。下図の赤色で示した部分のパッド（銅箔）が生きていれば基板を作り直しせずに、Micro USB type B 端子をはんだし直すだけでとりあえずは通信させることができます。

[image]

RikuYokoo, 2022-03-04:
おそらくその4本は生きていると思います。

他の部分(正方形の4つのところなど)は強度を上げるためのハンダってことですか？

MasatoKubotera, 2022-03-04:
@RikuYokoo そうですね。
4箇所中3箇所が生きているようなので強度的には問題ないと思います。

RikuYokoo, 2022-03-04:
[image]
画像のように通信の接続部が曲がっているので新しいので対応したいと思います．

@MasatoKubotera 
在庫にあるMicro USB type Bを使って大丈夫ですか？

MasatoKubotera, 2022-03-04:
ZX62-B-5PA(33)
https://akizukidenshi.com/catalog/g/gC-11183/

RikuYokoo, 2022-03-05:
昨日の時点ではんだ付けを試みましたが，自分が手順を誤り通信の部分のハンダ付けがうまく行かなかったため本日再度やりたいと思います．

RikuYokoo, 2022-03-05:
とりあえず164についていたstartstopswitchのハンダは終わりました．
動作確認も終わりました．
[動作確認](https://drive.google.com/file/d/1z3CWwkDtlz_DJejmGfqVN7OYZYH-GbLr/view?usp=sharing)

closed 2022-03-27 by MasatoKubotera