SUSTAINA-OP/SUSTAINA-OP #284: スプレッドシートの機能共有

NaokiIwasawa, 2022-03-05:
現在サーボモータや備品の管理でスプレッドシートを用いているが，使われている関数などの機能が初心者には分からない．そのため，全ての機能を説明すのではないが最低限の機能の共有を行いたいと思う．
詳細は実際に経験した方が早いと思うので，その際の基本レベル（バッテリー管理シートレベルなど）を提示できればと思う．

NaokiIwasawa, 2022-03-05:
@MasatoKubotera @hiroki-0001 
初心者が効率よくチームで運用しているレベルに達せるよう協力お願いします．

NaokiIwasawa, 2022-03-05:
事前に @hiroki-0001 に相談した際に示された参考サイトは以下である

クエリ関数の基礎
https://tonari-it.com/spreadsheet-what-is-query/
クエリ関数　selectの使い方
https://tonari-it.com/spreadsheet-query-select/
クエリ関数　whereの使い方
https://tonari-it.com/spreadsheet-query-where-comparison/

MasatoKubotera, 2022-03-05:
>初心者が効率よくチームで運用しているレベルに達せるよう協力お願いします．

下記に示します関数は汎用的なものなので実用レベルで使えるようになってください．

最低限覚える必要がある関数や機能
- SUM
- SUMIF
- MIN
- MAX
- AVERAGE
- AVERAGEIF
- VLOOKUP
- IF
- COUNT
- COUNTIF
- COUNTIFS
- IMPORTRANGE
- QUERY
- チェックボックス
- データの入力規則
- フィルタの作成
- 条件付き書式
- 交互の背景色
- 固定

管理システムなどの構築で必要な応用的な関数や機能
- UNIQUE
- MID
- AppScript
- IFNA
- CONCATENATE

NaokiIwasawa, 2022-03-12:
共有が一旦終了したのと，今後は以下で扱う方針になったのでcloseします．
- https://github.com/citbrains/citbrains_humanoid/issues/1711

closed 2022-03-12 by NaokiIwasawa