SUSTAINA-OP/SUSTAINA-OP #286: 新回路実装後のhr時挙動がおかしい

hiroki-0001, 2022-03-08:
新回路を導入したver4.1のhr時の挙動がおかしくなる。
症状としては、全身のB3Mサーボモータがエラーを示すLEDが点滅し、
機体の以下の部位

- 右足膝上ピッチ　(ID = 4)
- 右肩ピッチ　(ID = 7)
- 左足付け根ロール　(ID = 14)
- 左ひじピッチ　(ID = 18)

のB3Mサーボモータが力が入らず位置が固定されなくなる。

[その時の動画](https://photos.app.goo.gl/9Txhg8fmjxsGaxMJ8)

対処方法として、B3Mサーボモータの通信に使用しているUSBケーブルをVer4.1に使用しているL時のUSBケーブルからストレートのUSBケーブルのものに変更すると正常に動作した。

hiroki-0001, 2022-03-08:
hrの挙動がおかしくなるときに使用していたL字のUSBケーブル
[image] 

hrが正常に動作する際に使用していたストレートのUSBケーブル
[image]

hiroki-0001, 2022-03-08:
この問題はUSBの静電気保護機能の影響で起きているものと考え、Ver2.1 MainBoardの

- CG0603MLC-05E　×　4
- BLM18SN220TN1D　×　2

[image] 

[image] 

上図の青い枠のパーツを物理的に切断しました。
(できれば、はんだで外したかったですが、パーツが小さく取り外すことのできる技術を持った人が誰もいなかったためニッパーで切断しました)

[image] 

[image] 

-  BLM18SN220TN1D2を2つ切断したパターン 
- BLM18SN220TN1Dを2つとCG0603MLC-05Eを4つ切断したパターン 

の2つパターンでhrが正常動作するか確認しましたが、2つのパターンのどれもが正常に動作しませんでした。
また動かないB3Mサーボモータもhttps://github.com/citbrains/SUSTAINA-OP/issues/286#issue-1162307900 のときと変わらず

- 右足膝上ピッチ　(ID = 4)
- 右肩ピッチ　(ID = 7)
- 左足付け根ロール　(ID = 14)
- 左ひじピッチ　(ID = 18)

の部位が動かなったです。

MasatoKubotera, 2022-03-08:
@hiroki-0001 @MibuchiYuta 

D8，D9のCG0603MLC-05Eも切断してみてください．
これは，静電気保護ではありませんがMain Board Ver2.0→Main Board Ver2.1の変更点の変更点でもあります．

こちらを切断後に，L字ケーブルで検証
⇓
動作しない場合は，MainBoardにUSBを接続せずに外部からUSB/RS485変換基板を使って信号線(D+・D-)のみをMainBoardに接続して検証

してみて下さい．

MibuchiYuta, 2022-03-08:
@MasatoKubotera 
信号線のみをmainbordに接続する場合について質問です。6ポートあるサーボ用xhコネクタ（4ピン）とusb/rs485変換基板のxhコネクタをつなぐので問題ないでしょうか？

MasatoKubotera, 2022-03-08:
@MibuchiYuta 問題ありません。接続する時には2線ケーブルを使用してください。

MibuchiYuta, 2022-03-09:
>D8，D9のCG0603MLC-05Eも切断してみてください．
症状は改善しました

MibuchiYuta, 2022-03-09:
@MasatoKubotera 

>動作しない場合は，MainBoardにUSBを接続せずに外部からUSB/RS485変換基板を使って信号線(D+・D-)のみをMainBoardに接続して検証

症状は改善しませんでした
@MasatoKubotera

MibuchiYuta, 2022-03-09:
[image]

D8,D9 は切断後このようになりました

YukiShigematsu, 2022-03-11:
Ver2.0とVer2.1のFT232RL周辺の回路を比較したところ以下の赤印箇所の抵抗・配線の変更が確認できました。
[image] 
その中で、FT232RLのRXDピンに抵抗4.7K(R10)を介して3.3Vとつながっていますが、これがLTC1480との通信に以下のような要因で影響している可能性はないでしょうか。
・RXDがプルアップされている
・LTC1480のVCCにR10とR11によって分圧された電圧が供給されている

YukiShigematsu, 2022-03-11:
Ver2.0とver2.1のFT232RL周辺の回路を比較したところ写真の赤印箇所の抵抗・配線が変更されていることを確認したのですが、抵抗4.7K(R10)が接続されていることで以下のような理由で通信に影響を与えている可能性はないでしょうか。
・R10が接続されることでRXDがプルアップされている。
・R10とR11により、LTC1480のVCCに3.3Vが分圧されて印加されている。
[image]

MibuchiYuta, 2022-03-12:
J120の故障の疑いがあったため162機体のj120と交換し検証してみました。
結果として症状は改善しませんでした。

MibuchiYuta, 2022-03-12:
上記の作業の続きで、金メッキのUSB MiniBケーブルをしようしてhrと歩行をさせました。hrはできましたが歩行はできませんでした。また、終始サーボは点滅し各サーボの点滅は位相がずれていました。
[動画](https://drive.google.com/file/d/1BC-0uxoF8TB1DwVCoWHWK4GNt_QiSRb7/view?usp=sharing)

MibuchiYuta, 2022-03-12:
金メッキのUSB MiniBケーブルを使用し、サーボ通信のみ外部の変換基盤を用いて歩行させました。hrはできましたが歩行はできませんでした。また、終始サーボは点滅し各サーボの点滅は位相がずれていました。
[動画](https://drive.google.com/file/d/1RlbdFkVkaDj6y_TniL0Y6RDU1Fz6_q63/view?usp=sharing)

yasuohayashibara, 2022-03-13:
すでに分解してすぐに使える状態であれば良いですが，できるだけ部品の行き来はさせないほうが良いので，お渡ししたJ120を使用したほうが良いと思います．

> J120の故障の疑いがあったため162機体のj120と交換し検証してみました。
> 結果として症状は改善しませんでした。

yasuohayashibara, 2022-03-13:
ちょっと反応できない時間があり議論に加われなくて申し訳ありませんが，USBケーブルが問題ということはありませんでしょうか．
マイナーなメーカのUSBケーブルはUSB2.0の規格を満たしていないものがわりとあります．

長谷川君の研究で使用していた計測装置ですが，以下があります．
（どこにあるか曖昧なので，探さないと出てこないかもしれませんが．．．）
https://www.amazon.co.jp/dp/B07Y8BPVV4/ref=pe_2107282_266464282_TE_dp_1

MibuchiYuta, 2022-03-14:
## Mainboardを交換し戦略が動くところまで確認
検証試合を行うために動く状態にすることを現状の目標としました。以下修理内容です。
- JetsonTX2・J120共に170機体に使われていたものに戻した
- MainBoard2.1から2.0に交換した
- Mainboard変更に伴いケーブルの反転をおこなった
  - サーボケーブル用のxhコネクタ(4ピン)
  - Jetsonファン用電源ケーブル(2ピン)
  - J120給電ケーブル(2ピン)

バッテリ駆動させるためにXT60コネクタ部分を延長する必要があるが、そちらは本日対応します
[image]
反転させる前のJetsonファン用電源ケーブルとJ120給電ケーブル
[image]
Mainboard2.0

MibuchiYuta, 2022-03-14:
@yasuohayashibara 
>すでに分解してすぐに使える状態であれば良いですが，できるだけ部品の行き来はさせないほうが良いので，お渡ししたJ120を使用したほうが良いと思います．

こちらは170,162共に戻しました。

RikuYokoo, 2022-03-14:
>MainBoard2.1から2.0に交換した
 `PoseInitializer.sh`により初期位置を調整していたところ[動画](https://drive.google.com/file/d/1thA5jL2-lfHCgzWrLXRsNaPpdqJCjz2V/view?usp=sharing)にあるように一部サーボに力が入らなくなる現象が起きた．
その後，サーボ電源は落とさず，もう一度`connect`して`ServoOff`し`ServoOn`したら回復しました．(`connect`はいるかはわかりません)

yasuohayashibara, 2022-03-14:
以下の続きです．
https://github.com/citbrains/SUSTAINA-OP/issues/286#issuecomment-1066042252

USBケーブルチェッカーでケーブルを計測してみました．
接続されているかどうかと，電源の抵抗値しか出してくれませんでしたが，一応参考までに結果を載せます．

通信できなかったケーブル
[image]

通信できたケーブル
[image]

MasatoKubotera, 2022-04-03:
USB-HUBの他ポートにUSBメモリーを挿抜し内部データの読み込み書き込みを行いましたが通常通り動作したため，USB-RS485変換のどこかしらで問題が発生していたものと思われます．

こちらについて次期基板では，#290 で検証を行ったFT4232H MINI Moduleを用いるため，必然的に解決されます．

closed 2022-04-05 by MasatoKubotera