SUSTAINA-OP/SUSTAINA-OP #287: 新回路実装後、Jetsonが瞬電することがある

hiroki-0001, 2022-03-08:
hrによる直立姿勢の際にB3Mサーボモータの電源をスイッチで切り、その後スイッチでB3Mサーボモータの電源をつけ直すとJetsonの電源が瞬電します。
この現象自体は2022/3/8時点ではまだ1回しか起きていない現象ですが、念の為こちらのissueに記録として残しておきます。今後この症状が頻繁に起きてしまうようであれば、対策と原因の究明を行うようにしてください。

MasatoKubotera, 2022-03-08:
Ver2.1 MainBoard・SUSTAINA170機に変更するにあたりVer2.0 MainBoard・Ver4.0 171機からケーブル類を流用しています。

長さなどは適切に合わせておらずケーブル類の接触不良の疑いがあります。
Jetson電源部はかなりギリギリになってるかとおもいます。

こちらのコックピットに合わせたケーブルは、自作することも可能ですが難易度が高いのでGankenkunver4.0からはアセンブリワイヤ（圧着済み）を使用しています。SUSTAINAにあったケーブルはまだ発注していません。

昨日までに導通確認しかしていませんので今後、波形などについは確認する予定です。

yasuohayashibara, 2022-03-13:
> hrによる直立姿勢の際にB3Mサーボモータの電源をスイッチで切り、その後スイッチでB3Mサーボモータの電源をつけ直すとJetsonの電源が瞬電します。

すでに対処している場合は，無視していただいて良いコメントですが，サーボモータON/OFF次の電圧をオシロスコープで確認したほうが良いと思います．
サージが発生していると今すぐ壊れなくても，徐々に電気回路が壊れていきます．

こちらもすでに紹介したと思いますが，先輩もだいぶ悩まされました．
https://github.com/citbrains/citbrains_humanoid/issues/699

MasatoKubotera, 2022-06-18:
https://github.com/citbrains/citbrains_humanoid/issues/1528 で引き継ぎます．

closed 2022-06-18 by MasatoKubotera