SUSTAINA-OP/SUSTAINA-OP #288: 切削の技術継承

NaokiIwasawa, 2022-03-16:
現在切削に関して主に担当しているのが @NaokiIwasawa で，精度・ミスを減らすためのポイントが多くなってきた．
そのため，本issueには関連する主なissue，ドキュメントのリンクを示し目次のようなものになればと思う．

closeのタイミングについて，技術継承が目的のため関連のissueやドキュメントの整備が一旦完了したらcloseしたいと思う．後に追記する場合は念のため各issueなどにログを残すようにする．

NaokiIwasawa, 2022-03-16:
# 現在ある主なissueとドキュメント

---
issueなどを追加したら編集ログをこの欄に記載する

---

## CAM 
### 今後CAM継承ののメインissue
- https://github.com/citbrains/SUSTAINA-OP/issues/285 

### 関連
- [Autodesk Inventor CAM 2020を用いたGankenKun切削パスの生成法](https://docs.google.com/document/d/1MHM7T2BDNjA8xqrB54MIc1QrvLaiD6PXMtrIb-hadEg/edit)
- https://github.com/citbrains/SUSTAINA-OP/issues/273 (テンプレート更新について)
- https://github.com/citbrains/SUSTAINA-OP/issues/145 (初期テンプレート導入)
- https://github.com/citbrains/SUSTAINA-OP/issues/101 (切削送り速度検証)

直接ではないが関係がある
- https://github.com/citbrains/SUSTAINA-OP/issues/259 (切削できないポケット)

## マシニングセンタ関連
- [12号館4FのCNCの利用方法](https://docs.google.com/document/d/1n4-vayE9ptCkjq2pT49LtsxbtcDr-6Vd7hGP43_WFaM/edit)
- https://github.com/citbrains/SUSTAINA-OP/issues/253
- https://github.com/citbrains/SUSTAINA-OP/issues/197
- https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1018630634 (切削にかかる時間と計画)

## 精度
- https://github.com/citbrains/SUSTAINA-OP/issues/231 (嵌め合いの経験がまとめてある)
- https://github.com/citbrains/SUSTAINA-OP/issues/245#issuecomment-1014143724 (クーラントによる浮き報告)
- https://github.com/citbrains/SUSTAINA-OP/issues/264 (嵌め合い参考)

### 切り残しの実例は各機体の製作ログにある
- https://github.com/citbrains/SUSTAINA-OP/issues/254
- https://github.com/citbrains/SUSTAINA-OP/issues/245

----

### 未記載事項
- 両面テープを板にはるコツ
- 板材のバリ取り
- PPシートとオリジナルマインド製捨て板の使い分け
- マイクロメータでの計測とその活用
- 板をマシニングセンタに設置するコツ
- 曲がって貼ってしまったときの対処
- 切削中のトラブル対処法(停止・再開)
- ボトムオフセットの調整と切り残しの関係
- 切削送り速度の使い分け
- 板の再利用方法

など

NaokiIwasawa, 2022-03-17:
メモ
エンドミルの管理を記録する
今はミスミの袋とケースに開封日・管理番号・板・機体番号などを記載しているが不十分なためスプレッドシートか何かに記録する．

NaokiIwasawa, 2022-03-27:
> メモ エンドミルの管理を記録する 今はミスミの袋とケースに開封日・管理番号・板・機体番号などを記載しているが不十分なためスプレッドシートか何かに記録する．

[スプレッドシート](https://docs.google.com/spreadsheets/d/19B36EVSvOROwArWcq4vHEkkiur8F0hfsv_6M9Is0TVQ/edit#gid=1125225198)に状態を記録する場所を設けました．現在の使用状況はこちらで確認できます．

Keita1Mori, 2022-03-27:
切削について詳しい人に聞いたことを[**部品製作ログ**](https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1079969092)に報告しました。

KentaTsukino, 2022-05-24:
メモ程度に残しておきます
アルミを切削するときクーラントを流しますが、この状態はコックが全開になっている状態で工場の人曰くこんなにクーラントを出さなくても良いそうです
[image]

NaokiIwasawa, 2022-06-24:
# チャックホルダーの錆に関して

自身が久しぶりにチャックホルダーを確認したところ明らかに錆が進行していた．以前も少しぽつぽつ程度にあった気もするが画像1程ではなかった．そこで，工場の人に相談した．
[image]
画像1：錆びたチャックホルダー

## 相談と対応
梅雨の時期はどうしても触る部分は特に錆びやすいとのこと，その上で錆びたチャックホルダーに関しては紙やすりで軽くこすり錆を落としていただいた．
使用していたヤスリはポリネットというもので＃240か＃320のどちらかを使っていた．
[image]
画像2：ポリネット

### ヤスリがけを行った後
完全ではないが表面の錆が取れた
[image]
画像3：ヤスリがけ後

### 今後
錆びやすいものに関しては画像3に示すスプレーを布などに吹きかけパーツに塗ることで錆防止になるとのことなので，必要に応じて利用しようと思う．