SUSTAINA-OP/SUSTAINA-OP #289: 足裏の大きさの決定

MasatoKubotera, 2022-03-16:


MibuchiYuta, 2022-03-16:
足裏のサイズを決定するための判断材料があまりにも少なかったため、急遽過去のパーツを使用してSUSTAINA Prototype機体で検証を行いました。一部の組み合わせは実験できておりません。
- 厚み
  - T2
  - T3
- 面積 GankenKunVer3.0で使用されている面積のもの、ただし肉抜きがありません。
  - サイズを変えていないもの
  - 縦の長さを短くしたもの
  - 幅の長さを短くしたもの(今回はカットする時間がなかったためスパイクの位置を内側にすることで想定した幅に近づけております。)
- スパイクの本数
  - 4本
  - 6本

MibuchiYuta, 2022-03-16:
本日の検証全体を通してなんとなくわかってきたことです。

- T2を使用するとモーション再生時にたわみが発生しているようで転倒の回数が増え、歩行時も不安定だと思われる。
- 縦の長さを短くすると歩行は大幅に不安定になりキック時には必ず転倒してしまう。
- 幅の長さを短くすると歩行が少し不安定になる。
- スパイクの本数の影響はまだわからない
本日 @AD58-3104 @hiroki-0001 と話し合った結果以下のようになりました。
足のサイズをどのように削減するのが最適かを今すぐに判断することはできないが大会までのスケジュールもあるため横幅を10mmほど削ったT3の足裏をSUSTAINA Prototypeでの検証用に制作をメカ班に依頼する。

MasatoKubotera, 2022-03-17:
足裏の大きさに関して

レギュレーションでは重心の高さにより最大面積が決定され比率の制限も存在する．

- 重心の高さによる最大面積

>Each foot must fit into a rectangle of area (2.2・HCOM)^2/32. A foot is defined as the minimum encapsulating rectangle covering all mechanical parts below the ankle joint. The encapsulating rectangle should be in a plane parallel to the bottom contact surface of the foot.

各足は、面積 (2.2・HCOM)^2/32の長方形に収まっていなければならない。足とは、足首の関節より下のすべての機械部分を覆う最小限の長方形と定義されます。内包する長方形は、足裏の接触面に平行な面でなければならない。

- 比率

>The ratio between the longest and the shortest side of the encapsulating rectangle should be between 1.2 and 3.5

内包する長方形の長辺と短辺の比は、1.2～3.5であること。

参照：https://github.com/citbrains/SUSTAINA-OP/issues/166

---

現状の足裏は下記の仕様になっている．

Ver3.0　176.5 x 92.5 x 3mm

[image]

[image]

足裏の面積は，176.5 x 92.5 = 16,326.25[mm^2]であるため，重心高さが328.546mmよりも高い必要がある．

https://github.com/citbrains/citbrains_humanoid/issues/1142 で議論されているが，Ver3.0の重心の高さは，約34cmであるとされていた．

---

SUSTAINAのprototypeでは，https://github.com/citbrains/SUSTAINA-OP/issues/169#issuecomment-1062652355 にて重心高さ320mmとされている．
今後，Jetson Xavier NX，SCIカメラなどになった時に重心の高さが今よりも低くなることが予想されている．
そのため，重心の高さを低く見積もっておく．

https://github.com/citbrains/SUSTAINA-OP/issues/289#issuecomment-1069402775 の要望に従い，Ver3.0から大幅に面積を減らす必要があるが，この時に長辺の長さはできるだけ変えずに短辺の長さを削る方針とする．

検証用足裏は下記のような仕様にする．

175 x 80 x 3 
面積：14,000 mm^2
許容重心の高さ： 304.24mm以上
スパイクの本数：最大6本

[image]

---

短辺の長さは92.5mm→80mm（-12.5）に短縮されている．モデルの足裏は機体の外側から削っているがこれが正しいとは限らないため，外側にずらせるように調整用の穴を用意している．調整の間隔は，4.75mmであり三段階・最大で14.250mm外側にずらすことができる．

[image]

MasatoKubotera, 2022-03-17:
@MibuchiYuta 

量産機体用の切削都合により，3/22 火 夕方頃までに仕様の決定をお願いします．

MasatoKubotera, 2022-03-17:
長辺の長さは変更していませんが，前後にズラすための穴増設依頼があったため増設しました．

前後ともに5mmずつズラせる仕様になっています．

[image]

yasuohayashibara, 2022-03-17:
足裏はつま先など後で機能を追加するおそれがありますので，従来どおり４方向に取り付け用の穴は残しておいたほうが良いと思います．

MibuchiYuta, 2022-03-21:
[image]

yasuohayashibara, 2022-03-22:
上記の写真を見て，170の歩行の挙動が少し不安定である理由を思いつきました．
後で說明しますが，もし制御でなんとかならない場合は切削を遅らせたほうが良いと思います．

MasatoKubotera, 2022-03-22:
足裏の大きさについては、Jetson Xavier NX/EN715が届き次第質量・重心の計測を行って確定になります。

24日木曜日・25日金曜日のどちらかに切削出来ればと思います。

@MibuchiYuta @hiroki-0001

MibuchiYuta, 2022-04-02:
https://drive.google.com/file/d/12c--BspatIS3NrwgaoAU-wqqC4dP55wY/view?usp=sharing 
@yasuohayashibara 
こちら動画になります

yasuohayashibara, 2022-04-04:
勘違いさせてしまいました．
申し訳ありません．

前後方向ではなく，横方向から力を加えたときの動画があればお願いします．
ただし，自分で実ロボットを触りながら直接確認したので，優先度はだいぶ低いです．
（データとして残しておいたほうが良いと思いますが，実施しなくても構いません．）

MasatoKubotera, 2022-06-17:
SUSTAINA-OP Ver1.0の重心の計測を下記で行っています．
https://github.com/citbrains/SUSTAINA-OP/issues/169#issuecomment-1146883306

2022/06/06時点では重心高さが323.5mmとなっており，SUSTAINA-OP Ver1.0量産機体で採用している175 x 80 x 3mm　面積：14,000 mm^2の仕様では．許容重心の高さが304.24mm以上であるためルールに遵守していることを確認しました．

RoBoCup2022 Thailandでの足裏については，決定稿としてCloseします．

closed 2022-06-17 by MasatoKubotera