SUSTAINA-OP/SUSTAINA-OP #290: FT4232動作確認用の回路の作成

TakuyaNagamine, 2022-03-18:


TakuyaNagamine, 2022-03-18:
[こちらのissue](https://github.com/citbrains/SUSTAINA-OP/issues/120)で作成されていた回路は予算的な問題で発注が難しいので,ユニバーサル基板で作成する.

TakuyaNagamine, 2022-03-19:
今回の動作確認用の回路には,表面実装型のLTC1480を使用する予定であったが長谷川先輩からスルーホール実装型のLTC1480をいただいたので,ユニバーサル基板で作成する都合上スルーホール実装型のLTC1480を使用したいと思う(上記の二つに性能の差はない).

TakuyaNagamine, 2022-03-19:
@MasatoKubotera @hiroki-0001 
## 進捗報告
・部品のある程度の配置を行いました.
・中央にFT4232Hを配置し,左右にLTC1480を二つずつ接続する形にしました.
・12Vの電力供給のためのXT60はユニバーサル基板に配置することができないので,耐熱電子ワイヤーで接続しようと思います.
### 写真
[image]
・回路図(参考までに)
<img src="https://user-images.githubusercontent.com/53966390/151833752-469dac6c-698d-4e21-af0f-8ddab29a29ba.png" 
width="320px">

TakuyaNagamine, 2022-03-21:
@MasatoKubotera @hiroki-0001 
## 進捗報告
・LTC1480とサーボ間の配線が終了しました.
・FT4232HとLTC1480間の配線は半分完了しています.
### 写真
[image]
[image]

TakuyaNagamine, 2022-03-24:
@MasatoKubotera @hiroki-0001 
## 進捗報告
・配線は完了しました.
・導通チェック済みです.
### 写真
[image]
<img src="https://user-images.githubusercontent.com/83112606/159845090-55fcefc7-4052-43de-92c4-fa76e2b66226.jpg" 
width="320px">

TakuyaNagamine, 2022-03-27:
@MasatoKubotera @hiroki-0001 
## 進捗報告
・12V電源を接続する部分を被膜付き導線に変更しました.
### 写真
<img src="https://user-images.githubusercontent.com/83112606/160280389-55698b65-f368-4162-9cdb-80f1576a82d3.jpg" 
width="320px">
[image]

MasatoKubotera, 2022-04-02:
機体に実装した中で問題が起きているようです．
下記を参照してください．
[4又のRS485-USB変換でB3Mとの通信を高速化したらモーション再生速度が早くなった #1742](https://github.com/citbrains/citbrains_humanoid/issues/1742)

回路的な問題はないと思いますので実装については予定通り行っていきます．

closed 2022-04-03 by MasatoKubotera