SUSTAINA-OP/SUSTAINA-OP #291: Jetson Xavier NX / Carrier Board EN715 の組み立てと管理方法の確立

MasatoKubotera, 2022-03-26:


MasatoKubotera, 2022-03-27:
モジュールの組み立てに必要なものについては，下記通りです．

- [Jetson Xavier NX モジュール](https://www.macnica.co.jp/business/semiconductor/manufacturers/nvidia/products/134047/) x1
[image]
- [AverMedia Carrier Board EN715](https://www.avermedia.com/professional/product/en715/overview) x1
モジュール固定用 付属ネジx2
[image]
- [Jetson Xavier NXモジュール用 ファン付きアルミヒートシンク](https://www.switch-science.com/catalog/7434/) x1
付属ネジx4，ヒートシンクホルダx1，熱伝導シート x1
[image]

工具
- プラス0ドライバー
- トルクスT7ドライバー
- マイナス2.0ドライバー

特殊ドライバ（トルクス）が含まれるため，備品ではなく私物の[VANBOKEE 精密ドライバーセット104種ビット 磁気ドライバー 特殊ドライバーセット](https://www.amazon.co.jp/gp/product/B081RGG8RN/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1)を使って組み立てました．

組み立ての様子は[動画](https://drive.google.com/file/d/1osBURwOfY2mgs6uI4nGKdqDu98a4_X_c/view?usp=sharing)に収めました．

ファンモジュールのケーブルが短いため，組み立てが難しくなっています．

MasatoKubotera, 2022-03-27:
Jetson Xavier NX ，Carrier Board EN715については管理番号を設けて管理したいと思います．

それぞれ下図のように管理番号を貼付しています．
[image]

管理番号は下記のようになります．
- Jetson Xavier NX：NX-01~NX-07
- Carrier Board EN715：EN715-01~EN715-07

それぞれログ用のドキュメントを設けましたので不具合や検証などに用いた時などはドキュメントに記載してください．
 - [Jetson Xavier NX ログドキュメントフォルダ](https://drive.google.com/drive/folders/1oR6By0PQfVsRTEwZJ8iMHvyhEWLYKXg6?usp=sharing)
 - [Carrier Board EN715 ログドキュメントフォルダ](https://drive.google.com/drive/folders/1pyA_pnJyAJsP5JvjJAqVyy8GHTtxs9n3?usp=sharing)

MasatoKubotera, 2022-03-27:
Carrier Board付属の電源アダプタを使用する場合には，下図のように電源コネクタ（[MB332-350M02](https://www.avnet.com/shop/japan/products/multicomp-pro/mb332-350m02-3074457345645968881/)）に付属ケーブルを挿入し固定してください．

[image]

MasatoKubotera, 2022-03-27:
[Carrier Board - EN715 USer Manual](https://storage.avermedia.com/web_release_www/EN715/AVerAI_UM_EN715_NO111B_NX211B_TN111B.pdf)の3.13 Optional Function Selectionに記載されているが，Carrier Boardのスライドスイッチでモードを変更することができる．

検証，SUSTAINA-OPに搭載する場合には，下図のようにスイッチを切り替える必要がある．

[image]

MibuchiYuta, 2022-04-02:
## 管理方法に関して

それぞれの本体の以下の箇所にテプラで管理番号をラベリングしました。以下の番号で振り分けています
- NVIDIA Jetson Xavier NX-01 〜07
- AVer Media EN715-01 〜 07
[image][image]

また、それぞれの状態、変更内容を記録するログファイルをGoogleDrive上の19_検証ドライブ内に作成しました。
- [01_Jetson Xavier NX](https://drive.google.com/drive/folders/1oR6By0PQfVsRTEwZJ8iMHvyhEWLYKXg6?usp=sharing)
- [02_AVerMedia EN175](https://drive.google.com/drive/folders/1pyA_pnJyAJsP5JvjJAqVyy8GHTtxs9n3?usp=sharing)

MibuchiYuta, 2022-04-14:
@MasatoKubotera 
ログドキュメントにこれまでの作業で記載するべき事項がありましたら教えてください。

MasatoKubotera, 2022-04-15:
現状，開発中なので貸し出し状況とかでしょうか．

TakuyaNagamine, 2022-04-18:
2022/4/18
以下のモジュールを組み立てました
・NX-04,EN715-04
・NX-04,EN715-04

MasatoKubotera, 2022-04-18:
組み立てたのみで通電・動作確認はしていません。

MasatoKubotera, 2022-04-18:
組み立てに下記の工具を使用しました．

- [アネックス(ANEX) フォーラインドライバー　レギュラータイプ　+1×100](http://www.anextool.co.jp/item/8400-1-100)
- [アネックス(ANEX) 精密用マイナスドライバー -2.5×50 1220](http://www.anextool.co.jp/item/1220-2-5-50)
- [ベッセル(VESSEL) パワーグリップ トルクスドライバー No.5400TX(T20H×120)](https://vessel.co.jp/product/screwdriver/122861/)

トルクスドライバーについて，ワンサイズ下でもよさそうな感じがしました．

[VANBOKEE 精密ドライバーセット104種ビット](https://www.amazon.co.jp/gp/product/B081RGG8RN/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1)のT7が一番しっくりきましたが，[ベッセル(VESSEL) パワーグリップ トルクスドライバー No.5400TX(T20H×120)](https://vessel.co.jp/product/screwdriver/122861/)と同種であるためどのサイズが良いかよくわかっていません．

MasatoKubotera, 2022-04-22:
組み立て並びに管理方法の確立に伴いドキュメントの整備を行いましたのでcloseします．

closed 2022-04-22 by MasatoKubotera