SUSTAINA-OP/SUSTAINA-OP #292: スイッチング電源の組み立てと管理方法の確立

MasatoKubotera, 2022-03-26:


MasatoKubotera, 2022-03-27:
1台のスイッチング電源の組み立てに必要な用品は下記の通りです．
- [LEDMO スイッチング電源 AC 100V / 240V〜DC 12V 20A 240W](https://www.amazon.co.jp/dp/B01E6S0JS4/ref=cm_sw_r_tw_dp_K3W6TWKR5763G84J1EYA) x1
- [14AWGシリコンケーブル線 3mm](https://www.amazon.co.jp/dp/B07CWW2CB9/ref=cm_sw_r_tw_dp_T2HJW36WG0A35CZBAQZ3?_encoding=UTF8&psc=1) x1
- [ニチフ 銅線用 環境配慮形 絶縁被覆付圧着端子 (R形)丸形](https://www.monotaro.com/p/8956/9025/) x4
- [ACコード 固定長（PSE） 片側切り落としタイププラグ (2M-3)](https://jp.misumi-ec.com/vona2/detail/110500082830/?HissuCode=2M-3) x1
- [AMASS XT60Hコネクタ メス](https://store.shopping.yahoo.co.jp/suzakulab/amass-xt60h-f.html?sc_i=shp_pc_store-item_lemstr) x1

工具
- 電光ペンチ
- ニッパー
- ワイヤーストリッパー
- はんだごて
- はんだ
- マイナスドライバー（入力電圧切り替え用）

MasatoKubotera, 2022-03-27:
[LEDMO スイッチング電源 AC 100V / 240V〜DC 12V 20A 240W](https://www.amazon.co.jp/dp/B01E6S0JS4/ref=cm_sw_r_tw_dp_K3W6TWKR5763G84J1EYA) x1

[image]

について，ラベル表記が100~120V 60Hzとなっており西日本でしか対応してないように思えるが，販売ページで50Hz/60Hzで使えることが明記されているため１１号館でも使用できる．

MasatoKubotera, 2022-03-27:
入力電圧が110Vと220Vをスイッチをスライドスライドすることで選択できる仕様になっている．
オフセットでは，220V側になっているのでコンセントプラグをコンセントにつなぐ前に変更する必要がある．
[image]

MibuchiYuta, 2022-03-31:
（追記します）

[image]
[image]
[image]

MibuchiYuta, 2022-04-03:
@MasatoKubotera 管理番号は貼付しますが、その他にログドキュメントの作成などは行いますか？

MibuchiYuta, 2022-04-04:
[image][image][image]

MasatoKubotera, 2022-04-05:
@MibuchiYuta 

>@MasatoKubotera 管理番号は貼付しますが、その他にログドキュメントの作成などは行いますか？

ログドキュメントの作成をお願いします．

MibuchiYuta, 2022-04-06:
SW Power-01をコンセントに挿してみたところ異臭とまでは言えませんが特有の匂いがしました。こちらに関して異常があるかどうかの判断はできません

MasatoKubotera, 2022-04-08:
@MibuchiYuta 12Vが出力されていることを確認してください．

MasatoKubotera, 2022-04-13:
@MibuchiYuta

MibuchiYuta, 2022-04-13:
今週中に行います

MibuchiYuta, 2022-04-14:
# 電圧確認
１10秒間電圧計で計測し、最後の値が以下になりました
| 電源管理番号 | 電圧 |
| :------: | :------: |
| SW Power-01 | 12.12v |
| SW Power-02 | 12.21v |
| SW Power-03 | 12.14v |
| SW Power-04 | 12.27v |
| SW Power-05 | 12.28v |
| SW Power-06 | 12.24v |

MibuchiYuta, 2022-04-14:
実機の実験のために01,02,03を使用しました。１時間程度問題なく使用できました。

MibuchiYuta, 2022-04-14:
@MasatoKubotera 
ログドキュメントですが、ここまでの作業で書いておいた方がいいことなどはありますか?

MibuchiYuta, 2022-04-14:
報告ド忘れしていました
こちら[ログドキュメント](https://drive.google.com/drive/folders/19HboZS6_uFZ4ag-LPdF3OL1M6SUP4s22?usp=sharing)です

MibuchiYuta, 2022-04-20:
[image][image][image]![image](https://user-images.githubusercontent.com/53966307/164133695-980377d2-e3ce-4a91-a1a5-4614198504b5.jpeg

MibuchiYuta, 2022-04-20:
# 実運用に関して
4/18よりスイッチング電源を使用しています。これまで電源に使用していたものは段ボールの中にしまいました。
[image] [image] [image]

MasatoKubotera, 2022-04-22:
旧Switching電源との入れ替えまで完了したためcloseします．

closed 2022-04-22 by MasatoKubotera