SUSTAINA-OP/SUSTAINA-OP #293: SUSTAINA ver1.0 組み立てログ

Keita1Mori, 2022-03-28:
このissueではSUSTAINA ver1.0 を組み立ての記録
# 記録内容
- 組み立て部位
- 所要時間
- 担当者
- 使用したパーツ
- 作業内容（その他気がついたこと、メモなど）

Keita1Mori, 2022-03-28:
# 22/03/27 作業内容
## 組み立て部位
- Ankle　×３
[image]

## 所要時間
- 2時間
## 担当者
@MasatoKubotera @NaokiIwasawa @Keita1Mori 
## 使用したパーツ
- Ankle_side_frame ×6
- Ankle_back_frame ×6
- Ankle_front_frame ×3
- S4-05-3W1 ×3
- SLSBF6-39.5-M3-N3 ×6
## 作業内容
Ankle パーツの組み立てを行った。
初めにはめ合うところの角を軽くバリ取りする。
[image]
[image]
[image]

はめ合い剤を塗る場所としては受け側のAnkle_side_frameの溝に薄く塗ります。はめた時に出てきたはめ合い剤はウエスやキムワイプなどでふき取ること。

はめる際にバイスを使用するとパーツが曲がってしまうため人力もしくは圧入機を使う。圧入機を使用する場合はT3のパーツのみを使用。（この場合はAnkle_front_frameのみ)
[image]

パーツを押す際に使用する工具として工作室にある板を使用することがオススメ
[image]

## 次回作業内容
今回動画を撮り忘れたため次回Ankleを組み立てる際動画を撮りここにアップロードすること

NaokiIwasawa, 2022-04-11:
# 22/04/08 作業内容
## 組み立て部位
- Leg_joint_pitch_frame
[image]

## 所要時間
- 1時間
## 担当者
@NaokiIwasawa @Keita1Mori @KentaTsukino @aiba0921 

## 使用したパーツ
- Leg_joint_pitch_frame　×4
- SF419　×4
- sf419_press-fit_Jig　×4

## 作業内容
1. Leg_joint_pitch_frameのSF419圧入用の穴側面に嵌め合い剤爪楊枝などで少量塗布する
2. 多い分はペーパータオルなどでふき取る
4. 圧入プレス機（手動）を床に置き本体が浮かないように体重を乗せる
5. sf419_press-fit_Jigを丁度重なるようにセットしたら力の入りやすい態勢でプレスする
※自重では不十分だったためバーに対して上向き方向の力を加え地面と突っ張るようにして力を加えた．この場合ある程度の筋力が無いと最後まで圧入しきれないが自身はこの方法で行った．

## sf419_press-fit_Jig
以前までは切削時の円形端材で行っていたが，サイズが合わず圧入部に偏りが生じてしまっていた．そのため，直接の因果関係は分からないがパーツが大きく歪んでしまう結果となりパーツを利用できなくなった．
そこで，今回から圧入しやすいようにT2の圧入よ治具を作成し試してみた．結果，前回より位置ずれによる不均一な圧入がへりSF419自体の歪みも少なく済んだ．
また，この外径は同じくSF419を圧入するAnkle_side_pitch_frameのポケットに合わせて作ったのだが，重ねてみた様子ではサイズに問題はなさそうであった．
[image]Ankle_side_pitch_frame

### 消耗具合
sf419_press-fit_Jigの消耗具合について，最後まで圧入する前に表面に凹みができ仕上げまでは1面だけではできない．そのため，複数回に分けて圧入を行う場合最後の1回は綺麗な面で行うため1つ圧入するのに2面程度消耗した．そのため，SF419の分だけsf419_press-fit_Jigを用意することを推奨する．