SUSTAINA-OP/SUSTAINA-OP #294: サーボ修理申請2022年度

Keita1Mori, 2022-03-29:
https://github.com/citbrains/SUSTAINA-OP/issues/99
こちらのissueを参考にしながら修理申請を出そうと思います。
## 必要書類&参考書類
[修理依頼書](https://kondo-robot.com/w/wp-content/uploads/repair_send20150817.pdf)
[故障リスト](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit#gid=372255127)
[B3Mの修理申請の手順について](https://docs.google.com/document/d/1MsO7cTQQ8Z5c2m2XbP8l9NtJM6mksZJ8x2GYoWaFl9U/edit#)

Keita1Mori, 2022-03-29:
# 追記報告
今日口頭で野口先輩からスプレッドシート内の故障登録されているサーボ数と実際あるサーボ数の数を比べてミスがないかチェックしてくださいとのことだったので明日の切削が終わり次第確認したいと思います。

MasatoKubotera, 2022-03-31:
@Keita1Mori 
下記のissueに目を通してください．
https://github.com/citbrains/SUSTAINA-OP/issues/282

修理項目などや優先順位の決め方などについて口頭で伝えていることをここに書き出しておいてください．

Keita1Mori, 2022-04-01:
# 追加報告

> 修理項目などや優先順位の決め方などについて口頭で伝えていることをここに書き出しておいてください．

とありましたので報告します。
- 修理項目
  - 回路
  - モータ
  - ギア
- 修理の優先順位の決め方
  - 40サーボモータが優先
  - ギアなど外部の故障より回路の故障が優先

Keita1Mori, 2022-04-02:
https://github.com/citbrains/SUSTAINA-OP/issues/277#issuecomment-1079559600 のissueコメントでサーボ本体の出力軸ネジ、フリー軸ネジのネジ剤が取れなかったものに関しては総修理数が少なければ洗浄できないかKONDOに確認する。多ければ修理には出さない。

MasatoKubotera, 2022-04-03:
@Keita1Mori 
>https://github.com/citbrains/SUSTAINA-OP/issues/277#issuecomment-1079559600 のissueコメントでサーボ本体の出力軸ネジ、フリー軸ネジのネジ剤が取れなかったものに関しては総修理数が少なければ洗浄できないかKONDOに確認する。多ければ修理には出さない。

洗浄ではなく，おそらくファイナルギア並びにボトムケースの交換となると思います．確認していただければ結論がわかると思います．

ネジ材の残程度によりますが，そのまま使うことやフリー軸を使わない部位に使うことを検討した方が良いかと思います．

MasatoKubotera, 2022-04-03:
@Keita1Mori 

https://github.com/citbrains/SUSTAINA-OP/issues/294#issuecomment-1085286514
>修理項目などや優先順位の決め方などについて口頭で伝えていることをここに書き出しておいてください．
とありましたので報告します。
修理項目
回路
モータ
ギア
修理の優先順位の決め方
40サーボモータが優先
ギアなど外部の故障より回路の故障が優先

XHコネクタについても1か所100円で修理が可能ですので，挿抜寿命が少ないため修理に出すものすべて交換してもらった方が良いかと思います．

どこかでファームウェアのバージョン違いがあったと聞いたので書き換えについてついでに問い合わせしていただければと思います．

Keita1Mori, 2022-04-09:
@yasuohayashibara @hiroki-0001 @aiba0921
お金の話であるため以下のスプレッドシートにまとめておきました。ので確認のほどよろしくお願いします。
[2022年度概算見積もり](https://docs.google.com/spreadsheets/d/1htqCgbhsu0IrcMZuzL0zGw9SH6oHLh6IAHdAOwkYViI/edit#gid=1315693907)
※チーム外秘

yasuohayashibara, 2022-04-09:
> お金の話であるため以下のスプレッドシートにまとめておきました。ので確認のほどよろしくお願いします。

総額いくらでしょうか？
仮合計が1300円になっています．

Keita1Mori, 2022-04-09:
@yasuohayashibara 
現状詳細を調査する項目はできたのですが、詳細についての原因究明ができていないため今日原因を究明したのちに総額を出したいと思います。

aiba0921, 2022-04-09:
https://github.com/citbrains/SUSTAINA-OP/issues/281#issuecomment-1057121937 に関して移植は @hiroki-0001 と相談し、難しいため行わないことになりました。また、[この表](https://docs.google.com/spreadsheets/d/1htqCgbhsu0IrcMZuzL0zGw9SH6oHLh6IAHdAOwkYViI/edit#gid=0)を参考にした場合、修理することになる可能性の高いケースやコネクタは費用があまりかかりません。今回の修理申請では基板などを優先しているため出さずに自分達で費用を負担する別の機会に出すことになりました。

Keita1Mori, 2022-04-09:
# 4/9　追加報告
午後サーボのガタ調査をした際に上記で示した40-1922モータを使用したところ、コネクタを下としたときにサーボが反時計回りに回る際動かなかったので再び故障認定しました。
なので現状合計**12個**のサーボモータが修理対象となっています。

hiroki-0001, 2022-04-10:
[新_B3M管理シート](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit?usp=sharing)上での40-2106の管理状態を故障から在庫に変更しておきました。

hiroki-0001, 2022-04-10:
https://github.com/citbrains/SUSTAINA-OP/issues/248 のガタの調査で特にひどい症状だった3つのB3M

- 70-1982
- 70-1921
- 40-1956

これらを修理に対象としたい思います。

hiroki-0001, 2022-04-10:
@Keita1Mori 修理申請に必要な書類を作成してくれていると思いますが、添削・確認等したいので完成したら、こちらのissueに投げてくださると助かります

yasuohayashibara, 2022-04-10:
> #248 のガタの調査で特にひどい症状だった3つのB3M
．．．
> これらも修理に出したいと思います。

手で触って明らかに問題があることが分かれば良いですが，分からない場合は問題無しで帰ってくるかもしれませんので，データを添付して送った方が良いかもしれません．

Keita1Mori, 2022-04-10:
# 修理申請に必要な書類
- 問い合わせメールのメッセージ
  - [連絡メール](https://docs.google.com/document/d/1V71QAztfOfgWtX0whzy1-JmTlX1K3rk8y8dff2AG73c/edit#)
- 修理依頼書
  - [2022年度修理依頼書](https://drive.google.com/file/d/1HuEMEum4UyibE9E2YGHhfBEMkUd3k-Rv/view?usp=sharing) 
- 故障状態についての詳細
  - [具体的な故障の症状について](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit#gid=372255127)
- サーボのガタデータ
  - [B3M評価計測 シート](https://docs.google.com/spreadsheets/d/1ytKEkW5EvP1CdU5-lgqlxm7ZQpeUXrU9aAGpEB-B7Rg/edit#gid=1328542408)

yasuohayashibara, 2022-04-10:
即時性の方が重要かもしれませんので，出してしまったほうが良いかもしれませんが，以下どのような不具合なのか分かりにくいかもしれません．

(元は岩澤くんが記入)　歩行中に動かなくなるためSCに両足で同時交換
↑ 具体的に何が問題か分かりにくいです．

出力ホーンが回らない
↑ こちらも何が問題か分かりにくいです．
　電源を入れない状態で，人が力を加えて動かないという話でしょうか？
　もしくは，電源を入れて目標角を入れても動かないという話でしょうか？
　こちらが何を故障として考えているかを明確に伝えたほうが良いと思います．
　（状況を再現できるように説明しないと，故障無しで返却されるおそれがあります．）

hiroki-0001, 2022-04-11:
@Keita1Mori 今確認しましたが、SBシリーズのB3Mは今回の修理には出さないので、**具体的な故障の症状について**という資料に関しては管理シートの故障表示をそのまま使うのではなく、別途で用意したほうが良いかもしれません。また故障内容に関しては林原先生がおっしゃる通りもう少し詳細に記入したほうがいいですね。時間があまりないので今回は自分が作成してもいいですがどうしますか？

hiroki-0001, 2022-04-12:
故障内容について記載した資料ができました、
https://docs.google.com/spreadsheets/d/1MWSca0fwiOxtNKtGUd6AFXDdAXeRtFhv8CxAHal2hTc/edit?usp=sharing

@yasuohayashibara @Keita1Mori 確認をお願いします。

hiroki-0001, 2022-04-12:
@yasuohayashibara 大学側に修理の申請を行う際に必要そうなので、https://github.com/citbrains/SUSTAINA-OP/issues/294#issuecomment-1096344948 で作成したものに、故障時期・故障時の状況・故障原因を追加したものも作成しておきました。利用できそうなところがあれば利用してください。

hiroki-0001, 2022-04-12:
@Keita1Mori が作成してくれた[2022年度修理依頼書](https://drive.google.com/file/d/1HuEMEum4UyibE9E2YGHhfBEMkUd3k-Rv/view?usp=sharing)ですが、今回修理に出すB3Mのfirmwareのバージョンを最新版に上げてほしいのでそれを依頼する旨を文の追加をお願いします。

hiroki-0001, 2022-04-13:
修理依頼書でしたが自分で編集してしまいました。
https://drive.google.com/file/d/1BrpELZ6c86jhQ4XnCZNVJAYctw7vRH2h/view?usp=sharing

hiroki-0001, 2022-04-13:
> 故障内容について記載した資料ができました、
[220412_故障内容について_kondo用](https://docs.google.com/spreadsheets/d/1MWSca0fwiOxtNKtGUd6AFXDdAXeRtFhv8CxAHal2hTc/edit?usp=sharing)

で作成した資料ですが、ガタに関するデータの情報も追加で加えました。

hiroki-0001, 2022-04-13:
修理申請に必要な資料がとりあえず完成しました。以下にリンクをまとめておきます。

- [修理依頼書](https://drive.google.com/file/d/1BrpELZ6c86jhQ4XnCZNVJAYctw7vRH2h/view?usp=sharing)
- [具体的な故障内容 + ガタの情報](https://docs.google.com/spreadsheets/d/1MWSca0fwiOxtNKtGUd6AFXDdAXeRtFhv8CxAHal2hTc/edit?usp=sharing)
- [CIT BrainsでのB3Mの使用状況](https://docs.google.com/document/d/1VXUUf6xh4dTPcMYsGC-lY1rX0N7WBy89WiCjNhpsRuI/edit?usp=sharing)
- [修理申請前の写真](https://drive.google.com/drive/folders/1jRqQwO4arlQmk-wbKP7SrverA2iG7NiW?usp=sharing)

hiroki-0001, 2022-04-14:
修理にかかる値段ですが、ホームページに記載されてありました。
https://kondo-robot.com/w/wp-content/uploads/parts_robot_servo20210911-1.pdf

pdfの4ページ目の下部分に記載されています。
以前書いた[B3M_修理費用指標](https://docs.google.com/spreadsheets/d/1htqCgbhsu0IrcMZuzL0zGw9SH6oHLh6IAHdAOwkYViI/edit?usp=sharing)の値段を更新しておきました。こちらの値段は最悪を想定して設定されているため値段は高めになっています

hiroki-0001, 2022-04-14:
森くんが荷物を配送してくれました。
@Keita1Mori 連絡が届き次第こちらのissueに報告してください。

hiroki-0001, 2022-04-15:
https://github.com/citbrains/SUSTAINA-OP/issues/248#issuecomment-1098896529 で判明したガタのあるB3Mサーボモータをチーム内の予算で修理に出したいと思います。(久保寺くん了承済みです)
それにあたり以下の資料を作成しました。

- [修理依頼書　チーム予算版]( https://drive.google.com/file/d/1eAOV_e1YBMjksNVig-BzoCCBcm0XBH-2/view?usp=sharing)
- [故障内容について](https://docs.google.com/spreadsheets/d/1qetqbp4f7TafjIDgd3oLkX8REvGhCDl4ecqp7axv9ow/edit?usp=sharing)
- [B3M Series サーボモータ ガタの調査方法について](https://docs.google.com/document/d/1ivEqceVjtf6yNF0DpCNuKa5BZCFlrh3ET17w9fB6Plk/edit?usp=sharing)
    -  今回の修理に出すB3Mのほとんどがガタの異常なので、どのようにガタを調査したかを簡易的に記載したもの
- [B3M_修理費用指標_2022年度版　チーム予算版](https://docs.google.com/spreadsheets/d/1htqCgbhsu0IrcMZuzL0zGw9SH6oHLh6IAHdAOwkYViI/edit#gid=1469250925)

これらの資料はすべて[こちらにおいてあります](https://drive.google.com/drive/folders/1e2PlJBx09eGDCOPO19JT4-s2jLLe7IcN?usp=sharing)

こちらの修理申請が長引くとSUSTAINAの量産に大きく影響されるため、資料等に問題なければ今日中に発送したいと考えています。

hiroki-0001, 2022-04-15:
> https://github.com/citbrains/SUSTAINA-OP/issues/248#issuecomment-1098896529 で判明したガタのあるB3Mサーボモータをチーム内の予算で修理に出したいと思います。(久保寺くん了承済みです)
それにあたり以下の資料を作成しました.
> - [修理依頼書　チーム予算版]( https://drive.google.com/file/d/1eAOV_e1YBMjksNVig-BzoCCBcm0XBH-2/view?usp=sharing)
> - [...]
    -  今回の修理に出すB3Mのほとんどがガタの異常なので、どのようにガタを調査したかを簡易的に記載したもの
> - [B3M_修理費用指標_2022年度版　チーム予算版](https://docs.google.com/spreadsheets/d/1htqCgbhsu0IrcMZuzL0zGw9SH6oHLh6IAHdAOwkYViI/edit#gid=1469250925)
    - ギア交換による一番高価な修理費用を選択しているため、実際の価格はこれよりも少なくなることが予測されます
これらの資料はすべて[こちらにおいてあります](https://drive.google.com/drive/folders/1e2PlJBx09eGDCOPO19JT4-s2jLLe7IcN?usp=sharing)
> こちらの修理申請が長引くとSUSTAINAの量産に大きく影響されるため、資料等に問題なければ今日中に発送したいと考えています。

こちらの資料に問題はなさそうなので、本日中に発送しておきます。

Keita1Mori, 2022-04-29:
@yasuohayashibara @hiroki-0001 @MasatoKubotera 

以下の見積書を近藤科学からいただきました。
チーム予算利用:[20220425_3_千葉工業大学様_修理見積.pdf](https://drive.google.com/file/d/19bosMEy3-NXL7GUskXMybvSuaYz4-tTy/view?usp=drivesdk)
大学予算利用:[20220425_2_千葉工業大学様_修理見積.pdf](https://drive.google.com/file/d/1nbUFGjCIC3-sU3vKxwJ-XcXYKX9p7tPJ/view?usp=drivesdk)

修理内容は以下のドキュメント内に近藤科学からいただいたものがあるので確認できます。

[問い合わせ内容その③：修理申請4/23の返信(4/25)その②](https://docs.google.com/document/d/1V71QAztfOfgWtX0whzy1-JmTlX1K3rk8y8dff2AG73c/edit)

Keita1Mori, 2022-05-20:
@yasuohayashibara @hiroki-0001 @MasatoKubotera 
本日サーボ修理が完了したものを @HarukiOgawa1 に運んでもらいました。修理後の写真はこちらのリンクから確認できます。
https://drive.google.com/drive/folders/1h7-_0_oR2dRRa2ote_SGiWLEb2Atahm5?usp=sharing
備品管理登録に反映できていないのでそちらが終わり次第このissueをcloseしたいと思います。

Keita1Mori, 2022-05-20:
上記の修理済みサーボの備品管理登録終わりました。サーボ用ケースに移動済みです。

MasatoKubotera, 2022-05-26:
修理後の評価については，下記のissueで行います．
- https://github.com/citbrains/SUSTAINA-OP/issues/316

closed 2022-05-26 by MasatoKubotera