SUSTAINA-OP/SUSTAINA-OP #295: GankenKun162機体のコックピット修理

MibuchiYuta, 2022-03-31:


MibuchiYuta, 2022-04-02:
3月中旬にGankenKun162をセットアップしている最中に突然Jetson TX2が再起動することが何度かあった。DCDCコンバータの不具合による電力供給の不安定化が原因と断定しDCDCコンバータの修理を行いました。

MibuchiYuta, 2022-04-02:
[DCDCコンバータ(Amazon)](https://www.amazon.co.jp/Aideepen-5%E5%80%8B%E3%82%BB%E3%83%83%E3%83%88XL6009-DC-DC%E6%98%87%E9%99%8D%E5%9C%A7%E9%9B%BB%E6%BA%90%E3%83%A2%E3%82%B8%E3%83%A5%E3%83%BC%E3%83%AB3-8%E3%80%9C32V-to-1-25%E3%80%9C35V%E9%9B%BB%E5%9C%A7%E3%83%AC%E3%82%AE%E3%83%A5%E3%83%AC%E3%83%BC%E3%82%BF%E3%82%B3%E3%83%B3%E3%83%90%E3%83%BC%E3%82%BF/dp/B07S8WTHHF/ref=pd_sbs_3/355-3191415-8692265?pd_rd_w=PGknj&pf_rd_p=9a8dc6f0-e2c4-43d1-957f-96153e4a70bc&pf_rd_r=2F5R43T3J7XG27AZJ8RG&pd_rd_r=e6510389-2716-460b-9ec3-8c2d1bbff13d&pd_rd_wg=BjK8x&pd_rd_i=B07MLL7JSH&th=1) が在庫にあったため、耐熱電子ワイヤー外径0.75ｍｍ（ＡＷＧ２２相当）[赤](https://akizukidenshi.com/catalog/g/gP-11610/) [黒](https://akizukidenshi.com/catalog/g/gP-11611/)を半田付けし、xhコネクタを圧着、取り付けました。

その後配線の間違いがないかを確認し、コックピット内のMainboardと12Vのスイッチング電源をもちいて出力電圧の計測を行いました。結果として導通させた瞬間に高周波音がし、OUT+付近が発熱したためスイッチング電源を切りました。 その後再度導通させ電圧を計測しましたが出力電流は0Vに近かったです(詳細の電圧を記録し忘れました)

画像は後日添付します

MibuchiYuta, 2022-04-02:
↑ 上記の発熱に関して地震が行った半田付けに問題がある可能性があったため目視でハンダがどこかに流れ込んでないか確認しましたが、問題点は見つけられませんでした。

画像は後日添付します

MibuchiYuta, 2022-04-02:
上記の問題をDCDCコンバータの初期不良と断定し、在庫からもう一つのDCDCコンバータを同じ手順で加工しました。その後安定化電源を用いて11.5Vを入力電源とし出力が12Vになるように調整しました。

MibuchiYuta, 2022-04-02:
コート上で平常通りにセットアップを行ったところ以下二点の問題が発生しました。
- Jetson TX2内 Ubuntuにてpoweroff処理を行なったにもかかわらず再起動した
- ready姿勢時に全身のサーボモータのledが消灯→ホーンがフリーモードのように(実際には確認しておりません)力が入らなくなり、Jetsonは電源が切れる。

MasatoKubotera, 2022-04-02:
>Jetson TX2内 Ubuntuにてpoweroff処理を行なったにもかかわらず再起動した

他の機体でも時たまみられる症状なので，システムの方の問題かもしれません．
電子回路周りだとすると，この機体の回路Verを把握していませんがロッカスイッチ，FET周りの故障が考えられます．意図せずに回路側のスイッチング回路がONOFFしている可能性があります．また，定格を超えて逆流防止のダイオードを使っていた気がしますのでそのあたりもでしょうか．．．．

>ready姿勢時に全身のサーボモータのledが消灯しサーボが脱力した、Jetsonは電源が切れる。

システム面のバグを除いて考えると，こちらもスイッチング回路周辺の故障が挙げられると思います．

このあたりの回路修理は技術的に@MibuchiYuta が対応できる範囲を超えていると思いますので，修理についてはいったん保留にしたいと考えますが大丈夫でしょうか．
FETについてですが，自分が把握しているBrains在庫表では在庫数0のため同型番の置換修理はできません．

[SUM110P06-07L-E3](https://www.mouser.jp/ProductDetail/Vishay-Semiconductors/SUM110P06-07L-E3?qs=xrZAeqbNnhhCq86%252BSoA9ZA%3D%3D)

機体台数的に修理が必要であれば，私物対応や代替部品での修理が可能です．

MibuchiYuta, 2022-04-04:
@MasatoKubotera 4/4現在、検証に必要な台数は揃っているため保留にしましょう。

yasuohayashibara, 2022-04-04:
現在4台での戦略チェックを行っています．
林原としては，動くものが4台あれば大丈夫ですが，大抵は直前に動かなくなるロボットがあるので，余裕はあったほうが嬉しいです．

MasatoKubotera, 2022-08-08:
現状，修理Jetsonが瞬断することがありますがその点については下記のissueで扱っています．
https://github.com/citbrains/citbrains_humanoid/issues/1528
修理に関しては終了していますのでcloseします．

closed 2022-08-08 by MasatoKubotera