SUSTAINA-OP/SUSTAINA-OP #297: e-CAM50_CUNXの設置計画

MasatoKubotera, 2022-04-10:
現状のカメラモジュール　シキノハイテックの[KBCR-S02MU](https://www.shikino.co.jp/products/product-kbcr-s02mu.php)の搭載場所に新しいカメラモジュール [e-CAM50_CUNX](https://www.e-consystems.com/jp/nvidia-cameras/jetson-nx-cameras/5mp-mipi-camera-board-jp.asp)搭載すると，画角やモジュールの大きさが異なるため，センターサークルに立った時に足元やゴールポストが見えない．

そのために，カメラの焦点をどこのロボット座標に持っていけばいいのか，どの角度にすれば一番効率が良いのか計画する．

MasatoKubotera, 2022-04-10:
カメラの焦点座標や画角がわかれば，#251 で実機でなくとも視野の予測ができます．

Dansato1203, 2022-04-10:
e-CAM50_CUNXに付属してきたレンズでは垂直視野角が足らないため、e-CAM50_CUNXに対応する視野角が広いレンズを探しています。
現在可能性が一番高いのはユニバース光学工業株式会社の[こちらのページ](http://www.universekk.com/lineup/smount.html)に載っている VL-2228 という製品です。
また、こちらの会社は既存製品のセミカスタム、あるいは仕様に合わせたカスタムレンズも小ロットで製造が可能とのことなので、以前使用していたHPB1033と同じ程度の視野角、F値をもったレンズの製造、販売が可能かを問い合わせ中です。

Dansato1203, 2022-04-10:
カメラを設置する位置の理想ですが、見える画像としては
- 足元のボールが見えること
- センターラインからゴールがほとんど見えること
の2点が重要だと考えます。
参考画像:
[image]
</br>
取り付ける角度を決めるにしてもレンズの持つ視野角次第のため上記のユニバース光学工業株式会社からの返答を待ちたいと考えています。   

3営業日以内に返信とのことなので水曜日まで指標を示せるかは微妙です。

RikuYokoo, 2022-05-11:
購入したレンズとホルダーの設計データ
メーカー名：TOWIN社
レンズ型番：S02512512524F
ホルダー：H0251251812
- [レンズ](https://drive.google.com/file/d/13kYuDL-6OxlT6gKRIAfj5rNTzNaBa07W/view?usp=sharing)
- [ホルダー](https://drive.google.com/file/d/1mZLT52QZjX_1OC8evmoAToiL9tajF4wu/view?usp=sharing)

RikuYokoo, 2022-05-11:
カメラの位置は**仮決定**で以下のようになりました．
レンズが届くのがおそらく来週(5/9の週)になるので届きしだい確認し本決定をしたいと思います．
このように決めた理由としては，171にシキノハイテック製のKBCRS02-muを付けて確認したところ画角的にちょうど良い位置にあり，画角はシキノハイテックのカメラと新しく来るレンズとで1度しか変わらないためその誤差は問題ないと判断しました．そのためレンズの位置が現在と同位置に配置するために画像のように変更します．

[image]

MasatoKubotera, 2022-05-11:
@RikuYokoo 

上記を踏まえて変更しました．
下図のような構成で大丈夫でしょうか．
ご返答お願いします．

[image]
[image]

@KentaTsukino 
切削部品の変更があるのは，GFRP T2 Head_side_frame x2のみになるように調整しました．
@RikuYokoo の確認が取れ次第切削お願いします．
仮確定のようですので，1機体分で期限は5/16 月までにバリ取りまで終えて機体に実装できるようにお願いします．

RikuYokoo, 2022-05-12:
@MasatoKubotera @KentaTsukino 
大丈夫です．
よろしくお願いします．

RikuYokoo, 2022-05-17:
レンズが到着し画角を確認したところ特に問題はなさそうでした。

@MasatoKubotera @KentaTsukino 
以下の画像のように、足元とゴールポストがセンターサークルから見えていることより、特に変更はなくて大丈夫そうです。
よろしくお願いします。

届いたTOWIN社S02512512524F

[image]

defaultのカメラレンズ

[image]

MasatoKubotera, 2022-05-23:
下図のように上記で決まった位置を保ちつつ，フレームの形状を一部変更しました．
これで，SUSTAINA-OP Ver1.0 のカメラ配置を確定させたいと思います．

[image]

MasatoKubotera, 2022-05-23:
配置が決定したのでCloseします．

closed 2022-05-23 by MasatoKubotera