SUSTAINA-OP/SUSTAINA-OP #298: キャリアボード拡張基板の設計

MasatoKubotera, 2022-04-13:
Carrier Board EN715に実装されてい20-Pin GPIOを拡張するための基板を設計する．

必要条件は下記の通りである．

- MPU9250 (SPIもしくはI2C接続)の実装
※接続方法についてはSPI：https://github.com/citbrains/citbrains_humanoid/issues/1620#issuecomment-1009249914 ， I2C： https://github.com/citbrains/citbrains_humanoid/issues/1620#issuecomment-1018707186 を参照．

- USB-HUBの実装
CarrierBoardに最大でWi-Fiドングル，USBカメラ，FT232RL/FT4232Hの3ポート必要である．

- 開発用のUART，I2Cポートコネクタを設置

RikuYokoo, 2022-04-23:
拡張基板でWi-Fiドングル,IMUの動作確認を行いました．

MasatoKubotera, 2022-04-23:
elecrowに発注したときの詳細を下記に記します．

Layer : 2 layers
Dimensions : 87 x 50
PCB Qty : 10
Different PCB Design : 1eg
PCB Thickness : 1.6
PCB Color : Black
Surface Finish : HASL Lead Free
Castellated Hole : No
Copper Weight : 1oz
PCB Stencil : Stencil 15cm X 15cm no frame

[image]

MasatoKubotera, 2022-04-23:
使用する部品リストを下記に示します．
Eagle Num | Q'ty | Parts Name | Name
-- | -- | -- | --
C1, C2, C3 | 3 | [GRM188R6YA106MA73](https://akizukidenshi.com/catalog/g/gP-13161/) | チップ積層セラミックコンデンサー 10μF35V X5R 1608
C4 | 1 | [GRM188B11H103KA01](https://akizukidenshi.com/catalog/g/gP-13387/) | チップ積層セラミックコンデンサー 0.01μF50V B 1608
C5, C6 | 2 | [C1608C0G1H220J080AA](https://www.mouser.jp/ProductDetail/TDK/C1608C0G1H220J080AA?qs=NRhsANhppD%2FvEAPMU6eN6g%3D%3D) | 積層セラミックコンデンサ MLCC - SMD/SMT 0603 50VDC 22pF 5% C0G 0.8mm
CH1 | 1 | [FH-2x10SG](https://akizukidenshi.com/catalog/g/gC-00083/) | ピンソケット　（メス）　２×１０　（２０Ｐ）
CH2, CH3, CH4 | 3 | [S4B-XH-A](https://akizukidenshi.com/catalog/g/gC-12842/) | ＸＨコネクタ　ベース付ポスト　サイド型　４Ｐ　Ｓ４Ｂ－ＸＨ－Ａ（ＬＦ）（ＳＮ）
CR1 | 1 | [FA238V 12.000000MHz](https://akizukidenshi.com/catalog/g/gP-05225/) | 表面実装型クリスタル(水晶発振子) 12MHz
D1, D2, D3, D4, D5, D6 | 6 | [CG0603MLC-05E](https://www.mouser.jp/ProductDetail/Bourns/CG0603MLC-05E?qs=lDok7oSghXVLNKMK69q%252BFQ%3D%3D) | ESDサプレッサ/ TVSダイオード Chip Guard 0603 5 Volts
F1 | 1 | [MF-MSMF050-2](https://www.mouser.jp/ProductDetail/Bourns/MF-MSMF050-2?qs=t3shhpq1i1DZ7OBD5kLNoA%3D%3D) | リセッタブルヒューズ - PPTC 15V .5A-HD 100A MAX
FL1, FL2, FL3 | 3 | [BLM18SD220SN1D](https://www.mouser.jp/ProductDetail/Murata-Electronics/BLM18SD220SN1D?qs=glnLrWX4TPq%2Fm8%2FdPQltLg%3D%3D) | フェライトビーズ 22 OHM
IC1 | 1 | [FE1.1S](https://www.aitendo.com/product/16185) | USBハブコントローラチップ [FE1.1S]
LED1 | 1 | [150060VS86000](https://www.mouser.jp/ProductDetail/Wurth-Elektronik/150060VS86000?qs=GedFDFLaBXGW9qOjzZ1Hmw%3D%3D) | 標準 LED - SMD WL-SMCW TopVw Grn 2V 20mA 573nm
LED2,LED3,LED4 | 3 | [150060YS75000](https://www.mouser.jp/ProductDetail/Wurth-Elektronik/150060YS75000?qs=LlUlMxKIyB0nKmwefHgtZw%3D%3D) | 標準 LED - SMD WL-SMCW SMDMono TpVw Waterclr 0603 Yellow
M1 | 1 | [MPU-9250 Module](https://strawberry-linux.com/catalog/items?code=12250) | MPU-9250 ９軸センサモジュール(３軸加速度＋３軸ジャイロ＋３軸コンパス)
R1, R3, R4, R5 | 4 | [RC0603FR-101KL](https://www.mouser.jp/ProductDetail/YAGEO/RC0603FR-101KL?qs=EiqXWrxQq63MM0wzm69ZKg%3D%3D) | 厚膜抵抗器 - SMD 1 kOhms 100-200mW 0603 1%
R2 | 1 | [RC0603FR-102K7L](https://www.mouser.jp/ProductDetail/YAGEO/RC0603FR-102K7L?qs=Iqe6t0HYRD6Xaw3IgOqXnw%3D%3D) | 厚膜抵抗器 - SMD 2.7 kOhms 100-200 mW 0603 1%
USB1 | 1 | [MUSB-5B-NE-S175](https://akizukidenshi.com/catalog/g/gC-05843/) | 表面実装用USBコネクタ・ミニB・メス
USB2, USB3 | 2 | [USB1046-GF-0190-L-B-A](https://www.mouser.jp/ProductDetail/640-USB1046GF0190LBA) | USBコネクタ USB A Skt, Top Mount, SMT, R/A, GF, Black, With 1.9mm shell stake, T&R

MasatoKubotera, 2022-04-23:
実装後の様子です．

[image]
[image]
[image]

この基板について下記の問題があるため再製作の予定です．

- CarrierBoard EN715とExpantionBoardを接続するL字USBケーブル（USB Type A - Micro USB Mini B）が 干渉する
- StartStopSwitchもUSB type A接続となるため，さらに予備のUSBの追加ポートが必要になる．
- USB・LAN端子と基板が接触ぎりぎりで基板が装着しにくい．

再製作が決定していますが，IMUなどの動作確認のため，MPU9250を直接はんだ実装しました．
再利用できるように少し浮かせています．

CarrierBoardとは，現状ピンソケットをつないで接続しているだけなので，のちにスペーサなどで四隅を固定したいと思います．その時のスペーサ高さは12mmは予定しています．

MasatoKubotera, 2022-04-26:
Elecrowへの再発注時のオプションや選択項目については，下記のコメントと同様です．

https://github.com/citbrains/SUSTAINA-OP/issues/298#issuecomment-1107438466

MasatoKubotera, 2022-04-26:
https://github.com/citbrains/SUSTAINA-OP/issues/298#issuecomment-1107445580
>CarrierBoardとは，現状ピンソケットをつないで接続しているだけなので，のちにスペーサなどで四隅を固定したいと思います．その時のスペーサ高さは12mmは予定しています．

12.5mmに変更したいと思います．

12mmと判断したのは，ExpansionBoardに取り付けるピンソケットとCarrierBoardに実装されている表面実装のピンヘッダを装着したときの実測高さが12mmであったからです．

12.5㎜にしたいというのは，CarrierBoard凸部高さとUSBコネクタの高さを考えた値になります．
USBコネクタの高さ：[USB1046-GF-0190-L-B-A](https://www.mouser.jp/ProductDetail/640-USB1046GF0190LBA) 6.95mm
CarrierBoard 凸部（RTC Battery Connector　Pinrex, 721-94-02TWR9）高さ：5.50mm
※13mmでもよいのではないかと思ってもいる．

この場合，0.5mm程度ExpansionBoardからピンソケットの間に隙間を設けて実装する必要があります．

スペーサについては下記を使用することを検討しています．
[AS-312.5B](https://www.hirosugi-net.co.jp/shop/g/g4377/)

MasatoKubotera, 2022-04-29:
[image]

使用するスペーサを上記で提示した[AS-312.5B](https://www.hirosugi-net.co.jp/shop/g/g4377/)からスパイクでも使用している13mm長の[ASL-313BE](https://www.hirosugi-net.co.jp/shop/g/g13223/)に変更します．

また，コックピットとキャリアボードを接続するスペーサには，[BSL-306.5BE](https://www.hirosugi-net.co.jp/shop/g/g13299/)を使用したいと思います．

※ジュラコンスペーサだと締め付けトルクをほかの部位と同等にするとなめってすぐに使えなくなってしまうので，アルミに変更しました．

MasatoKubotera, 2022-05-05:
ExpansionBoard Ver1.1で必要な，電子部品を下記に示します．

Eagle Num|Q'ty|Parts Name|Name
-- | -- | -- | --
  | 1 | [6604S-40](https://akizukidenshi.com/catalog/g/gP-01591/) | 丸ピンＩＣ用ソケット　（シングル４０Ｐ）　１×４０
  | 1 | [PHA-1x40SG](https://akizukidenshi.com/catalog/g/gC-06631/) | 細ピンヘッダ　１×４０　（黒）
C1, C2, C3 | 3 | [GRM188R6YA106MA73](https://akizukidenshi.com/catalog/g/gP-13161/) | チップ積層セラミックコンデンサー 10μF35V X5R 1608
C4 | 1 | [GRM188B11H103KA01](https://akizukidenshi.com/catalog/g/gP-13387/) | チップ積層セラミックコンデンサー 0.01μF50V B 1608
C5, C6 | 2 | [C1608C0G1H220J080AA](https://www.mouser.jp/ProductDetail/TDK/C1608C0G1H220J080AA?qs=NRhsANhppD%2FvEAPMU6eN6g%3D%3D) | 積層セラミックコンデンサ MLCC - SMD/SMT 0603 50VDC 22pF 5% C0G 0.8mm
CH1 | 1 | [FH-2x10SG](https://akizukidenshi.com/catalog/g/gC-00083/) | ピンソケット　（メス）　２×１０　（２０Ｐ）
CR1 | 1 | [FA238V 12.000000MHz](https://akizukidenshi.com/catalog/g/gP-05225/) | 表面実装型クリスタル(水晶発振子) 12MHz
D1, D2, D3, D4, D5, D6, D7, D8, D9, D10 | 10 | [CG0603MLC-05E](https://www.mouser.jp/ProductDetail/Bourns/CG0603MLC-05E?qs=lDok7oSghXVLNKMK69q%252BFQ%3D%3D) | ESDサプレッサ/ TVSダイオード Chip Guard 0603 5 Volts
F1 | 1 | [MF-MSMF050-2](https://www.mouser.jp/ProductDetail/Bourns/MF-MSMF050-2?qs=t3shhpq1i1DZ7OBD5kLNoA%3D%3D) | リセッタブルヒューズ - PPTC 15V .5A-HD 100A MAX
FL1, FL2, FL3, FL4, FL5 | 5 | [BLM18SD220SN1D](https://www.mouser.jp/ProductDetail/Murata-Electronics/BLM18SD220SN1D?qs=glnLrWX4TPq%2Fm8%2FdPQltLg%3D%3D) | フェライトビーズ 22 OHM
IC1 | 1 | [FE1.1S](https://www.aitendo.com/product/16185) | USBハブコントローラチップ [FE1.1S]
LED1 | 1 | [150060VS86000](https://www.mouser.jp/ProductDetail/Wurth-Elektronik/150060VS86000?qs=GedFDFLaBXGW9qOjzZ1Hmw%3D%3D) | 標準 LED - SMD WL-SMCW TopVw Grn 2V 20mA 573nm
LED2, LED3, LED4, LED5, LED6 | 5 | [150060YS75000](https://www.mouser.jp/ProductDetail/Wurth-Elektronik/150060YS75000?qs=LlUlMxKIyB0nKmwefHgtZw%3D%3D) | 標準 LED - SMD WL-SMCW SMDMono TpVw Waterclr 0603 Yellow
M1 | 1 | [MPU-9250 Module](https://strawberry-linux.com/catalog/items?code=12250) | MPU-9250 ９軸センサモジュール(３軸加速度＋３軸ジャイロ＋３軸コンパス)
R1, R3, R4, R5 | 4 | [RC0603FR-101KL](https://www.mouser.jp/ProductDetail/YAGEO/RC0603FR-101KL?qs=EiqXWrxQq63MM0wzm69ZKg%3D%3D) | 厚膜抵抗器 - SMD 1 kOhms 100-200mW 0603 1%
R2 | 1 | [RC0603FR-102K7L](https://www.mouser.jp/ProductDetail/YAGEO/RC0603FR-102K7L?qs=Iqe6t0HYRD6Xaw3IgOqXnw%3D%3D) | 厚膜抵抗器 - SMD 2.7 kOhms 100-200 mW 0603 1%
USB1 | 1 | [MUSB-5B-NE-S175](https://akizukidenshi.com/catalog/g/gC-05843/) | 表面実装用USBコネクタ・ミニB・メス
USB2, USB3, USB4, USB5 | 4 | [USB1046-GF-0190-L-B-A](https://www.mouser.jp/ProductDetail/640-USB1046GF0190LBA) | USBコネクタ USB A Skt, Top Mount, SMT, R/A, GF, Black, With 1.9mm shell stake, T&R

[6604S-40](https://akizukidenshi.com/catalog/g/gP-01591/)，[PHA-1x40SG](https://akizukidenshi.com/catalog/g/gC-06631/)
上記２つの部品は，[MPU-9250 Module](https://strawberry-linux.com/catalog/items?code=12250)を取り付けるために40P中，10P使用します．
[MPU-9250 Module](https://strawberry-linux.com/catalog/items?code=12250)には購入時に，通常のピンヘッダとピンソケットが付属していますが，これを使用すると高さ的にキャリアボードと接触するために使用します．

MasatoKubotera, 2022-05-05:
実装について

ピンソケット（[FH-2x10SG](https://akizukidenshi.com/catalog/g/gC-00083/)，[6604S-40](https://akizukidenshi.com/catalog/g/gP-01591/)），USB A（[USB1046-GF-0190-L-B-A](https://www.mouser.jp/ProductDetail/640-USB1046GF0190LBA)），[MPU-9250 Module](https://strawberry-linux.com/catalog/items?code=12250)については，リフローで行えないので手はんだで実装する必要があります．

[USB1046-GF-0190-L-B-A](https://www.mouser.jp/ProductDetail/640-USB1046GF0190LBA)がリフローで実装できないのは，スルーホールの凸部が基板厚よりも大きいため，リフローで実装すると浮いて実装されるため避ける必要があります．

MPU9250について，Jetson Xavier NXとI2Cで接続するため，ショートランドをI２C側で４か所ショートさせる必要があります．

[image]
[image]
[image]
[image]
[image]

[image]「ショートランド」

MasatoKubotera, 2022-05-05:
実装（リフロー）の様子　[動画](https://drive.google.com/file/d/1vZQZhSyl96m8tp97vf5kRJtl9XUwW2Zp/view?usp=sharing)

MasatoKubotera, 2022-05-05:
動作確認について

USB Mini B にPCなどを接続して，USB AポートなどにUSBメモリーなどUSB機器を接続して動作すれば機能としては問題ありません．

I2C接続したMPU9250については，下記のwikiに記載されている方法で動作確認する必要があります．
https://github.com/citbrains/citbrains_humanoid/wiki/%E3%82%A4%E3%83%A1%E3%83%BC%E3%82%B8%E6%9B%B8%E3%81%8D%E8%BE%BC%E3%81%BF%E5%BE%8C%E3%80%80%E7%92%B0%E5%A2%83%E6%A7%8B%E7%AF%89%E3%80%90EN715%E3%80%91#IMU

MasatoKubotera, 2022-05-06:
IMUの座標軸に関して

https://github.com/citbrains/citbrains_humanoid/issues/1777#issuecomment-1120057393
にも記載しましたが，転載します．

<blockquote>
SUSTAINA-OP Ver1.0でのIMUの向きに関して

[MPU-9250 ９軸センサモジュール(３軸加速度＋３軸ジャイロ＋３軸コンパス) ](https://strawberry-linux.com/catalog/items?code=12250)を使用しています．

軸方向は，ロボット座標系とセンサの座標系（下図の左側の座標系）が一致するようになっています．

[image]
https://strawberry-linux.com/pub/mpu-9250-manual.pdf
</blockquote>

MasatoKubotera, 2022-05-10:
172機，173機用のExpansionBoard Ver1.1の製作，USB-HUB，IMUの動作確認が終了しました．

MasatoKubotera, 2022-05-10:
上記の製作時，@sekitorimisaki，@TakuyaNagamineと作業しました．

@sekitorimisaki の作成基板については，製作途中．

@TakuyaNagamineの作成基板について，LED4が微妙に点灯し，USB-HUBが使用できない問題があり，FE1.1Sチップを交換しましたが改善しませんでしたので保留扱いにしています．

TakuyaNagamine, 2022-05-12:
実装時の様子　[動画](https://drive.google.com/drive/folders/1HRZ_uV975-Ee89dltmhZ4bun3qLYi11W)

TakuyaNagamine, 2022-05-14:
@sekitorimisakiが作成した基板にも同様の問題が見られたため,確認したところ
CR1 [表面実装型クリスタル(水晶発振子) 12MHz](https://akizukidenshi.com/catalog/g/gP-05225/)が正しく実装されていなかったことが原因だということが分かりました.
修正を加えたところ,どちらの基板も正常に機能しました.

MasatoKubotera, 2022-05-26:
174機，175機分のExpansionBoard Ver1.1 No.4/5のUSB-HUBとIMUの動作確認が取れました．

基板底面に製作日・製作者・管理番号を貼付しています．

No.1~No.3：@MasatoKubotera
No.4：@sekitorimisaki 
No.5：@TakuyaNagamine

[image]「No.3 添付例」

MasatoKubotera, 2022-05-31:
不具合ログ
- [ ] No.4 USB-A Port2のコネクタ部がぐらつく
  ※2本足固定のはんだ量が足りない
- [ ] No5 USB-A Port1のみが使えない
  ※FE1.1Sチップ，USBコネクタのはんだ量が足りない．玉はんだ．

MasatoKubotera, 2022-06-06:
やることリスト

No.6
- [ ] 管理テプラ貼付
- [ ] IMUの動作確認
- [ ] USB-HUBの動作確認(Jetson)
- [ ] ロボットに実装して動作確認

No.7
- [ ] 管理テプラ貼付
- [ ] IMUの動作確認
- [ ] USB-HUBの動作確認(Jetson)
- [ ] ロボットに実装して動作確認

No.8
- [ ] 管理テプラ貼付
- [ ] IMUの動作確認
- [ ] USB-HUBの動作確認(Jetson)

No.9
- [ ] 管理テプラ貼付
- [ ] IMUの動作確認
- [ ] USB-HUBの動作確認(Jetson)

No.10
- [ ] 管理テプラ貼付
- [ ] IMUの基板実装
- [ ] IMUの動作確認
- [ ] USB-HUBの動作確認(Jetson)

MasatoKubotera, 2022-06-07:
製作過程を[Youtube](https://youtu.be/xLqRejphQpI)にアップしました．

MasatoKubotera, 2022-06-08:
@hiroki-0001 
SPI/I2Cを選択するショートランドが未接続でした．再度検証お願いします．

hiroki-0001, 2022-06-08:
No.9の動作確認を行いましたが、IMUは動作しませんでした。
MPU9250を交換したところ、No.9でも動作したので、**No.9に搭載されたMPU9250が悪そうです**

MasatoKubotera, 2022-06-08:
No.9の[MPU9250](https://strawberry-linux.com/catalog/items?code=20948)について通信不良による故障と判断します．

No.10のMPU9250が使用できているので，そちらをNo.9に実装し，No.10はMPU9250を実装しない予備基板としたいと思います．

MasatoKubotera, 2022-06-27:
設計・製作・動作確認まで終わっているためCloseします．

closed 2022-06-27 by MasatoKubotera