SUSTAINA-OP/SUSTAINA-OP #302: 足関節が硬いことによる対応

MasatoKubotera, 2022-05-02:


MasatoKubotera, 2022-05-02:
prototypeからVer1.0にかけて変更した箇所の影響により，
モータを介さずとも関節が硬くなっていることが確認されている．

prototypeからの関節回りの変更は下記の3つである．
- GFRPの変更 https://github.com/citbrains/SUSTAINA-OP/issues/62#issuecomment-1021098528
- ワッシャの塗装なしへ変更 https://github.com/citbrains/SUSTAINA-OP/issues/255#issuecomment-1051106640
- 軸スペーサの特注 https://github.com/citbrains/SUSTAINA-OP/issues/255#issuecomment-1049902362 https://github.com/citbrains/SUSTAINA-OP/issues/261

MibuchiYuta, 2022-05-12:
[交換前の様子](https://drive.google.com/file/d/1vCyloaMMo9ygKU8uBivMBhqv_HDlSbva/view?usp=drivesdk)
[交換後の様子](https://drive.google.com/file/d/1efcKUWXwr33kjIcTOfkUKXsMWLGtWEBT/view?usp=drivesdk)

yasuohayashibara, 2022-05-13:
ワッシャですが，同じ形状で厚さが異なるものがあるとしたら何か対策したほうが良いと思います．
おそらく十分に把握していない方がメンテナンスした場合，間違った厚さのワッシャを入れるおそれがありますので，何か対策をしておいたほうが良いと思います．
（複数ある場合はすべてを緩い方に合わせても良いかもしれません．）

混じるおそれがない場合は無視していただいて構いません．

MasatoKubotera, 2022-05-14:
@yasuohayashibara 

ワッシャについて，同外径・内径で厚さ違いのものを使用しています．

UUW-0614-03H：0.3mm厚　24枚/機
UUW-0614-05H：0.5mm厚　  4枚/機

厚みの違いは数量の触ればわかる程度で危ういため，数量の少ない[UUW-0614-05H](https://www.hirosugi-net.co.jp/shop/g/g59176/)を同一材質の[UUW-0614-05BH](https://www.hirosugi-net.co.jp/shop/g/g59960/)に変更して対応したいと思います．

[image]「UUW-0614-05H」

[image]「UUW-0614-05BH」

[image]

MasatoKubotera, 2022-05-24:
対応が完了し，モデルも修正したのでcloseします．

closed 2022-05-24 by MasatoKubotera