SUSTAINA-OP/SUSTAINA-OP #305: 世界大会(RoboCup2022)に持っていくよびB3M

hiroki-0001, 2022-05-04:


MasatoKubotera, 2022-05-06:
新品のサーボモータはhttps://github.com/citbrains/SUSTAINA-OP/issues/269#issue-1143097732 に示されているように下記の事項を追加して行ってください．

- 出力軸ネジの交換
- ケース固定ネジを抜く

サーボIDは０，バウンドレートは実機と合わせおくと大会当日の故障したときの対応がスムーズになります．
ID0はID設定をせずに機体に搭載してしまってもIDがほかのものと被ることがなく対応することができ，万が一hrした時も故障リスクをなくすことができます．

hiroki-0001, 2022-05-14:
https://github.com/citbrains/SUSTAINA-OP/issues/248#issuecomment-1126633640
でも報告させていただきましたが

ガタの調査方法を記載したマニュアルを作成しました。
https://docs.google.com/document/d/1iAS3x8LAxV03nM-gmYImkfOl_cfmT6Za8mNcNyWYxik/edit?usp=sharing

こちらのマニュアルを見ながら作業を問題なく行えるか確認してほしいです。

hiroki-0001, 2022-05-18:
出力の計測方法のやり方を記載したマニュアルも完成しました。
https://docs.google.com/document/d/10HbfmLSczo-6LDLIbtwnOn-_tKc3f8HeFCe6b0getcQ/edit?usp=sharing

こちらのマニュアルも見ながら作業を問題なく行えるか確認してほしいです。

KentaTsukino, 2022-05-26:
5/25
ガタの調査をしました
行ったサーボ
40-0001
40-0002
40-0003
40-0004
40-0005
40-2220
40-2221
40-2222
40-2223
40-2224
40-2225

MasatoKubotera, 2022-05-26:
世界大会用の新品サーボモータについては下記の管理番号のものになります．
確認漏れを防ぐためにリストを作成しました．
スプレッドシートに登録が終わり次第，チェックを付けてください．

**SC-1040**　計10個
- [ ] 40-0001
- [ ] 40-0002
- [ ] 40-0003
- [ ] 40-0004
- [ ] 40-0005
- [ ] 40-0006
- [ ] 40-0007
- [ ] 40-0008
- [ ] 40-0009
- [ ] 40-0010

**SC-1170**　計10個
- [ ] 70-2121
- [ ] 70-2122
- [ ] 70-2123
- [ ] 70-2124
- [ ] 70-2125
- [ ] 70-0006
- [ ] 70-0007
- [ ] 70-0008
- [ ] 70-0009
- [ ] 70-0010

hiroki-0001, 2022-06-01:
https://github.com/citbrains/SUSTAINA-OP/issues/305#issuecomment-1138972132
のB3Mの出力及びガタの計測が終了しました。予備用の20個のB3Mすべて**使用可能**です

hiroki-0001, 2022-06-01:
> 新品でないものはこちらの要項も確認してください．
https://github.com/citbrains/SUSTAINA-OP/issues/309#issuecomment-1141981228

こちら現在在庫にあるものを確認しました。初期位置はすべて0°になるようにしておきました。

MasatoKubotera, 2022-06-12:
新規サーボを導入するため，確認物品が増えましたのでreopenします．

MasatoKubotera, 2022-06-12:
新規導入B3Mは，B3M-SC-1040　×20個，B3M-SC-1170　×20個です．

管理番号は，
B3M-SC-1040 : 40-2226～40-2245
B3M-SC-1170 : 70-2201～70-2220
とします．

これにあたって，管理番号シール(テプラ)を作成する必要があります．
@MibuchiYuta 作成お願いします．

hiroki-0001, 2022-06-14:
新規購入したB3Mが到着しました。
以下のタスクの処理をお願いします。

- [ ] B3Mの管理番号シールの貼り付け
- [ ] 管理システムへの新規登録
- [ ] 取付用のねじ穴にネジが挿入されている場合は取り除く
- [ ] ClampHorn，B3Mのねじ穴がつぶれていないかバカになっていないか
- [ ] ClampHorn，B3Mのねじ穴がつぶれていないかバカになっていないか
- [ ] ClampHornネジを六角穴付フランジボルトの[620103008](https://jp.misumi-ec.com/vona2/detail/221000862966/?PNSearch=620103008&HissuCode=620103008)に交換する

KiyoshiroKawanabe, 2022-06-17:
管理番号，40-2230～40-2245，70-2201～70-2220のガタ計測行いました．
70-2209のみガタ異常が見られました．

yasuohayashibara, 2022-06-17:
> 70-2209のみガタ異常が見られました．

新品であればデータとともに林原までお願いします．

hiroki-0001, 2022-06-18:
@yasuohayashibara 

70-2209は新品になります。
ガタデータはこちらになります

| 0°　ガタ | 45°　ガタ | 90°　ガタ | -45°　ガタ | -90°　ガタ | 
| -------- | --------- | --------- | ---------- | ---------- | 
| 0.51     | 0.35      | 0.27      | 0.58       | 0.74       | 

比較用に正常の新品サーボのガタのデータも載せておきます

| 0°　ガタ | 45°　ガタ | 90°　ガタ | -45°　ガタ | -90°　ガタ | 
| -------- | --------- | --------- | ---------- | ---------- | 
| 0.16     | 0.16      | 0.16      | 0.16       | 0.18       |

KiyoshiroKawanabe, 2022-06-18:
管理番号，40-2230～40-2245，70-2201～70-2208，70-2210～70-2220の出力計測行いました．
問題のあるサーボはありませんでした．

Keita1Mori, 2022-06-25:
世界大会に持っていく予備サーボのダブルチェックを行い問題ありませんでした。
管理番号，70-2201～70-2208，70-2210～70-2220のID:0, パラメータ, バウンドレートの設定＆チェック終わりました。

Keita1Mori, 2022-06-25:
管理番号，40-2240～40-2245のID:0, パラメータ, バウンドレートの設定＆チェック終わりました。
明日残りの40をチェックします。

Keita1Mori, 2022-06-26:
管理番号，40-2230～40-2239のID:0, パラメータ, バウンドレートの設定＆チェック終わりました。

MasatoKubotera, 2022-07-11:
準備が終了していますのでCloseします．

closed 2022-07-11 by MasatoKubotera