SUSTAINA-OP/SUSTAINA-OP #306: LI-HV_Batteryの3セル間の電圧の誤差を無くす

TakuyaNagamine, 2022-05-05:
現在GanKenKunやSUSTAINA-OPで使用されているバッテリー「LI-HV_Battery」の3セル間の電圧差に0.05V以上(CIT Brainsでは各セルの電圧差が0.05V以上の場合,異常電圧と判定している)の誤差が生じている.
原因としてはバランス・ストレージ充電を行った際の充電スピードが影響していると考えられるので現状の充電速度から速度を落として充電を行っていく.

TakuyaNagamine, 2022-05-05:
## 検証1
### ・検証方法
現在LI-HV_Batteryを充電する際の設定は以下の通りである
2.8A 11.4V
今回の検証では充電速度を落とすことを目的として,電流値を2.8Aから1.0Aに設定を変更しバランス充電,ストレージ充電を1セットとして異常電圧でなくなるまで行う.
今回検証で使用するバッテリーは22-30C-02,03,04,05
### ・検証結果
充電前
・02   ①3.952V ②3.946V ③3.947V(最大電圧差0.06Vで異常電圧)
・03   ①3.947V ②3.938V ③3.946V(最大電圧差0.09Vで異常電圧)
・04   ①3.896V ②3.886V ③3.901V(最大電圧差0.15Vで異常電圧)
・05   ①3.960V ②3.941V ③3.967V(最大電圧差0.26Vで異常電圧)
1セット目
・02   ①3.904V ②3.908V ③3.909V(最大電圧差0.05Vとなり**異常電圧ではないが危険な値ではあるので引き続き検証を行う**)
・03   ①3.908V ②3.910V ③3.917V(最大電圧差0.09Vで異常電圧)
・04   ①3.901V ②3.903V ③3.911V(最大電圧差0.10Vで異常電圧)
・05   ①3.908V ②3.910V ③3.917V(最大電圧差0.09Vで異常電圧)
2セット目
・02   ①3.902V ②3.905V ③3.907V(最大電圧差0.05Vとなり**異常電圧でなくなった**)
・03   ①3.907V ②3.907V ③3.906V(最大電圧差0.01Vとなり**異常電圧でなくなった**)
・04   ①3.902V ②3.903V ③3.906V(最大電圧差0.04Vとなり**異常電圧でなくなった**)
・05   ①3.904V ②3.908V ③3.909V(最大電圧差0.05Vとなり**異常電圧でなくなった**)
バランス充電,ストレージ充電を2セット行った結果,4つのバッテリーは異常電圧外となったため2日ほど放置し,電圧差の変動を見る.
2日後
・02   ①3.900V ②3.900V ③3.903V(最大電圧差0.03Vで**異常電圧外**)
・03   ①3.902V ②3.905V ③3.906V(最大電圧差0.04Vで**異常電圧外**)
・04   ①3.898V ②3.901V ③3.900V(最大電圧差0.03Vで**異常電圧外**)
・05   ①3.900V ②3.902V ③3.905V(最大電圧差0.05Vで**異常電圧外**)
### ・まとめ
今回の検証結果からバランス充電,ストレージ充電を2セット行った結果,4つのバッテリーは異常電圧外となり,保管後の電圧差も異常電圧外の値を保つことができていた.

MasatoKubotera, 2022-06-21:
作業が終了しているのでCloseします．

closed 2022-06-21 by MasatoKubotera