SUSTAINA-OP/SUSTAINA-OP #307: 一定期間運用したサーボの損耗を調査する

MibuchiYuta, 2022-05-05:


yasuohayashibara, 2022-05-05:
こちらロボットを安定して動かすためには重要な知見になると思います．
ちょっと話題が変わるかもしれませんが，どの関節で何時間利用したかのデータを自動的に収集する仕組みを作れれば精度の高い故障予測ができるのですが，何か良いアイディアがありましたらお願いします．
（サーボモータのどこかにシリアル番号を埋め込めれば，自動的にデータ収集できると思います．）

MibuchiYuta, 2022-05-15:
5月末を目処に170に対して行う予定です。 @MasatoKubotera との話で以下の状態を確認する予定となりました。
- ワッシャ
- ブッシュ
- サーボ(ホーン含め)
- 段付きボルト

MasatoKubotera, 2022-06-03:
チェックリスト

ガタ調査
- [ ] ID1
- [ ] ID2
- [ ] ID3
- [ ] ID4
- [ ] ID5
- [ ] ID6
- [ ] ID7
- [ ] ID8
- [ ] ID9
- [ ] ID10
- [ ] ID11
- [ ] ID12
- [ ] ID13
- [ ] ID14
- [ ] ID15
- [ ] ID16
- [ ] ID17
- [ ] ID18
- [ ] ID19

出力評価
- [ ] ID1
- [ ] ID2
- [ ] ID3
- [ ] ID4
- [ ] ID5
- [ ] ID6
- [ ] ID7
- [ ] ID8
- [ ] ID9
- [ ] ID10
- [ ] ID11
- [ ] ID12
- [ ] ID13
- [ ] ID14
- [ ] ID15
- [ ] ID16
- [ ] ID17
- [ ] ID18
- [ ] ID19

MasatoKubotera, 2022-06-25:
調査は終了しており，[B3M評価計測 シート](https://docs.google.com/spreadsheets/d/1ytKEkW5EvP1CdU5-lgqlxm7ZQpeUXrU9aAGpEB-B7Rg/edit?usp=sharing)に記録が残っていますので参照してください．

closed 2022-06-25 by MasatoKubotera