SUSTAINA-OP/SUSTAINA-OP #309: サーボモータのIDとパラメータ書き込み方法

MasatoKubotera, 2022-05-06:


MasatoKubotera, 2022-05-07:
# サーボの書き込みについて

## 使用するサーボモータについて

SUSTAINA-OP Ver1.0では，下記のサーボモータを使用しています．

ＩＤ１　（右　足首Ｒ）：B3M-SC-1040-A
ＩＤ２　（右　足首Ｐ）：B3M-SC-1170-A
ＩＤ３　（右　膝下Ｐ）：B3M-SC-1170-A
ＩＤ４　（右　膝上Ｐ）：B3M-SC-1170-A
ＩＤ５　（右　付根Ｒ）：B3M-SC-1170-A
ＩＤ６　（右　付根Ｙ）：B3M-SC-1170-A
ＩＤ７　（右　肩　Ｐ）：B3M-SC-1040-A
ＩＤ８　（右　肩　Ｒ）：B3M-SC-1040-A
ＩＤ９　（右　肘　Ｐ）：B3M-SC-1040-A

ＩＤ１０（左　足首Ｒ）：B3M-SC-1040-A
ＩＤ１１（左　足首Ｐ）：B3M-SC-1170-A
ＩＤ１２（左　膝下Ｐ）：B3M-SC-1170-A
ＩＤ１３（左　膝上Ｐ）：B3M-SC-1170-A
ＩＤ１４（左　付根Ｒ）：B3M-SC-1170-A
ＩＤ１５（左　付根Ｙ）：B3M-SC-1170-A
ＩＤ１６（左　肩　Ｐ）：B3M-SC-1040-A
ＩＤ１７（左　肩　Ｒ）：B3M-SC-1040-A
ＩＤ１８（左　肘　Ｐ）：B3M-SC-1040-A

ＩＤ１９（首　Ｙ）：B3M-SC-1040-A or B3M-SB-1040-A
※SCシリーズの使用を推奨しているが，在庫状況によりSBシリーズを使用している．重さが６ｇ差があるが，そのほかのトルクや速度などは負荷がかかる場所ではないため無視できる．

参照：
[B3M-SB-1040-A](https://kondo-robot.com/product/03072)
[B3M-SC-1040-A](https://kondo-robot.com/product/03090)
[B3M-SC-1170-A](https://kondo-robot.com/product/03092)

## パラメータについて

パラメータについては，[citbrains_humanoid/wiki/](https://github.com/citbrains/citbrains_humanoid/wiki/B3M%E3%80%80%E3%83%91%E3%83%A9%E3%83%A1%E3%83%BC%E3%82%BF%E8%A8%AD%E5%AE%9A)に記載されている．

<blockquote>

## 共通パラメータ
- Pos Minmum = -32000[deg]
- Pos Maximum = 32000[deg]
- Baudrate  
    - SUSTAINA = 3000000[bps]
    - GankenKun = 1000000[bps]

- PWM Frequency = 8000[Hz]
- Dead Band = 15[deg]
- PWM Limit = 100[%]

**【注意】**
- Baudrate (通信速度)
- PWM Frequency (PWM制御の周波数)

**これらのパラメータに関しては設定をROMに保存し、再起動しないと設定が反映されない**

## 各種ゲインのパラメータ

### SC-1040
| PRESET No | Kp | Kd | Ki | Static Friction | Dynamic Fiction |
| :---------: | ------: | ----: | -----: | -----: | -----: |
| 0    | 35000 | 500 | 1800 | 0 | 0 |
| 1    | 5500  | 100 | 580  | 100 | 50 |
| 2    | 100000 | 500 | 10000 | 100 | 50 |

### SC-1170
| PRESET No | Kp | Kd | Ki | Static Friction | Dynamic Fiction |
| :---------: | ------: | ----: | -----: | -----: | -----: |
| 0    | 42000 | 400 | 1000 | 0 | 0 |
| 1    | 0 | 0 | 0 | 0 | 0 |
| 2    | 42000 | 750 | 6000 | 0 | 50 |

</blockquote>

上記の共通のパラメータの他に，SUSTAINA-OP Ver1.0 では，初期設定角度（Pose  Initializerでのsetangle，B3M ManagerでのCenter Position [deg]）がすべて0で直立姿勢になるようになっている．そのため，Center Position [deg]が0になっていることを確認する必要があります．

上記に提示されているパラメータ以外は，動作に関係がないパラメータとしている．

## 書き込み方法について

1. [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってID１～１９とBaudrate1,000,000[bps]を書き込む

※ SUSTAINA Ver1.0 動作時のBaudrate3,000,000[bps]ですが，B3MServoCheckerの仕様上Baudrate1,000,000[bps]でしかパラメータを書き込めない．
参照：https://github.com/citbrains/B3MServoChecker/issues/3

2. [B3MServoChecker](https://github.com/citbrains/B3MServoChecker)を使ってparameterを書き込む．
3.  [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってBaudrate3,000,000[bps]を書き込む．B3MServoCheckerでパラメータが書き込めているか確認する．

※Baudrate3,000,000[bps]を書き込むことができるのは，現状KONDO純正の[RS485USB/シリアル変換アダプター](https://kondo-robot.com/product/02133)に限ります．

**現状Baudrate3,000,000[bps]としていますが，今後のエラーレートを踏まえて変更する可能性があります．**

MasatoKubotera, 2022-05-08:
# SUSTAINA Ver1.0用の作業リスト

- [ ] 171
  - [ ] 選定（管理システムに登録）
  - [ ] 取付用のねじ穴にネジが挿入されている場合は取り除く
  - [ ] ClampHorn，B3Mのねじ穴がつぶれていないかバカになっていないか
  - [ ] ClampHornネジを六角穴付フランジボルトの[620103008](https://jp.misumi-ec.com/vona2/detail/221000862966/?PNSearch=620103008&HissuCode=620103008)に交換する（ネジ材塗布）
  - [ ] サーボモータ本体にIDをラベル付けする
  - [ ] [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってID1~19とBaudrate1,000,000[bps]を書き込む
  - [ ] [B3MServoChecker](https://github.com/citbrains/B3MServoChecker)でパラメータの書き込み
  - [ ] [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってBaudrate3,000,000[bps]を書き込む．B3MServoCheckerでパラメータが書き込めているか確認する．
---
- [ ] 172
  - [ ] 選定（管理システムに登録）
  - [ ] 取付用のねじ穴にネジが挿入されている場合は取り除く
  - [ ] ClampHorn，B3Mのねじ穴がつぶれていないかバカになっていないか
  - [ ] ClampHornネジを六角穴付フランジボルトの[620103008](https://jp.misumi-ec.com/vona2/detail/221000862966/?PNSearch=620103008&HissuCode=620103008)に交換する（ネジ材塗布）
  - [ ] サーボモータ本体にIDをラベル付けする
  - [ ] [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってID1~19とBaudrate1,000,000[bps]を書き込む
  - [ ] [B3MServoChecker](https://github.com/citbrains/B3MServoChecker)でパラメータの書き込み
  - [ ] [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってBaudrate3,000,000[bps]を書き込む．B3MServoCheckerでパラメータが書き込めているか確認する．
---
- [ ] 173
  - [ ] 選定（管理システムに登録）
  - [ ] 取付用のねじ穴にネジが挿入されている場合は取り除く
  - [ ] ClampHorn，B3Mのねじ穴がつぶれていないかバカになっていないか
  - [ ] ClampHornネジを六角穴付フランジボルトの[620103008](https://jp.misumi-ec.com/vona2/detail/221000862966/?PNSearch=620103008&HissuCode=620103008)に交換する（ネジ材塗布）
  - [ ] サーボモータ本体にIDをラベル付けする
  - [ ] [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってID1~19とBaudrate1,000,000[bps]を書き込む
  - [ ] [B3MServoChecker](https://github.com/citbrains/B3MServoChecker)でパラメータの書き込み
  - [ ] [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってBaudrate3,000,000[bps]を書き込む．B3MServoCheckerでパラメータが書き込めているか確認する．
---
- [ ] 174
  - [ ] 選定（管理システムに登録）
  - [ ] 取付用のねじ穴にネジが挿入されている場合は取り除く
  - [ ] ClampHorn，B3Mのねじ穴がつぶれていないかバカになっていないか
  - [ ] ClampHornネジを六角穴付フランジボルトの[620103008](https://jp.misumi-ec.com/vona2/detail/221000862966/?PNSearch=620103008&HissuCode=620103008)に交換する（ネジ材塗布）
  - [ ] サーボモータ本体にIDをラベル付けする
  - [ ] [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってID1~19とBaudrate1,000,000[bps]を書き込む
  - [ ] [B3MServoChecker](https://github.com/citbrains/B3MServoChecker)でパラメータの書き込み
  - [ ] [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってBaudrate3,000,000[bps]を書き込む．B3MServoCheckerでパラメータが書き込めているか確認する．
---
- [ ] 175
  - [ ] 選定（管理システムに登録）
  - [ ] 取付用のねじ穴にネジが挿入されている場合は取り除く
  - [ ] ClampHorn，B3Mのねじ穴がつぶれていないかバカになっていないか
  - [ ] ClampHornネジを六角穴付フランジボルトの[620103008](https://jp.misumi-ec.com/vona2/detail/221000862966/?PNSearch=620103008&HissuCode=620103008)に交換する（ネジ材塗布）
  - [ ] サーボモータ本体にIDをラベル付けする
  - [ ] [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってID1~19とBaudrate1,000,000[bps]を書き込む
  - [ ] [B3MServoChecker](https://github.com/citbrains/B3MServoChecker)でパラメータの書き込み
  - [ ] [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってBaudrate3,000,000[bps]を書き込む．B3MServoCheckerでパラメータが書き込めているか確認する．
---
- [ ] 176
  - [ ] 選定（管理システムに登録）
  - [ ] 取付用のねじ穴にネジが挿入されている場合は取り除く
  - [ ] ClampHorn，B3Mのねじ穴がつぶれていないかバカになっていないか
  - [ ] ClampHornネジを六角穴付フランジボルトの[620103008](https://jp.misumi-ec.com/vona2/detail/221000862966/?PNSearch=620103008&HissuCode=620103008)に交換する（ネジ材塗布）
  - [ ] サーボモータ本体にIDをラベル付けする
  - [ ] [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってID1~19とBaudrate1,000,000[bps]を書き込む
  - [ ] [B3MServoChecker](https://github.com/citbrains/B3MServoChecker)でパラメータの書き込み
  - [ ] [B3M Manager](https://kondo-robot.com/faq/b3m-manager-software-2)を使ってBaudrate3,000,000[bps]を書き込む．B3MServoCheckerでパラメータが書き込めているか確認する．

Keita1Mori, 2022-05-20:
# 報告
以前１７３機体ID６番ように用意したサーボ70-1976で通信エラーを起こしていたので新しいサーボを野口先輩に用意してもらいました。
[故障記録](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit#gid=372255127
)

70-1976 → 70-2003
[機体番号による抽出](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit#gid=824152206)

MasatoKubotera, 2022-05-21:
[B3Mサーボモータのリバースモード機能の廃止 #1787](https://github.com/citbrains/citbrains_humanoid/issues/1787)について，RoboCup2022 Thailandまでにリバースモードが廃止とならないため，リバースモードについても書き込む必要がある．

リバースモードを書き込むのは，GankenKun Ver3.0，SUSTAINA-OP Ver1.0共通でID３，４，７，１２，１３，１６の6個/機体である．

MasatoKubotera, 2022-05-31:
# 【重要】

位置制御において角度＝０度のときに，出力ホーンが既定の位置に来ない場合がある．

CentorPotision = 0度　になっていてもズレる場合は，出力ホーンとファイナルギアの組み合わせ位置を変える必要がある．

TakuyaNagamine, 2022-06-13:
## サーボパラメーター確認チェックリスト
  - [ ] 171
  - [ ] 172
  - [ ] 173
  - [ ] 174
  - [ ] 175
  - [ ] 176

MasatoKubotera, 2022-06-25:
SUSTAINA Ver1.0 171~176に使用するサーボの準備は終了したのでCloseします．

closed 2022-06-25 by MasatoKubotera