SUSTAINA-OP/SUSTAINA-OP #310: SUSTAINA用ゼッケン作成の為のレザーカッター技術の確立

KentaTsukino, 2022-05-07:
SUSUTAINAのゼッケン作成の為にレザーカッターを使用したい
Inventorのスケッチだとレザーカッターを使用するとき、スケッチのつなぎ目で止まってしまうので、それを解消することが目的。

MasatoKubotera, 2022-05-08:
@KentaTsukino 

ゼッケン用のDXFファイルを作成して上げておきました．
        UpperBody/Arm/ForeArm_outside_cloth.dxf
        UpperBody/Cockpit/Cockpit_back_frame_cloth.dxf
        UpperBody/Cockpit/Cockpit_front_frame_cloth.dxf

DXFのエクスポート方法は，下図のように面を右クリックすると，「面に名前を付けてエクスポート」からエクスポートできます．
ファイル形式は，AutoCAD DXFです．

[image]

「CUT2D」でパスが切れているか確認しましたが，切れている様子が確認できませんでした．
[image]

「AutoCAD」を使って結合することもできそうですがもしかしたらパスの結合作業は不要かもしれません．
https://knowledge.autodesk.com/ja/support/autocad/learn-explore/caas/CloudHelp/cloudhelp/2016/JPN/AutoCAD-Core/files/GUID-2E6DCB79-28F1-4BF8-AADC-2F16F85744F9-htm.html

MasatoKubotera, 2022-05-10:
新習志野キャンパス11号館 工作センタにあるレーザーカッターは[レーザー加工機 HAJIME CL1 PLUS](https://www.oh-laser.com/products/hajime)でした．

駆動用のソフトウェアは，[Oh-Laserダウンロードサイト](https://dl.oh-laser.jp/index.php)から[一括インストール](https://dl.oh-laser.jp/HARUKA-20220214-0.96.12/HARUKA-AllinOne.exe)でインストールできるHARUKAを使用します．

インストール後，レーザー加工機 HAJIME CL1 PLUSとUSB接続するだけで簡単に駆動させることができます．

上記のコメントで提唱していたパスの結合作業は不要で，用意していたDXFファイルで切断が可能でした．

ソフトウェア側の注意点として，DXFではパスの順番を選択できないようで内側→外側など選べるようになっていないようです．

また，複数のDXFをインポートして連続運転することはできません．（複数のDXFファイルを並べることができない）
そのため，複数の部品を一度に切断したい場合は，1つのDXFファイルでインポートする必要がありそうです．

KentaTsukino, 2022-05-13:
# 途中
### 1. レザーカッターに電源を入れる
HARUKAを使わないと付けられないようになっているので、工場の人にコネクターなどをもらう。
付ける場所は白い方の機械の裏にある
[image]  [image]

### 2. 水を入れる
白い奴と黒い奴にそれぞれ水を入れるが、ペットボトルを使う。白いのには蓋がついているので取る
[image][image]
[image][image]

KentaTsukino, 2022-05-13:
# レザーカッターの黒い方を壊した　
## 作業者
@KentaTsukino  @NaokiIwasawa 
## 
5/10
レザーカッターの検証が終わり、片付けの為に黒い方の水を排出するとこを本来回すはずのない六角の所を回して取ってしまい、戻らなくなった
[image] [image]

KentaTsukino, 2022-05-16:
@MasatoKubotera 

5/16
津田沼にあるレーザーカッターはEpilog Fusion 32 75Wで、Illustratorしか対応していませんでした
あと、出力が高いのかPPはきれいに加工できないと言われました
その代案として、水流カッターを勧められました
ですが現在壊れていて水曜日(5/18)に修理するらしいので、詳しくは木曜日以降に行ったときになります
津田沼にある水流カッターはDXFファイルで動き、操作自体も難しくないとの事でした

MasatoKubotera, 2022-06-01:
ゼッケン用データ（.dxf）

**ForeArm_outside_cloth** x2枚/機体
https://github.com/citbrains/SUSTAINA-OP/blob/master/UpperBody/Arm/ForeArm_outside_cloth.dxf

**Cockpit_front_frame_cloth_No1~8**  x1枚/機体
https://github.com/citbrains/SUSTAINA-OP/blob/master/UpperBody/Cockpit/Cockpit_front_frame_cloth_No1.dxf
https://github.com/citbrains/SUSTAINA-OP/blob/master/UpperBody/Cockpit/Cockpit_front_frame_cloth_No2.dxf
https://github.com/citbrains/SUSTAINA-OP/blob/master/UpperBody/Cockpit/Cockpit_front_frame_cloth_No3.dxf
https://github.com/citbrains/SUSTAINA-OP/blob/master/UpperBody/Cockpit/Cockpit_front_frame_cloth_No4.dxf
https://github.com/citbrains/SUSTAINA-OP/blob/master/UpperBody/Cockpit/Cockpit_front_frame_cloth_No5.dxf
https://github.com/citbrains/SUSTAINA-OP/blob/master/UpperBody/Cockpit/Cockpit_front_frame_cloth_No6.dxf
https://github.com/citbrains/SUSTAINA-OP/blob/master/UpperBody/Cockpit/Cockpit_front_frame_cloth_No7.dxf
https://github.com/citbrains/SUSTAINA-OP/blob/master/UpperBody/Cockpit/Cockpit_front_frame_cloth_No8.dxf

**Cockpit_back_frame_cloth**  x1枚/機体
https://github.com/citbrains/SUSTAINA-OP/blob/master/UpperBody/Cockpit/Cockpit_back_frame_cloth.dxf

MasatoKubotera, 2022-06-01:
製作チェックリスト（機体実装用）

## 171
- [ ] ForeArm_outside_cloth(赤) x2枚
- [ ] Cockpit_front_frame_cloth_No1(赤)  x1枚
- [ ] Cockpit_back_frame_cloth(赤)  x1枚
- [ ] ForeArm_outside_cloth(青) x2枚
- [ ] Cockpit_front_frame_cloth_No1(青)  x1枚
- [ ] Cockpit_back_frame_cloth(青)  x1枚

## 172
- [ ] ForeArm_outside_cloth(赤) x2枚
- [ ] Cockpit_front_frame_cloth_No2(赤)  x1枚
- [ ] Cockpit_back_frame_cloth(赤)  x1枚
- [ ] ForeArm_outside_cloth(青) x2枚
- [ ] Cockpit_front_frame_cloth_No2(青)  x1枚
- [ ] Cockpit_back_frame_cloth(青)  x1枚

## 173
- [ ] ForeArm_outside_cloth(赤) x2枚
- [ ] Cockpit_front_frame_cloth_No3(赤)  x1枚
- [ ] Cockpit_back_frame_cloth(赤)  x1枚
- [ ] ForeArm_outside_cloth(青) x2枚
- [ ] Cockpit_front_frame_cloth_No3(青)  x1枚
- [ ] Cockpit_back_frame_cloth(青)  x1枚

## 174
- [ ] ForeArm_outside_cloth(赤) x2枚
- [ ] Cockpit_front_frame_cloth_No4(赤)  x1枚
- [ ] Cockpit_back_frame_cloth(赤)  x1枚
- [ ] ForeArm_outside_cloth(青) x2枚
- [ ] Cockpit_front_frame_cloth_No4(青)  x1枚
- [ ] Cockpit_back_frame_cloth(青)  x1枚

## 175
- [ ] ForeArm_outside_cloth(赤) x2枚
- [ ] Cockpit_front_frame_cloth_No5(赤)  x1枚
- [ ] Cockpit_back_frame_cloth(赤)  x1枚
- [ ] ForeArm_outside_cloth(青) x2枚
- [ ] Cockpit_front_frame_cloth_No5(青)  x1枚
- [ ] Cockpit_back_frame_cloth(青)  x1枚

## 176
- [ ] ForeArm_outside_cloth(赤) x2枚
- [ ] Cockpit_front_frame_cloth_No6(赤)  x1枚
- [ ] Cockpit_back_frame_cloth(赤)  x1枚
- [ ] ForeArm_outside_cloth(青) x2枚
- [ ] Cockpit_front_frame_cloth_No6(青)  x1枚
- [ ] Cockpit_back_frame_cloth(青)  x1枚

## 177 (予備機体)
- [ ] ForeArm_outside_cloth(赤) x2枚
- [ ] Cockpit_front_frame_cloth_No7(赤)  x1枚
- [ ] Cockpit_back_frame_cloth(赤)  x1枚
- [ ] ForeArm_outside_cloth(青) x2枚
- [ ] Cockpit_front_frame_cloth_No7(青)  x1枚
- [ ] Cockpit_back_frame_cloth(青)  x1枚

MasatoKubotera, 2022-06-13:
## 発注

世界大会予備として，下記の製作をお願いします．
 - [ ] ForeArm_outside_cloth(赤) x2枚
 - [ ] Cockpit_front_frame_cloth_No7(赤) x1枚
 - [ ] Cockpit_front_frame_cloth_No7(赤) x1枚

 - [ ] Cockpit_back_frame_cloth(赤) x1枚
 - [ ] ForeArm_outside_cloth(青) x2枚
 - [ ] Cockpit_front_frame_cloth_No7(青) x1枚
 - [ ] Cockpit_back_frame_cloth(青) x1枚

IkuoShige, 2022-06-17:
@IkuoShige と @HirokiHasegawa777 でレーザーカッターの[マニュアル(ドキュメント)](https://docs.google.com/document/d/1AcNu-jnLQG2p36R0GTy4FbnFDLDkTZMZH3aMVPyjGP0/edit#)を作成しています
あと、足りていない写真を三枚ほど挿入して完成となります

MasatoKubotera, 2022-06-20:
お待たせいたしました．PPクラフトシート 黒（スポンサー添付用）の2Dモデルを作成しました．
製作のほどお願いします．
@IkuoShige @HirokiHasegawa777 

## PPクラフトシート 黒

- [ ] Trunk_chest_frame_cover_sheet x7枚 (1枚/機)
- [ ] FrontLink_diagonal_cover_sheet x28枚 (4枚/機)

MasatoKubotera, 2022-06-21:
レーザーカッターで使用する2Dデータについては，[SUSTAINA-OP](https://github.com/citbrains/SUSTAINA-OP)/LaserCutter/ に追加していきます．

IkuoShige, 2022-06-24:
@MasatoKubotera @KentaTsukino 
レーザーカッターの[マニュアル(ドキュメント)](https://docs.google.com/document/d/1AcNu-jnLQG2p36R0GTy4FbnFDLDkTZMZH3aMVPyjGP0/edit#)が完成しました。
確認お願いしたいです。

MasatoKubotera, 2022-06-25:
@IkuoShige @HirokiHasegawa777 

戦略側から要望がありました首後方に取り付ける機体番号マーキング用シートを作成しました．
[Head_back_cloth.dxf](https://github.com/citbrains/SUSTAINA-OP/blob/master/LaserCutter/Head_back_cloth.dxf)

[image]

下記の数量をカットお願いします．

- [ ] 橙色 PPクラフトシート T0.75 Head_back_cloth ×5
- [ ] 黒色 PPクラフトシート T0.75 Head_back_cloth ×14

MasatoKubotera, 2022-07-11:
技術確立のためのマニュアルの作成・ゼッケンの作成が終了しているためCloseします．

closed 2022-07-11 by MasatoKubotera