SUSTAINA-OP/SUSTAINA-OP #312: サーボモータの交換タスク

Keita1Mori, 2022-05-12:


Keita1Mori, 2022-05-12:
# 報告

現状あるGankenKunのサーボからSC40を確保するためにSB40に変えるタスクである。

---
## 165機体
ID:9
SC 40-1913 → SB 40-1908
交換済み

---
## 167機体
ID:8
SC 40-1914 → SB 40-1912
交換済み

ID:17
SC 40-1948 → SB 40-1910
交換済み

SC 40-1913, SC 40-1914, SC 40-1948は新B3M管理システムで在庫登録済みです。

Keita1Mori, 2022-05-13:
# 3/13の作業報告
昨日取り出したSCサーボの40-1913, 40-1914ですが今日野口先輩と一緒に出力軸とガタの調査を行いました。

## 結果
## 40-1914
0 deg
backlash：0.16

45deg
backlash：0.26

90deg
backlash：0.26

-45deg
backlash：0.18

-90deg
backlash：0.09

出力順
0.04A：221.111
0.06A：153.905

出力逆
0.04A：228.938
0.06A：163.956

---

## 40-1913
0deg
backlash：0.16

45deg
backlash：0.17

90deg
backlash：0.18

-45deg
backlash：0.26

-90deg
backlash：0.17

出力順
0.04A：242.021
0.06A：172.470

出力逆
0.04A：242.706
0.06A：175.759

---

上記内容は[B3M 評価計測 入力フォーム](https://docs.google.com/forms/d/e/1FAIpQLSfG6C2SnpBKjR3XnGXMD15MVS8d1qMKrj4wrJnNCsGt4dekHQ/viewform)に登録済みです。

Keita1Mori, 2022-05-14:
# 使用可能の有無
[新_B3M管理シート](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit#gid=765323291)内にある”型番による抽出”のシートを確認したところ使用可能でした。

hiroki-0001, 2022-05-24:
@Keita1Mori
 40-1948は調査を行いましたか？行っているなら評価結果を教えて欲しいです

Keita1Mori, 2022-05-24:
@hiroki-0001 申し訳ないですが、まだ行っていない状態です。

Keita1Mori, 2022-05-31:
こちらに関してですが @TakuyaNagamine が計測をして大丈夫とのことで174機体に搭載済みです。
ID：10

closed 2022-06-01 by MasatoKubotera