SUSTAINA-OP/SUSTAINA-OP #314: e-CAM50_CUNX カメラレンズ換装

MasatoKubotera, 2022-05-16:


MasatoKubotera, 2022-05-16:
カメラレンズ換装のほかに，カメラを取り付けるためのネジ・スペーサの取り付けも同時に行う．

その時に使う既製品は下記の4つである．
- CSHHB-SUS-M2-15 (廃版・在庫品)ｘ４個
- [C-2006-4](https://www.hirosugi-net.co.jp/shop/g/g5920/)ｘ４個
- [C-2002.5-4](https://www.hirosugi-net.co.jp/shop/g/g5913/)ｘ４個
- M2ナットｘ４個 ※機体に搭載はしない．保管時にネジ抜けを抑えるために使用

RikuYokoo, 2022-05-17:
レンズホルダーはカメラと合わなかったためもともとついていたものを使用．
カメラ7個，レンズの交換，スペーサ等の取り付けが完了しました．

RikuYokoo, 2022-05-17:
もともとついていたカメラレンズ
[FIFO-03028B8M](https://drive.google.com/file/d/11HB2Y_tcRjT_WATWErCewJJDsw_2MNqN/view?usp=sharing)
ホルダー
S-Mount Lens Holder
ホルダーについて名称がどこにもなくこのような記載になってしまいました．

MasatoKubotera, 2022-05-20:
https://github.com/citbrains/SUSTAINA-OP/issues/314#issuecomment-1127630593
でコメントした，スペーサの取り付けについて下図のように取り付けます．

[image]
[image]

使用するスペーサ・ネジについては下図のようにInventorモデルにも反映しています．
[image]

MasatoKubotera, 2022-05-20:
取り替えるときの注意事項として下記の項目を口頭で伝えしました．

1. レンズ交換時にはカメラ内部へほこりなど入らないようにカメラを下向きにしてレンズホルダから外す．
2. カメラを下向きにした状態で、交換したいレンズを装着する．

ほこりが付着した場合などは[エアダスター ](https://www.amazon.co.jp/gp/product/B005TH1HU6)などを使って除去する．

レンズに指を触れてしまった場合，[無水エタノール](https://www.amazon.co.jp/dp/B000TKDKA8)と[キムワイプ](https://www.amazon.co.jp/dp/B001EHI9XI?th=1)を使って拭きとる．

MasatoKubotera, 2022-05-20:
https://github.com/citbrains/SUSTAINA-OP/issues/314#issuecomment-1128710618
<blockquote>

もともとついていたカメラレンズ
[FIFO-03028B8M](https://drive.google.com/file/d/11HB2Y_tcRjT_WATWErCewJJDsw_2MNqN/view?usp=sharing)
ホルダー
S-Mount Lens Holder
ホルダーについて名称がどこにもなくこのような記載になってしまいました．

</blockquote>

交換したカメラのレンズは[TOWIN](https://www.towinlens.com/) S02512512524Fです．

MasatoKubotera, 2022-05-22:
レンズ換装後
[image]

MasatoKubotera, 2022-05-22:
作業が終了しているのでCloseします．

closed 2022-05-22 by MasatoKubotera