SUSTAINA-OP/SUSTAINA-OP #315: e-CAM50_CUNXが発熱する問題の対策

RikuYokoo, 2022-05-16:
e-CAM50_CUNXを付けた状態でjetsonを起動すると発熱し、カメラのプログラムを動かすとさらに発熱するという問題がある。
販売元のe-con systemsにも問い合わせているところであるが、
熱を持ちながら動作させることは故障の原因に繋がるので、ファンなどを取り付け対策を講じる。

MasatoKubotera, 2022-05-17:
参考：[e-con Systemsのcsiカメラe-CAM50_CUNXが発熱する問題 #1767](https://github.com/citbrains/citbrains_humanoid/issues/1767)

MasatoKubotera, 2022-05-20:
[ＤＣファン　１２Ｖ　６０×６０×１５ｍｍ](https://akizukidenshi.com/catalog/g/gP-16827/)をカメラ背面に取り付けて様子見したいと思います．

[image]

MasatoKubotera, 2022-05-20:
このような感じで実装したいと考えています．
[image]

MasatoKubotera, 2022-06-18:
e-CAM50_CUNXはヒートシンクとセットで使用することが推奨されているようです．
https://github.com/citbrains/citbrains_humanoid/issues/1767#issuecomment-1151898253

RikuYokoo, 2022-06-18:
https://github.com/citbrains/citbrains_humanoid/issues/1767#issuecomment-1159334742
こちらについて
メールでおすすめ？のヒートシンクと取り付け場所を聞いてみようと思います．

MasatoKubotera, 2022-06-25:
ヒートシンクについて

@RikuYokoo がメーカ側に問い合わせところ推奨のヒートシンクや取付方法についてはなく，ユーザ側で決める事項とのこと．

MasatoKubotera, 2022-07-11:
機体を夜から朝まで電源が入っていたことがあったようですが，そのあとも問題なく動作しているようです．
発熱に関しては，現段階では大きな問題と捉えずに大会を迎えました．
対策については，今後の課題としてCloseします．

closed 2022-07-11 by MasatoKubotera