SUSTAINA-OP/SUSTAINA-OP #316: 修理から戻ってきたサーボの状態確認

MasatoKubotera, 2022-05-20:
対象は2022/4/15，18に修理依頼した計29個．

SC-1040
- [ ] 40-1916
- [ ] 40-1922
- [ ] 40-1927
- [ ] 40-1933
- [ ] 40-1945
- [ ] 40-1949
- [ ] 40-1956
- [ ] 40-1959
- [ ] 40-2108

SC-1170
- [ ] 70-1914
- [ ] 70-1919
- [ ] 70-1921
- [ ] 70-1922
- [ ] 70-1923
- [ ] 70-1927
- [ ] 70-1928
- [ ] 70-1929
- [ ] 70-1939
- [ ] 70-1940
- [ ] 70-1972
- [ ] 70-1979
- [ ] 70-1982
- [ ] 70-1987
- [ ] 70-1988
- [ ] 70-2002
- [ ] 70-2005
- [ ] 70-2008
- [ ] 70-2110
- [ ] 70-2113

MasatoKubotera, 2022-05-20:
確認内容

下記を要項を行ってください．すべての工程が終わり次第，上記コメントの該当管理番号にチェックを入れてください．

ガタ調査，パウダーブレーキを用いたサーボの評価については，[B3M 評価計測 入力フォーム](https://docs.google.com/forms/d/e/1FAIpQLSfG6C2SnpBKjR3XnGXMD15MVS8d1qMKrj4wrJnNCsGt4dekHQ/viewform)にてシステムに登録を行ってください．

**1. ガタ調査**

  参考 : 
   - https://github.com/citbrains/SUSTAINA-OP/issues/248
   - [220430_B3Mサーボモータガタの計測方法](https://docs.google.com/document/d/1iAS3x8LAxV03nM-gmYImkfOl_cfmT6Za8mNcNyWYxik/edit?usp=sharing)

SC-1040
- [ ] 40-1916
- [ ] 40-1922
- [ ] 40-1927
- [ ] 40-1933
- [ ] 40-1945
- [ ] 40-1949
- [ ] 40-1956
- [ ] 40-1959
- [ ] 40-2108

SC-1170
- [ ] 70-1914
- [ ] 70-1919
- [ ] 70-1921
- [ ] 70-1922
- [ ] 70-1923
- [ ] 70-1927
- [ ] 70-1928
- [ ] 70-1929
- [ ] 70-1939
- [ ] 70-1940
- [ ] 70-1972
- [ ] 70-1979
- [ ] 70-1982
- [ ] 70-1987
- [ ] 70-1988
- [ ] 70-2002
- [ ] 70-2005
- [ ] 70-2008
- [ ] 70-2110
- [ ] 70-2113

**2. パウダーブレーキを用いたサーボの評価**

  参考 : 
   - https://github.com/citbrains/SUSTAINA-OP/issues/228
   - [220306_B3MSeries 一定負荷による回転速度を用いたモータ性能評価方法](https://docs.google.com/document/d/10HbfmLSczo-6LDLIbtwnOn-_tKc3f8HeFCe6b0getcQ/edit?usp=sharing)
 
SC-1040
- [ ] 40-1916
- [ ] 40-1922
- [ ] 40-1927
- [ ] 40-1933
- [ ] 40-1945
- [ ] 40-1949
- [ ] 40-1956
- [ ] 40-1959
- [ ] 40-2108

SC-1170
- [ ] 70-1914
- [ ] 70-1919
- [ ] 70-1921
- [ ] 70-1922
- [ ] 70-1923
- [ ] 70-1927
- [ ] 70-1928
- [ ] 70-1929
- [ ] 70-1939
- [ ] 70-1940
- [ ] 70-1972
- [ ] 70-1979
- [ ] 70-1982
- [ ] 70-1987
- [ ] 70-1988
- [ ] 70-2002
- [ ] 70-2005
- [ ] 70-2008
- [ ] 70-2110
- [ ] 70-2113

yasuohayashibara, 2022-06-02:
後で１個サンプルを下さい．

MasatoKubotera, 2022-06-27:
角度情報読み込みの誤差が大きいB3Mに関しては下記のissueで引き継ぎます．
https://github.com/citbrains/SUSTAINA-OP/issues/325

作業が終了していますのでcloseします．