SUSTAINA-OP/SUSTAINA-OP #317: B3Mサーボモータの選定方法の共有

hiroki-0001, 2022-05-21:
出力評価・ガタ評価をした際の情報はこの[スプレッドシート](https://docs.google.com/spreadsheets/d/1ytKEkW5EvP1CdU5-lgqlxm7ZQpeUXrU9aAGpEB-B7Rg/edit?usp=sharing)にまとめられている。
これらの情報をもとに機体に使用するb3mを選定するための方法を記載する。

hiroki-0001, 2022-05-22:
B3Mサーボモータのガタが大きいと歩行・モーションに大きな影響を与える。
(このガタに関する議論はhttps://github.com/citbrains/SUSTAINA-OP/issues/248#issuecomment-1094766010 でされました)

そのため、特定の部位にはガタの少ないB3Mサーボモータを使用する必要がある。
それらをまとめたのが以下の表1になる。

[表1]

| SC-1170                  | SC-1040                | 
| ------------------------ | ---------------------- | 
| ☆　足首ピッチ軸 　=　2  | 肩・肘ピッチ軸　=　4   | 
| 膝・股ピッチ軸　=　４    | 肩ロール軸　=　2       | 
| ☆　付け根ロール軸　=　2 | 首ヨー軸　=　1         | 
| 付け根ヨー軸　=　2       | ☆　足首ロール軸　=　2 | 

**☆…ガタの影響を大きく受ける部位**

B3Mのガタの評価は新品のB3Mサーボモータの指標データの差分から3種類に分けられる。

- ガタの量が指標の2倍以下　→　『使用可能』
- ガタの量が指標の2倍以上　3倍以下　→　『限定使用可能』
- ガタの量が指標の3倍以上　→　『ガタ異常』

これらの評価されたB3Mは表1の各部位ごとに

☆印がついていないB3Mは『限定使用可』  
☆印がついているB3Mは『使用可能』

を使用する

hiroki-0001, 2022-05-23:
B3Mのガタの評価などの情報は[『B3M評価計測 シート』](https://docs.google.com/spreadsheets/d/1ytKEkW5EvP1CdU5-lgqlxm7ZQpeUXrU9aAGpEB-B7Rg/edit?usp=sharing)及び[『新_B3M管理シート』](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit?usp=sharing)でいずれかで確認することができる。

[『B3M評価計測 シート』](https://docs.google.com/spreadsheets/d/1ytKEkW5EvP1CdU5-lgqlxm7ZQpeUXrU9aAGpEB-B7Rg/edit?usp=sharing)のほうが更新が早く、またガタの詳細なデータ等も確認する事ができるので、選定の際にはこちらを利用したほうがよい。

【[『B3M評価計測 シート』](https://docs.google.com/spreadsheets/d/1ytKEkW5EvP1CdU5-lgqlxm7ZQpeUXrU9aAGpEB-B7Rg/edit?usp=sharing)でのガタ・出力情報の表示】

[image] 

【[『B3M評価計測 シート』](https://docs.google.com/spreadsheets/d/1ytKEkW5EvP1CdU5-lgqlxm7ZQpeUXrU9aAGpEB-B7Rg/edit?usp=sharing)でのガタの詳細な状態の表示】

[image]

MasatoKubotera, 2022-05-23:
https://github.com/citbrains/SUSTAINA-OP/issues/317#issuecomment-1133924819 について初心者でもわかるようにまとめました．

ID | 部位名称 | 種類 | ガタ影響
-- | -- | -- | --
1，10 | 足首　ロール軸 | B3M-SC-1170-A | あり
2，11 | 足首　ピッチ軸 | B3M-SC-1040-A | あり
3，12 | 膝下　ピッチ軸 | B3M-SC-1170-A |  
4，13 | 膝上　ピッチ軸 | B3M-SC-1170-A |  
5，14 | 付け根　ロール軸 | B3M-SC-1170-A | あり
6，15 | 付け根　ヨー軸 | B3M-SC-1170-A |  
7，16 | 肩　ピッチ軸 | B3M-SC-1040-A |  
8，17 | 肩　ロール軸 | B3M-SC-1040-A |  
9，18 | 肘　ピッチ軸 | B3M-SC-1040-A |  
19 | 首　ヨー軸 | B3M-SC-1040-A |  

ガタが少ないモータは，**B3M-SC-1040-A：2個/機**，**B3M-SC-1170-A：4個/機** 必要．
この部分には，[『B3M評価計測 シート』](https://docs.google.com/spreadsheets/d/1ytKEkW5EvP1CdU5-lgqlxm7ZQpeUXrU9aAGpEB-B7Rg/edit?usp=sharing)または[『新_B3M管理シート』](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit?usp=sharing)で確認できる **ガタの量が指標の2倍以下である『使用可能』** のものを選定する．

[image]

MasatoKubotera, 2022-06-21:
共有が終了しているのでcloseします．

closed 2022-06-21 by MasatoKubotera