SUSTAINA-OP/SUSTAINA-OP #319: offset_angleがおかしい

MibuchiYuta, 2022-05-31:


MibuchiYuta, 2022-05-31:
@MasatoKubotera 
172
```
0       0       FOOT_ROLL_R
1       0       LEG_PITCH_R
2       0       KNEE_R1
3       0       KNEE_R2
4       0       LEG_ROLL_R
5       0       LEG_YAW_R
6       0       ARM_PITCH_R
7       0       ARM_ROLL_R
8       0       ELBOW_PITCH_R
9       0       FOOT_ROLL_L
10      0       LEG_PITCH_L
11      0       KNEE_L1
12      0       KNEE_L2
13      -100    LEG_ROLL_L
14      -100    LEG_YAW_L
15      0       ARM_PITCH_L
16      0       ARM_ROLL_L
17      0       ELBOW_PITCH_L
18      -900    HEAD_YAW
19      0       HEAD_PITCH
20      0       SPARE12
21      0       SPARE13
22      0       SPARE16
23      0       SPARE20
24      0       SPARE24
25      0       SPARE25
26      0       SPARE26
27      0       SPARE27
28      0       SPARE28
```

MibuchiYuta, 2022-05-31:
@MasatoKubotera 
172
```
0       0       FOOT_ROLL_R
1       0       LEG_PITCH_R
2       0       KNEE_R1
3       0       KNEE_R2
4       0       LEG_ROLL_R
5       0       LEG_YAW_R
6       0       ARM_PITCH_R
7       0       ARM_ROLL_R
8       0       ELBOW_PITCH_R
9       0       FOOT_ROLL_L
10      -1500   LEG_PITCH_L
11      0       KNEE_L1
12      0       KNEE_L2
13      0       LEG_ROLL_L
14      0       LEG_YAW_L
15      0       ARM_PITCH_L
16      0       ARM_ROLL_L
17      0       ELBOW_PITCH_L
18      -8500   HEAD_YAW
19      0       HEAD_PITCH
20      0       SPARE12
21      0       SPARE13
22      0       SPARE16
23      0       SPARE20
24      0       SPARE24
25      0       SPARE25
26      0       SPARE26
27      0       SPARE27
28      0       SPARE28
```

MibuchiYuta, 2022-05-31:
@MasatoKubotera 
173
```
0       0       FOOT_ROLL_R
1       -9000   LEG_PITCH_R
2       9000    KNEE_R1
3       0       KNEE_R2
4       0       LEG_ROLL_R
5       500     LEG_YAW_R
6       0       ARM_PITCH_R
7       0       ARM_ROLL_R
8       0       ELBOW_PITCH_R
9       0       FOOT_ROLL_L
10      0       LEG_PITCH_L
11      0       KNEE_L1
12      0       KNEE_L2
13      0       LEG_ROLL_L
14      0       LEG_YAW_L
15      0       ARM_PITCH_L
16      0       ARM_ROLL_L
17      0       ELBOW_PITCH_L
18      0       HEAD_YAW
19      0       HEAD_PITCH
20      0       SPARE12
21      0       SPARE13
22      0       SPARE16
23      0       SPARE20
24      0       SPARE24
25      0       SPARE25
26      0       SPARE26
27      0       SPARE27
28      0       SPARE28
```

MibuchiYuta, 2022-05-31:
@MasatoKubotera 
174
```
0       0       FOOT_ROLL_R
1       -1500   LEG_PITCH_R
2       0       KNEE_R1
3       0       KNEE_R2
4       0       LEG_ROLL_R
5       1500    LEG_YAW_R
6       9000    ARM_PITCH_R
7       0       ARM_ROLL_R
8       0       ELBOW_PITCH_R
9       0       FOOT_ROLL_L
10      0       LEG_PITCH_L
11      0       KNEE_L1
12      0       KNEE_L2
13      0       LEG_ROLL_L
14      0       LEG_YAW_L
15      0       ARM_PITCH_L
16      0       ARM_ROLL_L
17      0       ELBOW_PITCH_L
18      0       HEAD_YAW
19      0       HEAD_PITCH
20      0       SPARE12
21      0       SPARE13
22      0       SPARE16
23      0       SPARE20
24      0       SPARE24
25      0       SPARE25
26      0       SPARE26
27      0       SPARE27
28      0       SPARE28
```

MasatoKubotera, 2022-06-03:
offset_angleはJetson Xavier NXなどのコンピュータボード側で行う初期位置調整である．

上記のoffset_angleは，1列目がソフトウェア的なサーボID（B3M側のID x-1の値：0→ID1），2列目が調整値（0.01deg：9000=90deg），3列目が関節名称である．

この初期位置調整は，ハードウェア的に誤差がなければすべて0になるはずであるが，実際ガタや切削精度，組み立て精度により0になることは少ない．しかし，本来であればその誤差は数度であり，微調節のみのはずであるが上記の報告ではそうなっていない．

考えられる要因としては，https://github.com/citbrains/SUSTAINA-OP/issues/281 などで出力軸ホーン(ClampHorn)を取り出したときに本来の位置に戻さなかったこと，組み立て時の出力軸ホーン(ClampHorn)を規定位置基準として組み立てなかったことである．

ファイナルギアと出力軸ホーン(ClampHorn)はそれぞれ24歯であり，1歯取付を間違えると，360度÷24=15度ずれる．
[image]

これらを考慮すると対処が必要な個所は下記の通りである．
- [ ] 172：<code>10      -1500   LEG_PITCH_L</code> ※出力軸ホーン(ClampHorn)の再取り付け
- [ ] 172：<code>18      -8500   HEAD_YAW</code>※出力軸ホーン(ClampHorn)の再取り付け
- [ ] 173：<code>1       -9000   LEG_PITCH_R</code> ※組み立てミス
- [ ] 173：<code>2       9000    KNEE_R1</code> ※組み立てミス
- [ ] 174：<code>1       -1500   LEG_PITCH_R</code>※出力軸ホーン(ClampHorn)の再取り付け
- [ ] 174：<code>5       1500    LEG_YAW_R</code>※出力軸ホーン(ClampHorn)の再取り付け
- [ ] 174：<code>6       9000    ARM_PITCH_R</code> ※組み立てミス

MasatoKubotera, 2022-06-04:
@MibuchiYuta 
上記の問題に対してハードウェア的に対応しました．
最終的なoffset_angleをここに添付してください．

基本的にoffset_angleで対応するのは数度のずれとしたいと思います．

MasatoKubotera, 2022-06-05:
下記の箇所についてoffset_angleの箇所を見落としていました．順次対応します．

- [ ] 171 : <code>18      -900    HEAD_YAW</code> ※出力軸ホーン(ClampHorn)の再取り付け

MibuchiYuta, 2022-06-05:
## 171
```
0	0	FOOT_ROLL_R
1	0	LEG_PITCH_R
2	0	KNEE_R1
3	0	KNEE_R2
4	100	LEG_ROLL_R
5	0	LEG_YAW_R
6	0	ARM_PITCH_R
7	0	ARM_ROLL_R
8	0	ELBOW_PITCH_R
9	-200	FOOT_ROLL_L
10	0	LEG_PITCH_L
11	0	KNEE_L1
12	0	KNEE_L2
13	-400	LEG_ROLL_L
14	-100	LEG_YAW_L
15	0	ARM_PITCH_L
16	0	ARM_ROLL_L
17	0	ELBOW_PITCH_L
18	-900	HEAD_YAW
19	0	HEAD_PITCH
20	0	SPARE12
21	0	SPARE13
22	0	SPARE16
23	0	SPARE20
24	0	SPARE24
25	0	SPARE25
26	0	SPARE26
27	0	SPARE27
28	0	SPARE28
```

MibuchiYuta, 2022-06-05:
## 172
```
0	0	FOOT_ROLL_R
1	0	LEG_PITCH_R
2	0	KNEE_R1
3	0	KNEE_R2
4	0	LEG_ROLL_R
5	0	LEG_YAW_R
6	0	ARM_PITCH_R
7	0	ARM_ROLL_R
8	0	ELBOW_PITCH_R
9	0	FOOT_ROLL_L
10	0	LEG_PITCH_L
11	0	KNEE_L1
12	0	KNEE_L2
13	0	LEG_ROLL_L
14	0	LEG_YAW_L
15	0	ARM_PITCH_L
16	0	ARM_ROLL_L
17	0	ELBOW_PITCH_L
18	0	HEAD_YAW
19	0	HEAD_PITCH
20	0	SPARE12
21	0	SPARE13
22	0	SPARE16
23	0	SPARE20
24	0	SPARE24
25	0	SPARE25
26	0	SPARE26
27	0	SPARE27
28	0	SPARE28
```

MibuchiYuta, 2022-06-05:
## 173
```
0	-200	FOOT_ROLL_R
1	0	LEG_PITCH_R
2	0	KNEE_R1
3	0	KNEE_R2
4	0	LEG_ROLL_R
5	200	LEG_YAW_R
6	0	ARM_PITCH_R
7	-300	ARM_ROLL_R
8	0	ELBOW_PITCH_R
9	0	FOOT_ROLL_L
10	0	LEG_PITCH_L
11	0	KNEE_L1
12	0	KNEE_L2
13	100	LEG_ROLL_L
14	0	LEG_YAW_L
15	600	ARM_PITCH_L
16	0	ARM_ROLL_L
17	0	ELBOW_PITCH_L
18	0	HEAD_YAW
19	0	HEAD_PITCH
20	0	SPARE12
21	0	SPARE13
22	0	SPARE16
23	0	SPARE20
24	0	SPARE24
25	0	SPARE25
26	0	SPARE26
27	0	SPARE27
28	0	SPARE28
```

MibuchiYuta, 2022-06-05:
## 174
```
0	0	FOOT_ROLL_R
1	0	LEG_PITCH_R
2	0	KNEE_R1
3	0	KNEE_R2
4	100	LEG_ROLL_R
5	0	LEG_YAW_R
6	0	ARM_PITCH_R
7	0	ARM_ROLL_R
8	0	ELBOW_PITCH_R
9	0	FOOT_ROLL_L
10	-400	LEG_PITCH_L
11	0	KNEE_L1
12	0	KNEE_L2
13	-100	LEG_ROLL_L
14	0	LEG_YAW_L
15	0	ARM_PITCH_L
16	0	ARM_ROLL_L
17	0	ELBOW_PITCH_L
18	0	HEAD_YAW
19	0	HEAD_PITCH
20	0	SPARE12
21	0	SPARE13
22	0	SPARE16
23	0	SPARE20
24	0	SPARE24
25	0	SPARE25
26	0	SPARE26
27	0	SPARE27
28	0	SPARE28
```

MibuchiYuta, 2022-06-05:
## SUSTAINAのoffset_angleの読書感想文
きれいだとおもいました

MasatoKubotera, 2022-06-06:
171機のoffset_angle中の

<code>18      -900    HEAD_YAW</code>

について，https://github.com/citbrains/SUSTAINA-OP/issues/319#issuecomment-1145545735 に記載した通り１歯ズレるごとに15度ズレるため，ClampHornが1歯分（15度）ずれていて＋6度が取や切削精度による誤差とも考えられます．

取付時の誤差というのは，ガバ穴が基本的にネジ径の+0.2mmであるため，その分の誤差になります．

MasatoKubotera, 2022-06-06:
大きな問題は解決しましたのでcloseします．
175機，176機を量産したときに同様な問題が出た時はreopenしてください．

closed 2022-06-06 by MasatoKubotera