SUSTAINA-OP/SUSTAINA-OP #320: サーボモータ本体の固定ネジによる

MasatoKubotera, 2022-06-04:
[image]

MasatoKubotera, 2022-06-04:
機体については，転ぶことなく歩行していたため10分間動き続けていた．そのため，このサーボモータについてギア欠け，サーボの出力低下などが起きている可能性がある．

MasatoKubotera, 2022-06-04:
模擬試合中の様子 [YouTube](https://youtu.be/JzgJQcrpwPk)
[image]

MasatoKubotera, 2022-06-05:
該当のネジを取り出した状態で，2022/6/5に模擬試合を行ったが，目視では問題が確認できなかった．

世界大会用機体であるため，対象のサーボモータ**70-2111**について取り出して下記の検証を行う．

- [ ] ガタ調査
- [ ] 出力評価

@hiroki-0001 @KiyoshiroKawanabe 対応お願いします．

MasatoKubotera, 2022-06-05:
抜けたネジについて，下図に示しますが代替品がないので抜けたままになっています．
抜けた状態での動作には問題ありません．

[image]

hiroki-0001, 2022-06-07:
対象のサーボモータ70-2111のガタを調査しました。以下の表の用になりました

|          | 0°　ガタ | +45°　ガタ | +90°　ガタ | -45°　ガタ | -90°　ガタ | 
| -------- | -------- | ---------- | ---------- | ---------- | ---------- | 
| 問題　前 | 0.26     | 0.16       | 0.2        | 0.21       | 0.24       | 
| 問題　後 | 0.57     | 0.26       | 0.27       | 0.34       | 0.27       | 

初期位置付近のガタが大きくなっていることが判明しました。また、初期位置(0°)付近で回転させると、内部に欠けたギアが入ってしまったせいか、回転がカタつきます

MasatoKubotera, 2022-06-07:
サーボについて交換対応を取りました．交換先は70-0004です．

交換後hrして動作確認まで行いました．

MasatoKubotera, 2022-06-08:
対応が終了したので，closeします．

MasatoKubotera, 2022-06-13:
同箇所で問題が再発したのでreopenします．

MasatoKubotera, 2022-06-13:
2022/6/12の模擬試合において，172機 ID11（左足首ピッチ）のサーボモータにおいてサーボ本体の固定ネジが飛び出す方法が再発した．前回と同じ場所であり，増し締めして搭載していたがネジが緩み飛び出した．

[image]
[image]

飛び出したサーボモータ70-2111についは@hiroki-0001 の判断で70-0005に交換した．

MasatoKubotera, 2022-06-13:
ネジの増し締めだけでは再発が防止できなかったので，下図に示すネジについてはすべてのサーボモータを対象に取り外して搭載することとしたい．

[image]

yasuohayashibara, 2022-06-13:
干渉する場所を除いて，取り外したところにもともと入っていたネジを入れるようにしたほうが良いと思います．

hiroki-0001, 2022-06-14:
https://github.com/citbrains/SUSTAINA-OP/issues/320#issuecomment-1153631583 の交換に関してはガタ・出力等には問題ありませんでしたが、https://github.com/citbrains/SUSTAINA-OP/issues/325 で取り上げられた「12.10Vでシャットダウンするように安定化電源がセットされていたが落ちる」現象が発生し、駆動回路に異常が見られたので交換しました。

closed 2022-06-27 by MasatoKubotera