SUSTAINA-OP/SUSTAINA-OP #321: Mainboard ver 2.2no

MibuchiYuta, 2022-06-05:


MibuchiYuta, 2022-06-05:
## 電圧計の値がバラバラである
ロボットに同一の電源を挿しても表示される電圧は各機体によって変わっています。

  | 171 | 172 | 173 | 174
-- | -- | -- | -- | --
全てoff | 11.9 | 11.1 | 11.2 | 11.4
Jetson起動中 | 11.8 | 11.1 | 11.1 | 11.2
hr中 | 11.4 | 10.8 | 10.8 | 10.9

使用した電源はこちらです
SW Power-02
12.09V

MibuchiYuta, 2022-06-05:
## 電源の切り替えがうまくいかない機体がある
173,174機体にて6/4の試合前のバッテリー交換時に発覚しました。電源を落とし、スイッチング電源側のコネクタにバッテリを挿して使いました

MasatoKubotera, 2022-06-06:
各機体に搭載されている基板の管理番号は下記の通りです．

- 171 : No.0(動作確認用の試作，カラーコネクタ未対応)
- 172 : No.2
- 173 : No.3
- 174 : No.4

MasatoKubotera, 2022-06-06:
使用している部品はコネクタ以外すべて同一です．

MibuchiYuta, 2022-06-09:
> @MasatoKubotera が作成したMainboard ver2.2を搭載したロボット4台で模擬試合を行ったところ、多くの問題が見つかったためこちらに報告します。
> 
> 
> 
> - [ ] 電圧計の値がバラバラである
> - [ ] 172電源の切り替えがうまくいかない
> 
> - [ ] 173電源の切り替えがうまくいかない
> 
> - [ ] 174電源の切り替えがうまくいかない

@MasatoKubotera 172も電源切り替えに問題がありました。

TakuyaNagamine, 2022-06-15:
172機にスイッチング電源を接続したところ表示される電圧が9V台となっていた
[image]

MasatoKubotera, 2022-06-15:
<blockquote>
172機にスイッチング電源を接続したところ表示される電圧が9V台となっていた
[image]
</blockquote>

について，スイッチング電源並びにサーボモータの印加電圧は12Ｖ付近であったので電圧計の故障と判断し新しいものに交換した．
故障原因については判明していない．

KANBE8810, 2022-06-20:
@MasatoKubotera 
# 機体に搭載されている電圧計の誤差調査
6/20(月)時点で動作確認できる172，173，174の各機体に搭載されている電圧計の表示されている値と実際の値での誤差をスイッチング電源とバッテリーを用いて調査した。
測定に関しては、173はバッテリー用のケーブルに接続し、172、174はスイッチング電源用のコネクタに接続した。
[image][image]

尚、電圧計に表示されている値が瞬間的に何度も変化する場合は低い値を記載している。

#### 使用した電源
 - スイッチング電源(SW Power-05)  12.3[V]
 - LiPoバッテリー(22-30C-02) 	  13.06[V]

## 172
### スイッチング電源 
**誤差平均 0.963[V]**
|状況|電圧[V]|誤差[V]
|:----|:----|:----|
|ケーブルを刺した瞬間|11.5|0.8 |
|Jetsonの電源スイッチON|11.4|0.9 |
|サーボの電源スイッチON|11.2|1.1|
|hrコマンド実行|11.2|1.1 |
|servo_offコマンドの実行|11.2|1.1 |
|power offコマンドの実行 |11.2|1.1 |
|Jetsonの電源スイッチOFF|11.2|1.1 |
|サーボの電源スイッチOFF|11.5|0.8 |

### バッテリー
検証開始時 電圧：12.89[V]
検証終了後 電圧：12.76[V]
|状況|電圧[V]|誤差[V]
|:----|:----|:----|
|ケーブルを刺した瞬間|12.1| |
|Jetsonの電源スイッチON|12.0| |
|サーボの電源スイッチON|12.0| |
|hrコマンドの実行|11.8| |
|servo_offコマンドの実行|11.8| |
|power offコマンドの実行|11.8| |
|Jetsonの電源スイッチOFF|11.8| |
|サーボの電源スイッチOFF|12.1| |
<br>

## 173
### スイッチング電源
**誤差平均 1.075[V]**
|状況|電圧[V]|誤差[V]
|:----|:----|:----|
|ケーブルを刺した瞬間|11.4|0.9 |
|Jetsonの電源スイッチON|11.3|1.0 |
|サーボの電源スイッチON|11.1|1.2 |
|hrコマンドの実行|11.1|1.2 |
|servo_offコマンドの実行|11.2|1.1 |
|power offコマンドの実行|11.1|1.2 |
|Jetsonの電源スイッチOFF|11.1|1.2 |
|サーボの電源スイッチOFF|11.5|0.8 |

### バッテリー
検証開始時 電圧：12.76[V]
検証終了後 電圧：12.62[V]
|状況|電圧[V]|誤差[V]
|:----|:----|:----|
|ケーブルを刺した瞬間|11.9| |
|Jetsonの電源スイッチON|11.7| |
|サーボの電源スイッチON|11.6| |
|hrコマンドの実行|11.5| |
|servo_offコマンドの実行|11.5| |
|power offコマンドの実行|11.6| |
|Jetsonの電源スイッチOFF|11.6| |
|サーボの電源スイッチOFF|11.7| |

## 174
### スイッチング電源
**誤差平均：0.975[V]**
|状況|電圧[V]|誤差[V]
|:----|:----|:----|
|ケーブルを刺した瞬間|11.5|0.8 |
|Jetsonの電源スイッチON|11.4|0.9 |
|サーボの電源スイッチON|11.2|1.1 |
|hrコマンドの実行|11.2|1.1 |
|servo_offコマンドの実行|11.2|1.1 |
|power offコマンドの実行|11.2|1.1 |
|Jetsonの電源スイッチOFF|11.2|1.1 |
|サーボの電源スイッチOFF|11.7|0.6 |

### バッテリー
検証開始時 電圧：13.06[V]
検証終了後 電圧：12.89[V]
|状況|電圧[V]|誤差[V]
|:----|:----|:----|
|ケーブルを刺した瞬間|12.3| |
|Jetsonの電源スイッチON|12.1| |
|サーボの電源スイッチON|12.0| |
|hrコマンドの実行|11.9| |
|servo_offコマンドの実行|11.9| |
|power off コマンドの実行|11.9| |
|Jetsonの電源スイッチOFF|11.9| |
|サーボの電源スイッチOFF|12.1| |

hiroki-0001, 2022-07-05:
@MasatoKubotera 
https://github.com/citbrains/citbrains_humanoid/issues/1899
のissueでメインボード側に現在ファンを付けていると思いますが、それが実装済みの173機体で、B3M側の問題の調査の際に、ExpansionBoard 側のUSBハブから、FT4232H mini moduleに接続されてあるUSBを抜いたところ、Jetsonが落ちました。

この時の回路の状態は、

- ファンは動いている
- VNC接続が切れた
- USBの接続状況を示すオレンジのLEDは消えていた
- Jetsonの電源の供給を示すオレンジのLEDは点灯している

[その時の動画](https://youtu.be/2GdyNvMjNds?t=180)

この症状はメインボード側にファンを付ける前のコックピットでは起きていなかったため、このへんが原因なのかなと考えています

hiroki-0001, 2022-07-05:
上の状況の際のメインボード・拡張ボードのLEDの状態についても記載しておきます

[image]

②　スイッチング電源入力
④　サーボ電源入力
⑥　CPU電源入力

のみ点灯

[image] 

すべて消灯

MasatoKubotera, 2022-07-06:
@hiroki-0001 
上記の報告ですが，USBを抜いただけで落ちるというのは回路的には考えにくいです．
LEDの点灯状態的に回路自体から電源は供給されており，Jetsonがシャットダウン(落ちた)というのが正しいかもしれません．

貼付動画を拝見しましたが，USBを水平ではなく勢いよく斜め上方向に抜いてキャリアボードと接触しています．
場所的に3.3VとGNDにまたがるチップコンデンサに接触している恐れがあり落ちている要因かもしれません．

MasatoKubotera, 2022-07-11:
173機について，その後歩行・試合を通しましたが問題なく動作していました．

closed 2022-07-11 by MasatoKubotera