SUSTAINA-OP/SUSTAINA-OP #322: 一定期間運用したロボットの損耗を調査する

MibuchiYuta, 2022-06-05:
2月中旬より運用していた 170 SUSTAINA ProtoTypeのハードウェアの損耗を確認する。サーボモータに関しては #307 にて別途調査を行なっています。

MibuchiYuta, 2022-06-05:
# 右足

## 股付近
[image][image][image][image][image][image][image][image][image][image][image][image][image][image][image]

MibuchiYuta, 2022-06-05:
[image]
[image]
[image]
# 左足

一部のみの撮影になっています
[image]
[image]

MibuchiYuta, 2022-06-05:
# 上体
[image][image]

MasatoKubotera, 2022-06-25:
解体した170機については，下図のようにボックスにまとめました．
サーボモータのみ取り出しましたが，その他の部品については取り出していません．

[image]

必要最低限の写真での記録は残しましたので調査をするかどうかは今後の課題として一旦closeします．

他のロボットに実装されるとデータが変わってしまうサーボモータの調査については，下記のissueで行っています．
https://github.com/citbrains/SUSTAINA-OP/issues/307

closed 2022-06-25 by MasatoKubotera