SUSTAINA-OP/SUSTAINA-OP #323: 卓上CNC(Kitmill)での加工

yasuohayashibara, 2022-06-08:
津田沼キャンパスでのKitmillでの加工のログです．
正式な記録方法が分かりませんので，自由に追記していただいて構いません．

yasuohayashibara, 2022-06-08:
久しぶりに使用するので，まずは加工テーブルの面出しから始めました．
original mindが提供するGコードを使用しておよそ3時間程度です．
午後から切削を開始します．

[image]

yasuohayashibara, 2022-06-08:
InventorのCAMですが，『切断』に設定されているため，エラーが発生していました．
環境はAutodesk Inventor2022です．

[image]

『ミル』に変更したらエラーが出なくなりました．

[image]

yasuohayashibara, 2022-06-08:
切削始めました．
順調にいけば1枚目は3時ぐらいに終わると思います．
2枚目に関しても今日中に終了すると思います．

Keita1Mori, 2022-06-08:
@yasuohayashibara 
授業が終わり次第取りに伺います。

yasuohayashibara, 2022-06-08:
@Keita1Mori 
頂いたパスで切削していますが，怪しいところがありました．
取り外す前によく確認して下さい．
外す前でしたら，追加で切削できます．

yasuohayashibara, 2022-06-08:
切削完了していましたが，パスが不足しているところがあり，一部切削できていません．
追加で切削しますので，取り外さないようにしてください．
できれば再度パスをチェックして，問題があれば修正したデータを下さい．

[image]

yasuohayashibara, 2022-06-08:
以下の穴はCNCで切削しないということで良いでしょうか？
ちなみに，追加で切削する場所はここではありません．

[image]

Keita1Mori, 2022-06-08:
そうですね、上記で示した穴は後に手で追加工します。

yasuohayashibara, 2022-06-08:
1枚完成しました．
POM_T5_26も本日9時までには終了すると思います．
剥がして持っていっていただいて構いません．
（切削が完了しているかを丹念にチェックしてください．）

[image]

yasuohayashibara, 2022-06-09:
Kitmillでの加工は終了しました．
できれば評価を記載して，これ以上切削がない場合は閉じて下さい．

KiyoshiroKawanabe, 2022-06-10:
Cockpit_side_frameの下図で示す青色の部分の板厚公差を図りました．
どちらも1/100以下でしたので問題なく使用できると思います．
[image]
[image]
[image]

yasuohayashibara, 2022-06-11:
計測ありがとうございます．
issueを閉じます．

closed 2022-06-11 by yasuohayashibara