SUSTAINA-OP/SUSTAINA-OP #326: アルマイト加工の有用性の検討

ChikaraHanakawa, 2022-06-13:
初めてアルマイト加工をするので、実用可能なのかを実証する。

MasatoKubotera, 2022-06-13:
付属していたマニュアル：[Googleドライブ](https://drive.google.com/drive/folders/1Wzhs7fg0nhgD9KR53HrbKiYwM6neD03w?usp=sharing)

MasatoKubotera, 2022-06-14:
[アイストレー 14個取り 蓋付 H-5237](https://www.cainz.com/g/4976790552370.html)×７，[シールウェア OF－2](https://www.cainz.com/g/4973221053932.html)×１，[トワロン 151 Lサイズ](https://www.cainz.com/g/4907026015161.html)×２を追加で導入しました．

ChikaraHanakawa, 2022-06-14:
[image]
[image]
[image]
アルミ線を巻いた部分には塗装が施されないので、今後は目に当たらない様にする。
黒色の粉末染料が少なく、塗装予定のパーツ分足らない可能性がある。

ChikaraHanakawa, 2022-06-15:
[image]
[image]
孔封液に浸す際に誤って、パーツに触れてしまい、触れた部分だけ加工されていません。
今後、起こさないよう注意します。

MasatoKubotera, 2022-06-15:
マニュアル的なドキュメントを下記に残します．
https://docs.google.com/document/d/1WkH5zl4iEQ3SiRPdY8DgziSDmaT4UwKwvRA8MUlZRwk/edit#

MasatoKubotera, 2022-06-18:
[電解液　4L ３本](https://store.shopping.yahoo.co.jp/mc-across/LA-DE3.html)×1
[粉末染料　ブラックDM](https://store.shopping.yahoo.co.jp/mc-across/LAOP-SEN-BK-11.html)×2
[アルミ線＜ レッツアルマイト用＞](https://store.shopping.yahoo.co.jp/mc-across/LAOP-ALMS-11.html)×3
[鉛板](https://store.shopping.yahoo.co.jp/mc-across/LAOP-NAMAR-11.html)×4

を追加導入しました．

Dansato1203, 2022-06-18:
amzonで売っている鉛板で使えそうなのは[こちら](https://www.amazon.co.jp/%E5%85%89-Hikari-GZ555-%E9%89%9B%E6%9D%BF-0-5%C3%9750%C3%97300mm/dp/B00AJHJRW6/ref=sr_1_2?keywords=%E4%BA%BA%E6%B0%97%E3%81%AE%E9%89%9B%E6%9D%BF%E3%83%A9%E3%83%B3%E3%82%AD%E3%83%B3%E3%82%B0&qid=1655535719&sr=8-2)くらいです

ChikaraHanakawa, 2022-06-18:
[image]
[image]
一枚目は、穴が変色しています。
二枚目は、ネジが溶けてます。

ChikaraHanakawa, 2022-06-18:
[image]
作成しましたが、片方の一つを失敗しました。恐らく、鉛板のやすりがけを忘れた又は、電解液の寿命がきたのかどちらかだと思われます。

ChikaraHanakawa, 2022-06-18:
https://www.youtube.com/watch?v=eqXbj82pwAY&t=651s
アルマイト加工を参考にした動画です。

ChikaraHanakawa, 2022-06-19:
[image]
一回目に使用した鉛板の消耗が激しく、現在使用している鉛板は消耗がゆったりとしている。
電解液も同じく、一回目は液体の色が黄色に変色した。
パーツを洗浄していない事が原因だと思われる。
材料が限られており、消耗品を節約する為に洗浄は必要である。

ChikaraHanakawa, 2022-06-19:
[image]
174のパーツを加工しました。

ChikaraHanakawa, 2022-06-19:
[image]
[image]

Dansato1203, 2022-06-20:
## 認識から見たアルマイト加工の有効性
以下、アルマイト加工を施したパーツを組み込んだロボットとそうでないロボットで行ったフィールド検出の様子です。
- **アルマイト加工をしていないロボット**
[image]
https://www.youtube.com/watch?v=CC2fTmBPP4w
  
- **アルマイト加工したロボット**
[image]
https://www.youtube.com/watch?v=K6OX23yopg8

  
アルマイト加工を施した場合、明らかに該当部分の誤検出が少なっており**自己位置推定への影響が小さく**なっています。
そのため、アルマイト加工はこの面においては**有効**と言えそうです。

Dansato1203, 2022-06-20:
https://blog.lurestyle.jp/archives/14857
こちらを参考にアルマイト加工に失敗したパーツの皮膜を剥離させました。
[image]

これらのパーツをサンドペーパーで磨いてもう一度アルマイトしてみます

ChikaraHanakawa, 2022-06-23:
[image]
[image]
[image]
[image]
アルマイト加工の効率を上げる為にアルミ吊架棒を工場から頂いたアルミ棒を使用しました。
テスターを使用して電流が流れない事が判明した為、使用する部分のみをヤスリで削り、電流が流れる事を確認しました。
テスト用に1つのパーツに絞りました。
結果、失敗しました。原因は現状、不明です。

ChikaraHanakawa, 2022-06-23:
[image]
今日の進捗

ChikaraHanakawa, 2022-06-24:
https://docs.google.com/spreadsheets/d/1HcaccV7aFjciqpPa2d_VJO1ASei7_OJqTJKlwr4NjKI/edit#gid=1711021521
こちらでアルマイト加工のパーツを管理しています。

ChikaraHanakawa, 2022-06-27:
@MasatoKubotera 
封孔液を作成する為の封孔剤が残り、1L分しかありません。電解液が1回交換できる量しかありません。

MasatoKubotera, 2022-07-11:
有効性については，カメラによる白線認識の妨げにならないことを確認しました．

白線認識の妨げとなっていてアルマイトを実施した部品は下記の通りです．
- A2017 T5	Shoulder_horn_frame
- A2017 T2	Shoulder_side_frame
- A2017 T4	Trunk_shoulder_frame
- A2017 T4	Trunk_shoulder_frame_MIR
- A2017 T4	Trunk_neck_frame
- A2017 T4	Trunk_servo_bottom_frame

また，足首以下を黒くしなければならないというルール面からの変更に対しては下記の部品を対応しました．

- A2017 T2	Ankle_back_frame
- A2017 T2	Ankle_side_frame
- A2017 T3	Ankle_front_frame
- A2017 T4	Ankle_side_pitch_frame
- A2017 T2	Foot_back_frame
- A2017 T2	Foot_front_frame
- A2017 T5	Foot_root_block

MasatoKubotera, 2022-07-11:
タイトルにある有効性の検討については終了したのでCloseします．

closed 2022-07-11 by MasatoKubotera