SUSTAINA-OP/SUSTAINA-OP #329: 歩行時に独立ピッチのパーツが周辺のビスと接触してしまう

MibuchiYuta, 2022-06-18:
歩行時に足首独立ピッチの内側にあるAnkle_side_pitch_frameと独立ピッチのサーボを固定しているビスの頭が接触していました。
[YouTube 1:30〜 ](https://youtu.be/NrYKs6UTG5c?t=89)

hiroki-0001, 2022-06-18:
この問題ですが、以下のパラメータを変更させた際に発覚しました

- xp_mv_straight.z3	hr時の重心高さ(歩行の計算の際に使用)
    - 302.0　→　323.5(変更後)
- xp_mv_ready.time	ready姿勢になるまでの時間
    - 1.5 →　0.5(変更後)
- xp_mv_walk.h_cog	歩行時の重心高さ(歩行の計算の際に使用)
    - 300.0 → 290.0(変更後)

パラメータを見た感じ、パラメータの変更による影響というよりも単純に足首ピッチのフリーホーン側のネジが緩んでしまい、それが原因で足首独立ピッチの内側のAnkle_side_pitch_frameが動く。

これにより、付近にあるネジと接触してしまうことがあったと考えています

hiroki-0001, 2022-06-18:
### 2個目の接触パターン

eeprom_list.txt(歩行調整用パラメータ)でロボットの股の開く量を小さくした(20 → 15 単位はおそらく[deg])
すると歩行時に足付け根ロールの[DBTS4-11](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4&PNSearch=DBTS4&KWSearch=DBTS4-11&searchFlow=results2products&list=PageSearchResult)の内側の後ろ側がガラエポのリンクパーツ(FrontLink_frame)と接触して傷がついていました。

[image] 

以下傷の状態

[image] 

[image] 

[image] 

これに対しては対象のパラメータ(ロボットの股を開く量)を既存の20のままにした場合接触するところは見られなかったので、このパラメータは今後小さくさせないようにするようにしていきたいと思います

hiroki-0001, 2022-06-18:
### 3個目の接触パターン??

1, 2回目の接触のパターンが起きないように歩行パラメータを調整したのにも関わらず、脚部から稀に『カチカチ』と言ったなにかが接触するような音が聞こえることがある。

観察したところ、ロール軸方向の接触は見られなかったため、おそらくピッチ軸部分での接触が考えられるが何が接触しているかどうかは不明

また芝生の状態によっても音がなるときとならないときがある。

メインフィールドの3Dプリンタ側の陣地の芝ではなる

工場側の陣地ではならない？

hiroki-0001, 2022-06-21:
>歩行時に足首独立ピッチの内側にあるAnkle_side_pitch_frameと独立ピッチのサーボを固定しているビスの頭が接触していました。
@MasatoKubotera
[YouTube 1:30〜](https://youtu.be/NrYKs6UTG5c?t=89)

の問題ですが172, 173, 174機体で発生していました

各機体のAnkle_side_pitch_frameの状態に関しては各機体ログの「220620」に写真として残しておきます

- [172機体ログ](https://docs.google.com/document/d/1Bu__7bFpr7pK8O6WlwEwFvpa3BMVsJl1QJU1Rt5ZLbg/edit?usp=sharing)
- [173機体ログ](https://docs.google.com/document/d/1U-59AQhLoF95vKk1aIkmdPmZ7NBgLv7Q0eTqMOcJjHc/edit?usp=sharing)
- [174機体ログ](https://docs.google.com/document/d/1YwnAH5AbXvdZMq6qINXNWuMH8qINUu5QY9R45Hn63rM/edit?usp=sharing)

写真を見ていただければわかるとは思いますがAnkle_side_pitch_frameの状態はネジとの接触で傷がついているというよりも跡がついているような感じでした

MasatoKubotera, 2022-06-21:
Ankle_side_pitch_frameにねじ頭の黒色塗料が付着している問題に関して，下図に示す青色で選択した2本のねじが要因です．

[image]

通常の歩行動作で接触するのは，上側のねじのみになります．
下側は，かなり独立ピッチ軸を動かすキック動作などでしか接触が起きません．

この原因については，単純にFreeHornの取り付け軸が負荷に耐え切れずぶれていることが原因です．ClampHorn側（出力軸）では，サーボモータ本体との取付が強固であるためこの現象はフリーホン軸特有の問題になります．

Dynamixelでは，軸ブレを抑えるため，アキシアル荷重に強固にするためにホーンとサーボモータ本体との間に挿入するスラストベアリングがオプションとして存在します．

[image]

https://e-shop.robotis.co.jp/product.php?id=152

スラストベアリング以外でも樹脂系のワッシャを挿入することでこの問題は解決できますが，
現状，フリーホンからネジが飛び出ていることやGankenKunでもこの問題は起きていているため，世界大会までは該当の2つのねじを取り外すことで対応したいと思います．

@hiroki-0001

MasatoKubotera, 2022-06-21:
下記のリストは上記で提示しているサーボモータのフリーホーン側２つの締結ねじを抜くタスクリストです．
抜いたらチェックつけていきます．

- [ ] 171
- [ ] 172
- [ ] 173
- [ ] 174
- [ ] 175
- [ ] 176

MasatoKubotera, 2022-06-23:
>歩行時に足首独立ピッチの内側にあるAnkle_side_pitch_frameと独立ピッチのサーボを固定しているビスの頭が接触していました。

間近で様子を撮影しました．

FreeHornブレ（足首ピッチ軸）：https://youtu.be/eIy3EqUrck0

hiroki-0001, 2022-06-25:
https://github.com/citbrains/SUSTAINA-OP/issues/329#issuecomment-1159478656

> 3個目の接触パターン??
1, 2回目の接触のパターンが起きないように歩行パラメータを調整したのにも関わらず、脚部から稀に『カチカチ』と言ったなにかが接触するような音が聞こえることがある。
観察したところ、ロール軸方向の接触は見られなかったため、おそらくピッチ軸部分での接触が考えられるが何が接触しているかどうかは不明
また芝生の状態によっても音がなるときとならないときがある。
メインフィールドの3Dプリンタ側の陣地の芝ではなる
工場側の陣地ではならない？

この音の原因ですがおそらくhttps://github.com/citbrains/SUSTAINA-OP/issues/330 で発見された脚部のピッチ軸のフリーホーンから鳴っていたと考えられます。

MasatoKubotera, 2022-06-27:
>歩行時に足首独立ピッチの内側にあるAnkle_side_pitch_frameと独立ピッチのサーボを固定しているビスの頭が接触していました。

に関しては下記のissueで引き継ぎます．
https://github.com/citbrains/SUSTAINA-OP/issues/335

上記でコメントされている1回目，2回目の接触についてはほかのissueで議論され，2回目の接触については人為的ミスと対応がそれぞれ決まったのでcloseします．

closed 2022-06-27 by MasatoKubotera