SUSTAINA-OP/SUSTAINA-OP #330: 膝下サーボのフリーホーンが沈む現象

KiyoshiroKawanabe, 2022-06-22:
172の右膝下(ID3)，174の左膝下(ID12)のフリーホーンが沈む．

下記isuueで議論されている影響で膝下に負荷がかかりすぎていると考えられる．
https://github.com/citbrains/citbrains_humanoid/issues/1843

KiyoshiroKawanabe, 2022-06-22:
世界大会前にフリーホーンの交換を行う．

MasatoKubotera, 2022-06-23:
@yasuohayashibara
>もし部品の交換で対応するということでしたら，予備を多めに用意したほうが良いと思います．
FreeHornについて，42個（10個/機）の新品予備を用意しています．

@hiroki-0001 
膝ピッチ軸に関しては，５本のねじを抜くことにより交換が可能ですので比較的交換しやすい場所ですが，下記のissueでも報告されているように，足首ピッチ軸でも同様の問題が発生しています．
https://github.com/citbrains/SUSTAINA-OP/issues/329#issuecomment-1161270453

足首ピッチ軸で交換対応となると，軸ネジを含む9本を外さなくては交換できないため，10~20分程度の交換時間を要します．

この問題は，GankenKunでも起きている現象であり，歩行制御だけでの解決は難しいと考えています．

そのため，下記のコメントで提示しているように，サーボモータ本体とFreeHornの間にPTFEワッシャを挿入する対策を緊急で行いたいと思います．
https://github.com/citbrains/SUSTAINA-OP/issues/329#issuecomment-1161313174

hiroki-0001, 2022-06-23:
実際にフリーホーン側が沈む問題を示すような動画がなかったので撮影しておきました。

https://youtu.be/maucm3aaglU

MasatoKubotera, 2022-06-23:
挿入イメージ

[image]
フリーホーンの外形はφ21で，理想のPTFEワッシャーは内径φ12.5 外径φ21 厚さ2です．

部品を変更せず，追加実装するとなるとこのあたりになりそうです．
[TT-1220-20](https://www.hirosugi-net.co.jp/shop/g/g31980/) 内径φ12.5 外径φ20 厚さ2
[TT-1321-20](https://www.hirosugi-net.co.jp/shop/g/g31985/) 内径φ13 外径φ21 厚さ2

現状，SUSTAINAで使用している[TT-1020-20](https://www.hirosugi-net.co.jp/shop/g/g31941/)が近かったので加工して装着してみました．
内径をφ10からφ13に広げました．

[image]

結果として，沈む・ぶれる問題を抑制することはできそうです．厚さ精度・サーボモータ側の距離間も一定でないこと（ものにより異なる）ことから100％の解決とまではいかなそうです．

---

問題がわかりやすいように動画に収めました．
FreeHorn沈み込み（膝下ピッチ軸）：https://youtu.be/kNmpu0GXDnQ
FreeHornブレ（足首ピッチ軸）：https://youtu.be/eIy3EqUrck0

サーボモータ単体にPTFEワッシャを挿入したときの比較
挿入後：https://youtu.be/LaY6Vv91a2A
挿入前：https://youtu.be/q6i7lx_H5M0

MasatoKubotera, 2022-06-23:
この問題は早急に対応することが望まれますが，足首ピッチ軸だけでも６か所に挿入する必要がある．
SUSTAINAで使用している[TT-1020-20](https://www.hirosugi-net.co.jp/shop/g/g31941/)を加工して挿入することはできるが数が足りないこと，加工が大変なため，対応するワッシャを購入して挿入するのが早そうです．

yasuohayashibara, 2022-06-24:
よく見たらフリーホーンの取付方向が逆です．
もっと早く気づくべきでした．

[image]

Kondoのサイト
https://kondo-robot.com/product/02137

MasatoKubotera, 2022-06-24:
ここで自分も提言していましたが間違えて取り付けていました．
完全に盲点でした
https://github.com/citbrains/SUSTAINA-OP/issues/99#issuecomment-922448953

MasatoKubotera, 2022-06-25:
今後の方針

FreeHornが沈む現象は，ベアリングの再取り付けで解決できますが https://github.com/citbrains/SUSTAINA-OP/issues/329#issuecomment-1161270453 で提示されている問題に関しては再取り付けだけでは解決できないので，樹脂ワッシャを挿入して対応したいと思います．

MasatoKubotera, 2022-06-27:
樹脂ワッシャ挿入に関しては下記のissueで引き継ぎます．
https://github.com/citbrains/SUSTAINA-OP/issues/335

また，FreeHornの表裏逆の対応については下記のタスクリストで対応します．
https://github.com/citbrains/SUSTAINA-OP/issues/332

closed 2022-06-27 by MasatoKubotera