SUSTAINA-OP/SUSTAINA-OP #331: StartStopSwitch がたまに機能しない問題について

hiroki-0001, 2022-06-23:
検証中、稀にStartStopSwitchの動作が機能しないことがある。
これの原因を調査する。

hiroki-0001, 2022-06-23:
StartStopSwitchのデバイスとして認識しているかどうかはlsusb・/dev內に生成されるデバイスファイルを参照することで確認できる

lsusbでの出力
```
ID 2341:8036 Arduino SA Leonardo (CDC ACM, HID)
```

/dev內のデバイスファイル
```
/dev/input/js*

* = ワイルドカード
```

hiroki-0001, 2022-06-23:
Switchの押し込み等の認識の確認方法としては2通りある

1. joystickを用いた確認方法
2. https://github.com/citbrains/citbrains_humanoid/tree/master/for2050/src/tools/joystick_testを用いた確認方法

2に関してはlibsdlを通った後のstartstopswitchで利用しているボタン0,1が利用できるかどうか、またStartStopSwitch本来の機能が使用できるかどうかの確認として有用

1に関しては/dev/input/js*として、つまりスイッチ単体をテストすることが可能になる。

MasatoKubotera, 2022-06-24:
@hiroki-0001 @AD58-3104 

>2に関してはlibsdlを通った後のstartstopswitchで利用しているボタン0,1が利用できるかどうか、またStartStopSwitch本来の機能が使用できるかどうかの確認として有用

https://github.com/citbrains/SUSTAINA-OP/issues/271#issuecomment-1139324746
に記載していますが，現状SUSTAINAに搭載しているStartStopSwitchはGankenKunとの互換性を保つために，button1，2に変更されています．

hiroki-0001, 2022-06-30:
6月30日の模擬試合中に174が転倒した際に同様の症状が発生しました。

/dev/input/にjs0があったためデバイス自体は認識されていそうだった。

そこで以下のツールを用いて、確認を行なった

- jstest-gtk
- [joysticktest](https://github.com/citbrains/citbrains_humanoid/tree/master/for2050/src/tools/joystick_test)
 
を用いて確認したがスイッチ自体は問題なく動作していたことは確認済み

しかしjstest-gtkをインストールするのに時間がかかったため、その間に回復した可能性は高い
(この作業中には制御スイッチのUSBは触っていない)

[作業中の動画](https://youtu.be/BT9LPdp1jds)

MasatoKubotera, 2022-06-30:
一番の問題として考えられるのは，Aruduino側のファームフェアが悪いことです．
最新バージョンのファームフェアについては，下記にあります．

MasatoKubotera, 2022-06-30:
@hiroki-0001 
>jstest-gtkをインストールするのに時間がかかったため、その間に回復した可能性は高い

この間にボタンを押下しましたか？

hiroki-0001, 2022-07-01:
先に[joysticktest](https://github.com/citbrains/citbrains_humanoid/tree/master/for2050/src/tools/joystick_test)を使用したので、その間にボタンは押下しました

MasatoKubotera, 2022-07-11:
下記のように，押下から解放されたときの処理をポートレジスタで高速化しました．

変更前
https://github.com/citbrains/SUSTAINA-OP/blob/ba2f0c7848bbbd35434bd6c1388189bd74b393aa/electronics/StartStopSwitch_ver3_0/firmware/StartStopSwitch/StartStopSwitch.ino#L32-L52

変更後
https://github.com/citbrains/SUSTAINA-OP/blob/0907be4b06165a5879109165c8dd803bea4ec1e8/electronics/StartStopSwitch_ver3_0/firmware/StartStopSwitch/StartStopSwitch.ino#L32-L52

参考：https://omoroya.com/arduino-extra-edition-22/

MasatoKubotera, 2022-07-11:
上記ファームウェアで検証しましたが，押された状態で止まることがあるようです．
ジョイスティックライブラリの仕様で通信信号を送信するとArduinoに実装されたLEDが点灯しますが，一度の押下で２度点灯点滅を繰り返していることが判明しました．

MasatoKubotera, 2022-07-11:
一応変更後の方が，動作的には安定しているので順次切り替えます．

- [ ] No.1
- [ ] No.2
- [ ] No.3
- [ ] No.4
- [ ] No.5
- [ ] No.6
- [ ] No.7
- [ ] No.8

MasatoKubotera, 2022-08-08:
押し続けられる現象について
基本的には，Arduinoのファームウェアが問題かと思います．
ジョイスティックコントローラのライブラリも使っていて，現状ライブラリに問題があるか，その他なのか特定できていません．

押し続けられる現象については，再度押せばもとに戻る仕様になっています．
現状のJetson側のシステムだと，**押されている**を感知しているようなので，押し続けられた誤情報で無限にボタンが押された行動を行うなど誤爆しないためには，押されてないー＞押されたこの０から１になるタイミングを検出しておいた方が良いかと思います．