SUSTAINA-OP/SUSTAINA-OP #333: csiカメラケーブルの取り回しの確立

RikuYokoo, 2022-06-25:
現在のcsiカメラケーブルの取り回しは以下の画像のとおりである．
[追加予定]

取り回し的にはいいがもう少し調整が必要である．
現在のままであると以下のように画像が読み込めないときがあるので対処が必要である．
[image]

MasatoKubotera, 2022-06-25:
@RikuYokoo 上記コメントでは，ケーブルの取り回しだけで起きている問題と断定ができません．
上記画像を取得したときの情報（機体番号やカメラ姿勢）などを追記してください．

連続して画像を取得している場合には，その前後の画像も添付してください．

MasatoKubotera, 2022-06-25:
読み込みエラー率についてわかればお願いします．

RikuYokoo, 2022-06-25:
本日発生したロボットは174です。
画像が見えなくなる前の画像を見ると首をほぼ最大角度(135度)に回転しているときだと思います。

時系列での画像です。
画像は約1秒周期で保存しています。

[image]
[image]

[image]
[image]

[image]

[image]

RikuYokoo, 2022-06-25:
画像にノイズが乗っているのもありました。
[image]

[image]

MasatoKubotera, 2022-06-26:
<blockquote>
[image]
</blockquote>

上記で報告があるカメラ映像ですが，前後の画像と，一部分だけ見えている部分を拡大したところ正面を向いているときだと思われます．
よって，
>画像が見えなくなる前の画像を見ると首をほぼ最大角度(135度)に回転しているときだと思います。

以外に画像が取得し出来ないことがあるようです．

yasuohayashibara, 2022-06-30:
該当するロボット固有の問題であれば大丈夫ですが，他のロボットでも生じそうな問題であれば最優先で対処したほうが良いと思います．

MasatoKubotera, 2022-06-30:
ケーブルが傷つく要因として大きく2つあります．

１．慣れてない人がコックピットの取り回し・配線を行う
２．首が回転したときになどで傷つく

１．に関しては，今回の検証で発生したように該当慣れてない人がかかわったことによる該当ロボット固有問題になるため，慣れていない人以外は触らないという解決ができます

２．に関しては，取り回しの方法関係なく，どの機体にも現状起こりえる問題です．

解決策として２つ示します．
ケーブルの幅が16mmということを踏まえて，

Ａ．φ17の収縮チューブ（畳まれた状態のもの）を装着する．※収縮はしない
https://item.rakuten.co.jp/gitamani/vgs-sstb-blk17mm-30m/

Ｂ．サーボケーブルと同じネットチューブの径が大きいものを装着する．
https://jp.misumi-ec.com/vona2/detail/222000189666/?HissuCode=SE75P-CR0

ケーブルの幅参考：https://www.we-online.com/catalog/datasheet/686715152001.pdf

RikuYokoo, 2022-07-02:
模擬試合で撮った画像を確認したところ171で一番初めの画像のみ以下のような画像になっていた．
[image]
その他の171で撮った画像は正常であった

RikuYokoo, 2022-07-02:
174では全て正常な画像であった．
()
175では
途中画像が固まる症状が起きた．以下連続画像
正常
[image]
異常
[image]
[image]
[image]
[image]
正常
[image]

歩行中に起きている

MasatoKubotera, 2022-07-06:
@RikuYokoo 

>175では途中画像が固まる症状が起きた．

上記はソフトウェア側の問題かと思います．

closed 2022-07-11 by MasatoKubotera