SUSTAINA-OP/SUSTAINA-OP #336: MainBoardのJetson側への電源電圧の変更による影響の調査

hiroki-0001, 2022-06-29:
https://github.com/citbrains/SUSTAINA-OP/issues/299 にて設計されているMainBoardですが以下の問題が発生している

- [jetsonのCPUが落ちる問題について #1528]( https://github.com/citbrains/citbrains_humanoid/issues/1528)
- [xavier nxで戦略を実行した際に"System throttled due to Over-current"と警告が出る原因の調査 #1771](https://github.com/citbrains/citbrains_humanoid/issues/1771)

これらの問題の解決のため、DCDCを通してJetson側に供給する電源電圧を19Vに変更する。

この変更による影響を調査する

hiroki-0001, 2022-06-29:
Jetson側に供給する電源電圧を19Vに変更した際の調査を行いました。
以下調査項目になります。

1. 通常時(OS起動後 無操作の状態)
2. build時
3. 戦略実行時
4. CPU・GPU最大負荷
5. 最大負荷(以下を実行)
    - 戦略実行
    - CPU・GPU最大負荷
    - HPLStatusUIの起動
    - wgetでisoファイルのダウンロード

これら項目ごとの電源電圧の波形の調査を行いました。

---

## 調査結果

以下の条件で波形を確認しました。

- スケール：2[V] (1マス 2Vずつ変化する)
- 位置：-16.00[V]
- 時間：250[ms]
- 波形の種類
    - 黄色：DCDC出力
    - 水色：DCDC入力

左が10W4core設定, 右が15W6core設定

### 1. 通常時　電圧波形

[image]  [image] 

### 2. build時

[image] [image] 

### 3. 戦略実行時

[image] [image] 

### 4. CPU・GPU最大負荷

[image] [image] 

### 5. 最大負荷時

[image] [image] 

---

### [計測時の動画](https://youtu.be/Fo7QQR-O4rQ)

hiroki-0001, 2022-06-29:
↑の調査ですが、調査中に稀に撮影した波形よりも乱れる事はありました。
動画には収めてあるので一度確認してもらえるとありがたいです。

しかし、https://github.com/citbrains/citbrains_humanoid/issues/1528#issuecomment-1148621649
のときのように、電圧が6V程度降下することはありませんでした。

MasatoKubotera, 2022-07-01:
上記の検証は電源切り替え回路が故障しているMainBoard No.2のDCDCコンバータ出力電圧を操作するR21，R22をそれぞれR21=[33k](https://akizukidenshi.com/catalog/g/gR-06333/)，R22=[2.4k](https://akizukidenshi.com/catalog/g/gR-06242/)にして，出力電圧を1.25*(1+33/2.4)=18.4375V（実測18.5V）にして検証を行ってもらいました．
[image]

---

@hiroki-0001 

上記の検証と過去に行ったの検証の波形（ https://github.com/citbrains/citbrains_humanoid/issues/1528 , https://github.com/citbrains/citbrains_humanoid/issues/1841 ）を踏まえて，大幅なスパイクノイズの改善が確認できませんでしたので従来通りの12V駆動で行いたいと思います．

MasatoKubotera, 2022-07-11:
調査が終了していますのでCloseします．

closed 2022-07-11 by MasatoKubotera