SUSTAINA-OP/SUSTAINA-OP #337: SUSTAINA用バンパーの衝突実験

hiroki-0001, 2022-06-29:


hiroki-0001, 2022-06-29:
転倒実験を行いました。詳細をまとめるのに時間がかかりそうなので、計測データがどこにあるかだけ載せておきます

https://drive.google.com/drive/folders/15AtwwdoDDcb46LN6fy25PNkimVYolGy6?usp=sharing

から飛べます。

hr姿勢時・ready姿勢時　各姿勢で前方・後方・右方・左方・斜め前・斜め後ろを計測しました

時間がなかったため、1回ずつしか行えていないです

hiroki-0001, 2022-06-29:
間違えてcloseしてしまいました....

MasatoKubotera, 2022-06-30:
@hiroki-0001 
機体改造（おもり乗せ）を行って計測していると思うのでその辺の情報もあとで追記お願いします．

hiroki-0001, 2022-06-30:
転倒実験を行いました。
検証対象は https://github.com/citbrains/SUSTAINA-OP/issues/337#issuecomment-1170622371 で述べられているように下記のコミット時点のものになります。

https://github.com/citbrains/SUSTAINA-OP/tree/583df4d54e541f563c366a1d94848a7a7c5797e6

また足裏の面積変更により重心高さを上げなければならないので、以下の写真のように肩部分に60g × 2 = 120gの重りが載っています

[image] 

また今回の転倒実験には「MA3-50AD-RDB-SS」という加速度センサを使用した。

この加速度センサは写真のようにロボットの正面に養生テープで固定した

[image] 

直立姿勢・ready姿勢の2種類の姿勢で

前後左右・斜め前・斜め後ろの合計6方向に転倒させました

## 実験結果

実験データは以下のディレクトリにまとめてあります

以下のグラフ表示です

### 【直立姿勢】

[image] [image] 

[image] [image] 

[image] [image] 

### 【ready姿勢】

[image] [image] 

[image] [image] 

[image] [image] 

## [実験中の動画](https://youtu.be/sF5ddDAhaXM)

yasuohayashibara, 2022-06-30:
明確な指針があるわけではありませんが，経験的に40G程度であれば故障の確率は低いと思います．
ただし，上記のグラフは軸ごとに出力していますので，`sqrt(x^2+y^2+z^2)`も計算したほうが良いです．

MasatoKubotera, 2022-07-02:
GKバンパーについては下記のissueで行っています．
https://github.com/citbrains/citbrains_humanoid/issues/1822

MasatoKubotera, 2022-08-23:
RoboCup2022以前にバンパーについて問題があるか検証するためのissueであるため，Closeします．
上記の結果でRoboCup2022に臨みましたが，試合中にロボットが損傷または断線等，ハードウエア要因による故障は1回もありませんでした．

closed 2022-08-23 by MasatoKubotera