SUSTAINA-OP/SUSTAINA-OP #338: 新しい仰向け起き上がり時に入力電圧エラーが発生する

hiroki-0001, 2022-06-30:
@MibuchiYuta が作成中の仰向け起き上がり動作の再生中に

- ID1 = 右足首ロール
- ID2 = 右足首ピッチ
- ID10 = 左足首ロール
- ID11 = 左足首ピッチ
- ID18 = 左肘ピッチ

から`input voltage`のエラーが出力されました。

このエラーは指定された最大入力電圧値を超えた瞬間に警告として出るものです。

この最大入力電圧を調べたところ、18.0Vでした。つまり仰向け起き上がり動作中にID1, 2, 19, 11, 18の計5箇所に対して18Vの負荷がかかっていることになります。

このままでは回路基板に大きな損傷を与えかねないので原因と対策を考える

hiroki-0001, 2022-06-30:
この問題が起きた際の動画は以下に載せておきます

- [B3M Managerの様子](https://youtu.be/PusiG6R8BVE)

yasuohayashibara, 2022-07-01:
調査ありがとうございます．
このような具体的なデータは問題解決するために重要ですね．

以下のP.42に
『入力電圧が上限値を超えたか、下限値を下回った場合に1になります』
とありますので，大電流を流したため電圧降下により下限値を超えた方が可能性としてはあると思います．

https://kondo-robot.com/w/wp-content/uploads/B3M_SoftwareManual1.2.0.3.pdf

あと，気になったところとしてフレーミングエラーも発生しているようです．
上記の資料のP.43に，
『フレミングエラー発生時に1になります』とありますが，これでは分かりづらいので，以下が参考になるかもしれません．

https://www.lineeye.co.jp/faq/a00082.html

MibuchiYuta, 2022-07-01:
@hiroki-0001 
この問題は176で起きた問題という認識でいいのでしょうか?

MibuchiYuta, 2022-07-01:
こちらは171で起きていました

> あと，気になったところとしてフレーミングエラーも発生しているようです．

MasatoKubotera, 2022-07-01:
>あと，気になったところとしてフレーミングエラーも発生しているようです．
上記の資料のP.43に，
『フレミングエラー発生時に1になります』とありますが，これでは分かりづらいので，以下が参考になるかもしれません．

https://www.lineeye.co.jp/faq/a00082.html
こちらに関しては，ソフトウェア分野になってしまうので，https://github.com/citbrains/citbrains_humanoid/issues で扱ってください．

@AD58-3104 @hiroki-0001

MasatoKubotera, 2022-07-01:
>以下のP.42に
『入力電圧が上限値を超えたか、下限値を下回った場合に1になります』
とありますので，大電流を流したため電圧降下により下限値を超えた方が可能性としてはあると思います．

明日の試合検証で @AD58-3104 がロギングシステムを導入するとのことなのでそちらで様子見したいと思います．

MibuchiYuta, 2022-07-01:
176のこの現象が起きた際の動画は撮影されてなかったそうです

AD58-3104, 2022-07-02:
176で問題の現象が発生したモーション実行時の電圧を計測しました。結果は以下です。
横軸が時間(10ms)縦軸が電圧(1mv)です。0秒時点で4番モーションのコマンドを受信し、そこから動作開始しています。
### 全体の画像
[image]
### 最低まで降下した箇所のアップ
[image]

AD58-3104, 2022-07-02:
174 回路変更前の機体での同じモーション再生結果
[image]

MasatoKubotera, 2022-07-02:
176と174の回路の違い

こちらのissueで対応している突入電流対策用のスイッチに直列で接続する抵抗値が異なります．
下記のような仕様になっています．
- 174 : 10k
- 176 : 20k

林原先生が口頭で提言して仮説として提言していましたがそちらが原因のようですね．

2022/7/4の模擬試合までに順次問題が起こらない10kに変更します．

MasatoKubotera, 2022-07-11:
上記で示されているように，ロードスイッチに直列な抵抗を10kとすることで，問題が解決されているため一旦closeします．

メモ：この時に使用されているMainBoardはNo.6であり，使用しているFETは[SQM40P10-40L_GE3](https://www.mouser.jp/ProductDetail/Vishay-Siliconix/SQM40P10-40L_GE3?qs=jHkklCh7amg3sgsHIUrcaw%3D%3D)でした．

closed 2022-07-11 by MasatoKubotera