SUSTAINA-OP/SUSTAINA-OP #342: RoboCup2022で使用・故障したB3Mの整理

hiroki-0001, 2022-07-30:
RoboCup2022では芝生の環境が悪かったため、多くのB3Mサーボモータに大きなガタが発生した。
その影響により多くのB3Mを交換したが、どのB3Mが故障しているのかどうかが曖昧になってしまったので調査・整理を行う

hiroki-0001, 2022-07-30:
現在故障しているおそれのあるB3Mはすべて故障登録を行いました。
故障しているB3Mの一覧は[新_B3M管理シートの「故障リスト」](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit?usp=sharing)を参照してください。

SBのB3Mは修理する対象では無いので、これらを除外すると故障しているB3Mは
**1040が8個、1170が20個の合計28個になります。**

これらのB3Mが故障のチェック等はこれから行なっていきます

hiroki-0001, 2022-07-30:
まだ正確な調査は済んでいませんが、**故障したB3Mの内の世界大会の芝生環境による影響でガタが生じたものは19個程度**になります

hiroki-0001, 2022-07-30:
一応議題にあったので共有しておきますが以下のpdfにB3Mの各部品の修理の値段が記載されています
https://kondo-robot.com/w/wp-content/uploads/parts_robot_servo20210911-1.pdf

hiroki-0001, 2022-07-30:
@KentaTsukino にやってほしいことを以下にまとめます

**1. [新_B3M管理シートの「故障リスト」](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit?usp=sharing)に記載されている故障の調査**
主に出力ホーンに大きなガタがあると判定されたもののガタを計測してください。

**2. 故障B3Mの修理費用の概算**
 https://github.com/citbrains/SUSTAINA-OP/issues/342#issuecomment-1200119521 にある、修理費用を元にそれぞれのB3Mの修理費用の概算を出してください。

yasuohayashibara, 2022-07-31:
可能であればですが，世界大会を通じてサーボモータのガタがどの程度大きくなったかを定量的に調べてもらえれば助かります．
ロボットごとに交換時期も含めて明確にしてもらえると貴重なデータになりそうな気がします．
なお，現在取り付けてあるモータに関しては取り外す必要はありません．
ちなみに，これ調べると研究発表できそうかも．

MasatoKubotera, 2022-07-31:
ガタ調査に関して，現状定義さてている計測方法では，
>定量的
な評価がまだ難しい部分があります．

評価する人によって，加える力の大きさが異なり，強すぎるとガタの数値が大きく評価されてしまいます．

そのため，一定トルク速度で回転できる機器を用いた方がより定量的に評価できるかと思います．

今は，クランプホーンを直接手で回転させることが難しいため，調査できていませんが．．．
機器を軸に容易に取り付けることができれば現状搭載されているものも調査対象に含めることができそうです．

計測することを前提としたハードウェア仕様にしても良いのかなと思います．

KentaTsukino, 2022-08-02:
## 電源入れただけで熱くなったサーボ
40-1927

KentaTsukino, 2022-08-03:
## サーボ調査
### 8/3

- 70-1985　B3M ManegerのSarvo Raramtersで出力軸を動かしたら、OVPが12.1V以上になったので止まった。OCPは測っているとき、最大で0.3Aほどだった
- 70-1921　Baudrateを何にしても入れなかった。B3MServoCheckerで入ってみたら、Baudrateが3000000でIDが４で止まった。他のBaudrateは29で止まる
-

yasuohayashibara, 2022-08-03:
そういえばですが，これらのデータを分かりやすく整理して，近藤科学に提供してあげれば喜ばれるような気がします．
こちらから問題を提示することで，改良してくれるかもしれませんので，こちらにもメリットがあります．

hiroki-0001, 2022-08-05:
一応ある程度調査したので、修理費用の概算を行いました。  
以下のスプレッドシートに費用等の情報が載っています
https://docs.google.com/spreadsheets/d/1U6zZ1ngHVIbdfdb_4UAU768oHHx5UNLALx8iGcP_BUA/edit?usp=sharing

**トータルで¥104,300になります。。。**

hiroki-0001, 2022-08-05:
前回の修理の際にいくつかのB3Mに関してはガタが改善されていなかったものもあったため、それらのデータを集めて問い合わせられるような資料を作成します

KentaTsukino, 2022-08-11:
## サーボ調査
### 8/11

- 40-001 
 故障判断の誤り。正常に使用可能です
- 40-002
故障判断の誤り。正常に使用可能です

hiroki-0001, 2022-10-12:
https://github.com/citbrains/SUSTAINA-OP/issues/342#issuecomment-1204533782
で指摘されているように、前回修理時にチームが正常と判断できるラインまでガタが修理されていなかったので、そのレベルまで修理が可能かどうかみたいなのを質問する内容の資料を作成します。
とりあえず今週中にある程度作成するつもりです

hiroki-0001, 2022-10-28:
@yasuohayashibara 
資料の作成がとりあえず終わりました。

[『221012_B3Mサーボモータ 出力軸のガタについて』](https://docs.google.com/document/d/10dJAnjl1gcs_qocyCsq9m92YMxGW_qUp2ZJC9gkRsDw/edit?usp=sharing
)に問い合わせる内容を記載してあります。

また、近藤科学側に提出する書類として、

- [『B3Mサーボモータ 出力軸に生じるガタの計測方法』](https://docs.google.com/document/d/1vZdy7MidPszqy6u9QeZGzGlUynMgIRMCaXT-9e0R4aU/edit?usp=sharing
)
  - B3Mのガタの計測方法に関して記載した資料

- [『修理前・修理後のB3Mサーボモータのガタ計測データ』](https://docs.google.com/spreadsheets/d/1PQakEwBxtImDY8kQLW5g3x3-FOLDDoWOOC64wPNNNbc/edit?usp=sharing)
  - 2022年4月25日の修理申請時の修理前・修理後のガタの計測データの比較したデータ

の2つも合わせて作成しました。これらの資料をまとめて提出しようと思います。

しかし誤りが無いかの確認していただきたいので、校閲をお願いします。

yasuohayashibara, 2022-11-03:
こちらに記載するのが遅れましたが，近藤科学に連絡しました．

hiroki-0001, 2022-11-25:
近藤科学とのやり取りをドキュメントにまとめておきました．
https://docs.google.com/document/d/1KUbJoYK34SZCzZQFY0DWqyWjMtj4y69VyGjFcDVgfzY/edit?usp=sharing

hiroki-0001, 2022-12-03:
B3Mの修理申請のためにB3Mの在庫情報と[管理シート](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit?usp=sharing)の情報を照合しました．

結果

- 70-1982
- 70-2209

が行方不明になっていることがわかりました．この2つのB3Mサーボは故障判定のため，見つけ次第修理に出したいと思います．

hiroki-0001, 2022-12-03:
これが見つかるまで，修理申請を止めてしまうと大幅に申請が遅れてしまうので，上記の2つのB3Mは一旦除いて修理に出したいと思います．

hiroki-0001, 2022-12-14:
@yasuohayashibara 
修理申請に必要な資料の作成が終わりました．今回は作成した資料が多いので，

[**大学に提出する資料**](https://drive.google.com/drive/folders/1ummGdrxfehkFadpgw5PZpIOASZtCEc4u?usp=share_link)

[**近藤科学に提出する資料**](https://drive.google.com/drive/folders/1isWU1ENWAo5KiKH3yLG1xVnKqiWTHJq8?usp=sharing)

の2つのディレクトリを分けておきました．それぞれに入っている資料は以下の通りです．

### 大学に提出する資料

- [修理前のB3Mサーボモータの写真](https://drive.google.com/drive/folders/1zAyscmYAD5OYSdgTQNdxQkQ2OCwAWUBU?usp=share_link)
- [故障内容・時期・状況・原因に関して書かれている資料](https://docs.google.com/spreadsheets/d/1RKYKTIwdmYISy0bb2GfliP4PxyUzylM8rCFdgWj6RFk/edit?usp=sharing)
- [CIT BrainsでのB3Mサーボモータの使用状況について](https://docs.google.com/document/d/16CxLMy2WzPqTqVSOCEqpcCxIEV4qu1XFzb1N6BS_TC0/edit?usp=sharing)

### 近藤科学に提出する資料

- [修理依頼書](https://drive.google.com/file/d/1jST3NoFOJDP9b3Sy30Zn5SKQpO-b_ABl/view?usp=sharing)
- [ガタの計測データ](https://docs.google.com/spreadsheets/d/126NKqPHB6cZfGgrsZRfF1AyQ5ANTWIGyacWOr1V22nY/edit?usp=sharing)
- [具体的な故障の症状について](https://docs.google.com/spreadsheets/d/1ZaGgKU-MvOv61YyueEe9KWpUxBayAMwD_PVrgFvNIPU/edit?usp=sharing)

以下，ガタの修理に関する説明の資料

- [B3Mサーボモータ 出力軸のガタの修理のお願い](https://docs.google.com/document/d/1eS3MotILW0Cd6Su8ZHDDKlfdwfhZMdM7ynCiLo1yAco/edit?usp=sharing)
- [B3Mサーボモータ出力軸に生じるガタの計測方法](https://docs.google.com/document/d/1Cj-SPKhQhi2--01exauldr-KtiqmdFxZvRtyHA1_GCQ/edit?usp=sharing)

お忙しいとは思いますが確認の方をお願いします．特に問題なさそうであれば，12/15には修理に出したいと思います．

hiroki-0001, 2022-12-14:
@yasuohayashibara 

大学側に出す書類には上記で作成したもの以外に以下の4種類があります．

- 修理代の見積書(社印があるもの)
    - 修理依頼書を近藤科学に配送した後にメールで送られてきます
- 修理作業工程表

- 修理完了報告書
    - 納品書に修理時の作業内容を記入したもので代用可能

- 修理後のB3Mサーボモータの写真

---

この内の**修理作業工程表**は例年だと，林原先生が作成してくださっているのですが，今回もお願いできますでしょうか？もし作成が厳しそうであれば，例年のものを送ってくだされば代わりに自分が作成します．

hiroki-0001, 2022-12-15:
修理予定のB3Mを発送しました

hiroki-0001, 2023-02-08:
修理から帰ってきたので、ぼちぼち出力とガタの計測を行います

closed 2023-07-25 by MasatoKubotera