SUSTAINA-OP/SUSTAINA-OP #343: MPU-9250の後継品への移行

MasatoKubotera, 2022-09-13:
RoboCup 2022仕様であるSUSTAINA-OP Ver1.0に実装されるストロベリーリナックス製のMPU-9250モジュール（https://strawberry-linux.com/catalog/items?code=12250）が生産終了となったため，RoboCup2023に向けて後継品に移行する．

TDK InvenSense MPU-9250については，RoboCup2022の開発時点ですでに生産終了となっていたが，開発のリソース的に見送っていた．

後継品については，TDK InvenSense ICM-20948がある．

ICM-20948
https://product.tdk.com/ja/search/sensor/mortion-inertial/imu/info?part_no=ICM-20948

TDK InvenSense ICM-20948については，MPU-9250のようなストロベリーリナックス製のモジュールが販売されている．
[ICM-20948 ９軸センサモジュール(３軸加速度＋３軸ジャイロ＋３軸コンパス)](https://strawberry-linux.com/catalog/items?code=20948)

しかし，I/O電圧の違いにより，モジュールのみを交換するだけで後継品に移行することはできない．
Jetson Xavier NX では，3V3レベルのI2C接続であるため，[TCA9406 1.8V/3.3Vレギュレータ付 I2Cレベル変換モジュール](https://strawberry-linux.com/catalog/items?code=19406)などのレベル変換が必要となる．

また，ソフトウェア開発でもレジスタマップがMPU-9250とICM-20948は異なるので後継品の移行に伴い変更が必要となる．

現状，https://github.com/RTIMULib/RTIMULib2 を使ってソフトウェア開発が行われているが対応していないため，こちらの開発もまた必要である．

参考：

- https://github.com/citbrains/citbrains_humanoid/issues/1777
- https://github.com/citbrains/citbrains_humanoid/issues/1620
- https://github.com/citbrains/citbrains_humanoid/issues/1594

MasatoKubotera, 2022-09-13:
SUSTAINA-OPはオープンプラットフォームとして公開しているため，日本でのみ販売されているストロベリーリナックス製のモジュールを使うか否かも検討の余地がある．

もう一つの案としては，ICM-20948が実装されたSparkFun製のSEN-15335がある．
- [SEN-15335 (Mouser)](https://www.mouser.jp/ProductDetail/SparkFun/SEN-15335?qs=uwxL4vQweFMcls1MYZT00A%3D%3D)
- [SEN-15335 (Digi-Key)]( https://www.digikey.jp/ja/products/detail/sparkfun-electronics/SEN-15335/10279707?utm_adgroup=IoT&utm_source=google&utm_medium=cpc&utm_campaign=PMax%20Shopping_Product_IoT&utm_term=&productid=10279707&gclid=CjwKCAjw1ICZBhAzEiwAFfvFhJYGmGpQfJbpWrPVaCATqOI1AaKjqeMuE6Itw66jri0-6P1Kz8IVLBoCqzoQAvD_BwE)

こちらのボードは，センサとレベル変換が同一モジュールに収められているため，直接Jetson Xavier NXに接続することができる．

MasatoKubotera, 2022-09-13:
SEN-15335については，SparkFunがドキュメントを様々な公開しています．
[Schematic](https://cdn.sparkfun.com/assets/c/7/8/f/8/15335_9DoF_Schematic.pdf)
[Eagle Files](https://cdn.sparkfun.com/assets/e/3/8/8/4/15335_9DoF_IMU_EagleFiles.zip)
[Hookup Guide](https://learn.sparkfun.com/tutorials/sparkfun-9dof-imu-icm-20948-breakout-hookup-guide)
[Datasheet](https://cdn.sparkfun.com/assets/7/f/e/c/d/DS-000189-ICM-20948-v1.3.pdf) (ICM-20948)
[Arduino Library](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary)
[Python Support (Qwiic_Py)](https://github.com/sparkfun/Qwiic_9DoF_IMU_ICM20948_Py)
[GitHub Hardware Repo](https://github.com/sparkfun/SparkFun_Qwiic_9DoF_IMU_Breakout)

MasatoKubotera, 2022-09-15:
参考：
- [ぱわぷろ活動日誌 ジャイロセンサ「ICM20948」で苦しんだ話](https://ss-sholaw-wmmc.hatenablog.com/entry/2020/12/12/235938)

MasatoKubotera, 2022-09-16:
[SEN-15335](https://www.mouser.jp/ProductDetail/SparkFun/SEN-15335?qs=uwxL4vQweFMcls1MYZT00A%3D%3D)を現状の拡張基板に載せるとなると基板設計がかなり難しくなりそうです．

[image]

下図の緑のエリアがセンサエリアになります．

[image]

MasatoKubotera, 2022-09-17:
候補3

Multiple Function Sensor Development Tools 6 Degrees of Freedom Breakout - LSM6DSO (Qwiic)

地磁気センサーが入っていない、6軸センサーです。

上記であげた、ICM-20948チップが実装されたSEN-15335よりも価格帯が安く在庫もありますが、LSM6DSOチップで情報量が少ない。

RTIMULIBも非対応
https://github.com/RTIMULib/RTIMULib2

>加速度センサとジャイロセンサに9 KB FIFOバッファと組み込みプロセス割り込み機能がついたブレイクアウトボードです。高性能で安価な慣性計測装置（IMU）LSM6DSOを搭載しています。用途に合わせて柔軟に設定が可能な設計です。衝撃、傾き、動き、タップ、歩数、温度の計測などが可能です。

https://www.sparkfun.com/products/18020

https://www.switch-science.com/catalog/7182/

https://www.digikey.jp/ja/products/detail/sparkfun-electronics/SEN-18020/14313960

https://www.mouser.jp/ProductDetail/SparkFun/SEN-18020?qs=Jslch3jnSjkJ2et%2FXANMTw%3D%3D

MasatoKubotera, 2022-09-19:
ICM-20948が実装されたSparkFun SEN-15335を搭載できる，AverMedia Carrier Board EN715の拡張基板の修正版の設計しておきました．（PCB未配線）

[image]

データは下記にあります．
EN715_ExpansionBoard_ver1_2 https://github.com/MasatoKubotera/EN715_ExpansionBoard_ver1_2

hiroki-0001, 2022-09-22:
そういえば思い出したので共有しておくと、EN715の信号ピンは前回オシロスコープを使って確認した際にI2Cの信号は来てたけどもSPI通信関連の信号は来てなかったので、使用するIMUはI2C対応のものじゃないと無理かも

今のところ上がっている候補3つはすべてI2C対応だから問題ないとは思うけど念のためです

MasatoKubotera, 2022-09-22:
@hiroki-0001 
今回もI2C接続を前提として，SPI接続もできるようにしておきました．

前回（RoboCup2022）のMPU-9250を使った開発は，https://github.com/RTIMULib/RTIMULib2 を前提にしていたと思いますが，今回はもっと下層からの開発になるため選択できるのであれば将来的にはSPIを選択した方が良いかもしれません．
https://www.rohm.co.jp/electronics-basics/memory/memory_what7

MasatoKubotera, 2022-09-29:
参考

Migrating from MPU-9250 to ICM-20948 (MPU-9250からICM-20948への移行について)
機能比較やハードウェア比較，ソフトウェア比較について書いてあります．
 - https://invensense.tdk.com/wp-content/uploads/2018/10/AN-000146-v2.0-TDK_Migration_MPU_9250toICM-20948.pdf

レジスタマップ
- MPU-9250 : https://invensense.tdk.com/wp-content/uploads/2017/11/RM-MPU-9250A-00-v1.6.pdf
- ICM-20948 : https://invensense.tdk.com/wp-content/uploads/2021/10/DS-000189-ICM-20948-v1.5.pdf
  7 REGISTER MAP FOR GYROSCOPE AND ACCELEROMETER を参照．

hiroki-0001, 2022-10-05:
@MasatoKubotera
先日入江さんにIMUの移行に関して相談したところ

**移行予定のICM-20948はディスコンになる可能性がある**と

指摘をいただきました。

詳細に関しては以下のpdfに記載されてあります。
https://www.symmetryelectronics.com/getmedia/98a0ce24-a39e-46c6-a7b9-90187746be98/AN-000360-ICM-20948-Transition-Guide-v1-1.pdf

**今すぐなくなるわけではなく、pdfには供給上の懸念から新規開発には向かない**との記載がありました。

またTDKが推奨する移行先として

- ICM-42670-P
- ICM-42688-P

の2つがありますが、ICM-42688-Pのほうが市場ターゲットにロボティクスと記載があったこと、センサ自体の精度もこちらのほうが高いことからICM-42688-Pを利用したほうが良いような気がします。

開発コストに関してですが、ICM-20948でもICM-42688-Pでも今のところ大きくソフトウェアを書き換える必要がありそうなことには変わりないので個人的にはどちらでも良いとは考えています。

しかし、部品在庫や市場の流れ。回路に組み込む際の労力等に関してはそちらのほうが詳しいと思うので、最終的な判断は委ねます

hiroki-0001, 2022-10-05:
また開発の際には以下の評価ボードを使用したほうが、はんだ等の回路的な問題で開発が滞るということがないので良さそうです。

- MIKROE-4237
[購入リンク](https://www.digikey.jp/ja/products/detail/mikroelektronika/MIKROE-4237/13166617?utm_adgroup=General&utm_source=google&utm_medium=cpc&utm_campaign=PMax%20Shopping_Product_Zombie%20SKUs&utm_term=&productid=13166617&gclid=CjwKCAjws--ZBhAXEiwAv-RNL1jQjSui7xikmHFMgA-SsNa2W_g_lzHeWykSvKeDdMu_IdJEKUb34xoCuFUQAvD_BwE)

- ICM-42688-P EVAL BOARD【DK-42688-P】
[購入リンク](https://www.marutsu.co.jp/pc/i/35058598/)

hiroki-0001, 2022-10-05:
また既に購入してしまっている

- SEN-15335
- SEN-18020

に関してはまだ未開封状態です。

MasatoKubotera, 2022-10-05:
## ICM-42688-Pに関して

## 性能比較
現行のロボットで必要なスペックかは置いておいて，[ICM-20948 Transition Guide](https://www.symmetryelectronics.com/getmedia/98a0ce24-a39e-46c6-a7b9-90187746be98/AN-000360-ICM-20948-Transition-Guide-v1-1.pdf) を見ても性能的に**ICM-42688-P** > **ICM-20948**．

## モジュール

実ロボットに搭載するときには，モジュール品を利用することになります．

その点に関して，ICM-20948とICM-42688-Pの大きな違いとしては，モジュール展開しているメーカが少ないことが挙げられる．

<a href="https://www.mouser.jp/ProductDetail/TDK-InvenSense/ICM-42688-P?qs=OlC7AqGiEDk8QrNAV728zw%3D%3D">ICM-42688-P (Mouser)</a>のモジュール
- <a href="https://www.mouser.jp/ProductDetail/Mikroe/MIKROE-4237?qs=hWgE7mdIu5Th0Vmu4C82%2FQ%3D%3">DMIKROE-4237 (Mouser)</a>

TDKからは評価ボードが出ているが，MikroElektronikaが出しているモジュール1種であり取り扱いもMouser，digikeyともにあるがsparkfunと比較すると在庫的に心配なところがある．

メーカページにはsparkfunと同等な使用は公開されている．
https://www.mikroe.com/6dof-imu-14-click

現行のExpansionBoard などに載せるとなっても，モジュールサイズ，固定方法的に自分たちでモジュールことになる確率が高いです．難易度的には低い．

チップが下記の状態であるため，年明けになりそう．
<a href="https://www.mouser.jp/ProductDetail/TDK-InvenSense/ICM-42688-P?qs=OlC7AqGiEDk8QrNAV728zw%3D%3D">ICM-42688-P (Mouser)</a>
取寄中:
888予想11-1-23
14,890予想20-1-23

---

以下メモ

<a href="https://www.mouser.jp/ProductDetail/TDK-InvenSense/ICM-42688-P?qs=OlC7AqGiEDk8QrNAV728zw%3D%3D">ICM-42688-P (Mouser)</a>
取寄中:
888予想11-1-23
14,890予想20-1-23
  - <a href="https://www.mouser.jp/ProductDetail/Mikroe/MIKROE-4237?qs=hWgE7mdIu5Th0Vmu4C82%2FQ%3D%3">DMIKROE-4237 (Mouser)</a> モジュール品
  - <a href="https://www.mouser.jp/ProductDetail/TDK-InvenSense/DK-42688-P?qs=OlC7AqGiEDmi4%252Blydwet0Q%3D%3D">DK-42688-P (Mouser)</a>
    - Development Kit for ICM-42688-P 6-Axis Motion Sensor 
    - <a href="https://product.tdk.com/system/files/dam/doc/product/sensor/mortion-inertial/imu/data_sheet/dk-42688-p-product-brief.pdf ">Product Brief</a>
  - <a href="https://www.mouser.jp/ProductDetail/TDK-InvenSense/EV_ICM-42688-P?qs=hWgE7mdIu5SqHj7syF9KYA%3D%3D">EV_ICM-42688-P (Mouser)</a>

ICM42688 Arduino Library
https://www.arduino.cc/reference/en/libraries/icm42688/

MasatoKubotera, 2022-10-05:
ハードウェアの開発コスト < ソフトウェアの開発コスト であり，

https://readable.jp/files/al-4aae8f8ae46443608c684d6cbcf96dc8.pdf

<blockquote>
The ICM-20948 has been a staple in the TDK InvenSense IMU portfolio for over 6 years. To date it is still one of best
performance 9-axis consumer IMU devices in the market and a very popular part with our customers. However
due to supply concerns, we recommend that customers do not use ICM-20948 for new developments and
investigate alternative IMUs for their application.
This guide will help answer the questions that customers will have in choosing an IMU to replace the ICM-20948. 
</blockquote>

と公式が提言しているのであれば，従った方が良いかと思います．

hiroki-0001, 2022-10-05:
@MasatoKubotera 
了解です。それならICM-42688-Pに移行する方針で進めていきましょう

MasatoKubotera, 2022-10-06:
@hiroki-0001 

下記リポジトリで試しにICM-42688-Pのセンサモジュールを作成しました．
[image]
https://github.com/MasatoKubotera/ICM-42688-P_Module

現状のSUSTAINA-OP(RoboCup2022仕様)で使用しているストロベリーリナックス製のMPU-9250センサモジュールと同等のペン配置・サイズにすることができたので，以前検討されてきたspakfun製のセンサモジュールを使用するために必要だった拡張基板(ExpansionBoard)の作成は不要になりそうです．
※モジュールを変更するだけで，ハードウェア側の変更は済みそうです．

MasatoKubotera, 2022-10-19:
手元に5V系のArduinoしかなかったので，レベル変換モジュールを通して，SPI，I2C接続でICM-42688-P搭載センサモジュール[MIKROE-4237](https://www.mikroe.com/6dof-imu-14-click)の動作確認を試みました．
- [８ビット双方向ロジックレベル変換モジュール](https://akizukidenshi.com/catalog/g/gM-04522/)
- [Arduino Nano Every](https://akizukidenshi.com/catalog/g/gM-14848/)
- [MIKROE-4237](https://www.mikroe.com/6dof-imu-14-click)

[image]

[MIKROE-4237](https://www.mikroe.com/6dof-imu-14-click)は，到着時にSPI接続の仕様になっているため，I2C接続ではジャンパ抵抗３個をはんだ付けしなおす必要がありました．
[image]

Arduino 公式ライブラリの[ICM42688](https://www.arduino.cc/reference/en/libraries/icm42688/)を使って検証してみましたが，接続できない旨を示す"IMU initialization unsuccessful"が表示されるだけでした．

３０分ほどしか時間をかけれなかったので，再度3.3V系のArduino などを使って検証します．

MasatoKubotera, 2022-10-26:
# SparkFun 9DoF IMU Breakout - ICM-20948 SEN-15335

ICM-20948を搭載したモジュールの[Arduino Due](https://akizukidenshi.com/catalog/g/gM-07386/)を使った動作確認が取れました．
製品 : [SparkFun 9DoF IMU Breakout - ICM-20948](https://www.sparkfun.com/products/15335)

Arduino DueをArduino IDEで使用するためには，ボードマネージャからボードライブラリをインストールする必要があります．

接続方法は下記の回路図に従っています．
[image]
※回路図出展 : https://github.com/MasatoKubotera/EN715_ExpansionBoard_ver1_2

通信方式はSUSTAINA-OPと同様のI2Cで確認しました．サンプルプログラムは下記からダウンロードできます．
サンプルプログラム : https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/archive/main.zip
サンプルプログラム Example2_Advanced内の<code>#define AD0_VAL 1</code>を<code>#define AD0_VAL 0</code>に変更する必要があります．

参考 : [SparkFun 9DoF IMU (ICM-20948) Breakout Hookup Guide](https://learn.sparkfun.com/tutorials/sparkfun-9dof-imu-icm-20948-breakout-hookup-guide?_ga=2.220520664.1100016345.1663489669-2064064442.1660846271#software-setup-and-example-code)

[image]
[image]

hiroki-0001, 2023-02-23:
ICM-42688-Pでの動作確認がArduino dueと[サンプルプログラム](https://github.com/finani/ICM42688)を用いることでできました。
Jetsonに搭載した際に同じようにデータ値が取得できるように検証していきます。
このissueに検証記録を記載してしまうとこのissueのログが巨大になるので、https://github.com/citbrains/citbrains_humanoid/issues/1966 で行います。

closed 2023-07-25 by MasatoKubotera