SUSTAINA-OP/SUSTAINA-OP #346: SUSTAINA-OP No.177 部品切削記録

MasatoKubotera, 2022-10-25:
SUSTAINA-OP 7機体目となる177機（No.177）の部品切削記録について残すIssue．

RikuYokoo, 2022-10-26:
# CAM報告
## CAMの名前
[POM_T5_01.iam](https://github.com/citbrains/SUSTAINA-OP/blob/number177/Cam/POM_T5/POM_T5_01.iam)
## 担当者名
@ChikaraHanakawa 
## commit ID
8dc0f872e72aac663ea43d4219d4edec7796ecce
## 配置したパーツと個数
Cockpit_side_frame * 1
Cockpit_battery_back_door * 1
Cockpit_top_frame * 1
Cockpit_side_frame_MIR * 1
## 切削時間
2:37:51
## 配置後のCAM画像
[image]

## 板厚公差
[image]
その他，共有事項等があれば載せる

## ボトムオフセット
ポケット切り抜き，輪郭を0にした

RikuYokoo, 2022-10-26:
# 切削報告
## 切削日
10月25日
## 切削担当
@ChikaraHanakawa @RikuYokoo 
## CAM名
[POM_T5_01.iam](https://github.com/citbrains/SUSTAINA-OP/blob/number177/Cam/POM_T5/POM_T5_01.iam)(8dc0f872e72aac663ea43d4219d4edec7796ecce)
## 使用エンドミル
WXL-3D-DE-2_01(共用エンドミル)
## 切削距離
42.1983m
## 所要時間
## 切削時間
2:37:51
## 切削したもの
Cockpit_side_frame * 1
Cockpit_battery_back_door * 1
Cockpit_top_frame * 1
Cockpit_side_frame_MIR * 1
## 切削後の写真

どの板を使ったかの説明
## 板圧交差
[image]
左下から時計周りに
-0.002
-0.003
-0.004
-0.01
-0.007
-0.017

## ボトムオフセット
オフセットは0にして行った

RikuYokoo, 2022-10-26:
# CAM報告
## CAMの名前(commit ID)
[POM_T5_02.iam](https://github.com/citbrains/SUSTAINA-OP/blob/number177/Cam/POM_T5/POM_T5_02.iam)(eab8ef6fe9fba93e88789256ea3780279a1e4f0a)
## 担当者名
@RikuYokoo 
## 配置したパーツと個数
Cockpit_carrier_board_frame * 1
## 切削時間
00:22:03
## 配置後のCAM画像
[image]

## 板厚公差
[image]

## ボトムオフセット
0に設定

RikuYokoo, 2022-10-26:
# 切削報告
## 切削日
10月26日
## 切削担当
@GakuKuwano @RikuYokoo @KiyoshiroKawanabe 
## CAM名
[POM_T5_02.iam](https://github.com/citbrains/SUSTAINA-OP/blob/number177/Cam/POM_T5/POM_T5_02.iam)(https://github.com/citbrains/SUSTAINA-OP/commit/eab8ef6fe9fba93e88789256ea3780279a1e4f0a)
## 使用エンドミル
WXL-3D-DE-2_01(共用エンドミル)
## 切削距離
6.32768m
## 所要時間
## 切削時間
00:22:03
## 切削したもの
Cockpit_carrier_board_frame * 1
## 切削後の写真
[image]

どの板を使ったかの説明
## 板圧交差
[image]
左下から時計周りに
-0.007
-0.009
-0.007
-0.01
-0.01
-0.01

## ボトムオフセット
オフセットは0にして行った

MasatoKubotera, 2023-03-07:
プロジェクトが中断されたのでCloseします．

closed 2023-03-07 by MasatoKubotera