SUSTAINA-OP/SUSTAINA-OP #347: サンプルバッテリの慣らし充電

MasatoKubotera, 2023-02-18:
DciTechに提供いただいたバッテリの慣らし充電の記録を行う．
RoboCup2022で用いたHyperion製バッテリの慣らし充電記録については，#230 にある．

MasatoKubotera, 2023-02-18:
バッテリ種類

提供いただいたバッテリは以下の2種類各10本です．

1. Prototype LiHV Battery-01 11.4V 2800mAh 25-50C
2. Prototype LiHV Battery-02 11.4V 2400mAh 25-50C

[image]

MasatoKubotera, 2023-02-18:
【準備】

１．書くバッテリにテプラで管理番号を添付
Prototype LiHV Battery-01 11.4V 2800mAh 25-50C : 23-2800-P-01～23-2800-P-10
Prototype LiHV Battery-02 11.4V 2400mAh 25-50C : 23-2400-P-01～23-2400-P-10

２．内部抵抗を計測（充電器で計測できる）

【慣らし充電の方法】

★　電圧記録
〇　0.2C（2800mAh:0.6A/2400mAh:0.5A）バランス充電
★　電圧・充放電容量・時間記録
〇　0.4C（2800mAh:1.2A/2400mAh:1.0A）放電（ストレージ）
★　電圧・充放電容量・時間記録
〇　0.7C（2800mAh:2.0A/2400mAh:1.7A）バランス充電
★　電圧・充放電容量・時間記録
〇　1.0C（2800mAh:2.8A/2400mAh:2.4A）放電（ストレージ）
★　電圧・充放電容量・時間記録
〇　1.0C（2800mAh:2.8A/2400mAh:2.4A）バランス充電
★　電圧・充放電容量・時間記録
〇　1.0C（2800mAh:2.8A/2400mAh:2.4A）放電（ストレージ）
★　電圧・充放電容量・時間記録

MasatoKubotera, 2023-02-18:
**重要**

提供いただいたバッテリのバッテリー側バランス端子の配列が，Hyperion製と異なり逆である．
接続の際には，注意が必要である．

量産版ではHyperionに合わせることが出来ないか打診する予定である．

黒XHコネクタ：Hyperion
白Xhコネクタ：提供バッテリ

[image]

MasatoKubotera, 2023-02-18:
管理番号とチーム名のテプラを添付しました．
QRコードは各管理番号になっていますので，
いつかカメラを使って認識してバッテリ管理できるシステムを構築します．

[image]

MibuchiYuta, 2023-02-19:
## 初回のメモ
バッテリ管理システムの用意ができていないため、[スプレッドシート](https://docs.google.com/spreadsheets/d/1kF5YmkmFDKDM6ZZ6VB_ocMgighTidKEMSVWuwV2aRFg/edit#gid=0)に記録していく。

充放電気は

MibuchiYuta, 2023-02-23:
## ケーブルのラベル

> 重要
>
>提供いただいたバッテリのバッテリー側バランス端子の配列が，Hyperion製と異なり逆である．
>接続の際には，注意が必要である．

こちらの問題に対して、2色のラベルをテプラで作成しわかりやすくしました。
それぞれ14本ずつ作成しました

<image src= "https://user-images.githubusercontent.com/53966307/220887165-fc7cab7d-54e0-43ff-afe4-eb8776c03601.jpg" width = "640">

MibuchiYuta, 2023-02-27:
> 

旧充放電器を含めて統一しました。

MibuchiYuta, 2023-02-27:
## 内部抵抗計測についての問題

充放電器の機能を使用して慣らし前に内部抵抗を計測していました。その際に2セル目の値が必ず高いことに違和感を感じたので調査を行ったところ**充放電器を用いた計測値は信頼できない**ことがわかりました。

### 問題が発覚した際の状況

新充放電器1番のチャンネルAで23-2800-P-07〜23-2800-P-10を計測すると以下のような値になった。

|  | 1セル目 | 2セル目 | 3セル目 
| :----- | :-----| :-----| :----- |
| 07 | 16.8 | 16.8 | 16.8 |
| 08 | 16.9 | 27.1 | 13.5 |
| 09 | 16.8 | 26.9 | 16.8 |
| 10 | 16.5 | 19.9 | 13.2 |

不思議に思い、23-2800-P-10をBチャンネルで計測したところ、以下の画像のようにAチャンネルとは違う値が出力された。
[image]

### 行った調査

上記の問題に関して原因究明のために以下の検証を行った。

- 再計測
- バランス端子を交換

MibuchiYuta, 2023-02-27:
## バランス端子の長さに関して
Hyperion製のバランス端子と比較すると30mmほど短く、旧充放電器では画像のように取り回しがしにくかったです。もし可能でしたら、量産型はもう少し長いと作業しやすいと思います。
[image]?

MibuchiYuta, 2023-03-04:
3周分の慣らし充電が完了しました。

closed 2023-07-25 by MasatoKubotera